espressif/esp-idf #2434: ble client connect with public address

chegewara, 2018-09-19:
Is it right behavior if we have `esp_ble_gattc_open` setup with `BLE_ADDR_TYPE_PUBLIC` and esp32 is connecting to peer device with random if we set local privacy to true `esp_ble_gap_config_local_privacy(true);`?
If yes, then it seems to be redundant, because it should be enough to set any value but `BLE_ADDR_TYPE_PUBLIC` to get random mac.

To be clear, im not complaining just trying to understand this option.

Weijian-Espressif, 2018-09-25:
@chegewara esp_ble_gattc_open(esp_gatt_if_t gattc_if, esp_bd_addr_t remote_bda, esp_ble_addr_type_t remote_addr_type, bool is_direct)
 'remote_addr_type' is remote device bluetooth device the address type.

chegewara, 2018-09-25:
@Weijian-Espressif Sorry, i dont know how i did miss that. But i dont understand something in that case. How do i know what address type got remote device and, what is more important, in almost all applications i have been trying ive got `BLE_ADDR_TYPE_PUBLIC`, no matter if ive been connecting with peer device with public address or with random address and i saw no difference. Maybe i am missing something here. Thanks for patience and answering my noob questions.

Weijian-Espressif, 2018-09-25:
@chegewara  you will get remote device address type  "scan_result->scan_rst.ble_addr_type" when you scan.

chegewara, 2018-09-25:
@Weijian-Espressif Thanks for explanation. Now when i understand that i think we have bug in @nkolban library, because we have hardcoded remote address type to PUBLIC, but like i said, it makes no difference (as far as i know) if we connect this way to peer device with public or random address.
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEClient.cpp#L114

Weijian-Espressif, 2018-09-26:
@chegewara Connecting to a random address device may fail. The number of adv packets is limited in lower layer, and sometimes the adv information of this device cannot be found.  

For example, connect a random address device, if we don't find the adv information,  We use the public type to connect, the connection will fail.  please refer to "BOOLEAN l2cble_init_direct_conn (tL2C_LCB *p_lcb)" in l2c_ble.c file.

closed 2018-09-26 by chegewara