espressif/esp-idf #2437: [feature request][build system][make][cmake] Make simpler running test from component/project directory

nicola-lunghi, 2018-09-19:
Dear customer support,

thanks for your support first you are a very responsive company!

Here's the issue
currently developing unit test for a module is a painful experience.
the ideal world would be that each module could have a "make test"
target that could be launched from the commandline (or a idf.py test?)

I think that now with the cmake system and the possiblity to specify module dependency could also be possible to have a make test inside the project directory that include all the test for the components that are needed by the project or for example

1) I specify that my application needs bt wifi and i2c components
2) i put my tests file in the test folder
3) i do a "make test"
4) I do a make flash
5) I have all the test for my main and components executed into the esp32

I know that is a much to ask but while you are revisiting the cmake system maybe is worth to have a look at it

Best Regard
Nicola Lunghi

ESP-Marius, 2022-08-31:
Hi!

Although we don't have exactly what you are describing here, I still think our new approach to testing with `pytest embedded` is pretty close to what you want.

See [Running tests locally](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/contribute/esp-idf-tests-with-pytest.html#run-the-tests-locally) for how to easily build and run a test app.

dobairoland, 2025-03-14:
I agree with the previous post. I don't think we will near term change that approach.

closed 2025-03-14 by dobairoland