espressif/esp-idf #2438: Bad i2c driver optimization (very important for big data flow, like screen buffers transfer, for example)

FoxKeys, 2018-09-19:
Hi all.

I have found not a bag, but very unoptimal code in i2c driver, seriously reducing maximum data rate.

In fact, CPU time to create and destroy cmd_link (i2c_cmd_link_create + i2c_master_start + i2c_master_write_byte (*1024) + i2c_master_stop + i2c_cmd_link_delete) is about equal to real data transfer time (i2c_master_cmd_begin)

Why so bad? I have found next things.
1. Big CPU and memory overhead from malloc/memcpy (especially, if use i2c_master_write_byte). But i2c_master_write is not much better, because it is still chunks provided data buffer by 255 bytes pieces and still make a copy of data anyway. So, it have less malloc but same memcpy. And there is no function to give prepared data buffer to driver and use it _without_ making copy.
This is pretty bad when working with things like big screen buffer, for example. Consume about size * 2 additional memory (because of cmd fields overhead. If using i2c_master_write_byte - it is about 6 times overhead because of additional cmd fields)

2. Another bad thing - once prepared cmd_link **can not be reused**. It is "damaged" while data transfer and *must* be destroyed after i2c_master_cmd_begin and created again even i  want to send same data in loop! Very, very bad solution!

So, because of this 2 things (and others i'm not found) i have about 60 fps on my screen instead of "theoretical" 120 fps on 1MHz i2c. And, this is with 100% utilization of one CPU core! Incredible....

Unfortunately, i'm not i2c expert and can't contribute usable solution. In current version of driver data buffering is pretty tight mixed with data transfer process.
So, i only can give some suggestions.

1. Simplest and important. Change transfer functions such way, that they _don not_ damage prepared data buffer (cmd_link created by i2c_cmd_link_create), so same link can be used many times to transfer same data.

2. Use data array instead of linked list for data storage and add functions for direct access to data in this buffer. This will be very usable for screen buffers. So, no more need to recreate whole cmd_link if i need to chage only one pixel!

3. Provide not-copy version of i2c_master_write, that just receive pointer to data buffer (same, good for screen buffer-like usages.

4. Export (put in i2c.h) all required data buffer structs, like i2c_cmd_t so it can be filled by end user if needed. Export required cmd_link struct to get direct access to data buffer. Usefull when user want many i2c_master_write_byte (just change data byte, but not all other fields and call i2c_master_write_byte for every data byte. Currently - it is creates new copy of struct on every call)

P.S. Unfortunately, i'm not so expirienced to hel you with coding such things, but ready to help any other way (testing, suggesting, etc)

Best regards, 
Sergey.

michprev, 2018-09-19:
You better should write your own I2C driver or move to an other MCU.

I still don't understand how could anyone think that using FreeRTOS and mallocs is a good idea in low level hardware drivers. You can take a look at https://esp32.com/viewtopic.php?f=2&t=6491

Especially ESP_Sprite's post:

> With regards to the separation between lower-level drivers and higher-level (FreeRTOS or something else based): This is on our ToDo list. We are aware that this would be helpful so people porting everything to a different OS. However, again, you have to keep in mind our priorities are partially based on the use cases of our customers, so it's not something that will be implemented tomorrow.

FoxKeys, 2018-09-19:
Yes. Not all ESP-IDF code is good.

But. 
1. Unlike of many other modern MCUs (will not write which exact, but, sure all knows which is) - Espressif provide almost all source code on github. And, this is very, very good thing, unlike of precompiled *.a libraries provided by many competitors.

2. FreeRTOS it self is *not* a problem for driver writing, because it mainly depend on ISR, but not on RTOS functions.

3. malloc is not RTOS part, but C/C++ std library part. Or i mistake? 

4. Because of ESP-IDF is open source - you can write better own driver, of course. In theory. But. There is one little problem. There is no publically awailable detailed ESP32 datasheet. Only "short" version, without registers and other hardware details. Or i mistake again?

Anyway, unfortunately i'm not smart enought to write own driver. So, this is why i'm opened this issue. My level is only enough to _find_ and _describe_ problem. But not to fix. But, this is open sourse, and there is many much more smart peoples present here. Hope, somebody can do better solution.

michprev, 2018-09-29:
1. I  am not sure which one you mean. Every MCU I've been working with had open source drivers.

2. It adds delays that are not necessary. Also if you decide that you want to run FreeRTOS only on the first core you have to implement all drivers yourself.

3. Sure it is not a part of RTOS. Some developers even refuse to use dynamic allocation on embedded systems. Again it adds additional delays that are not necessary.

4. Take a look for "Technical reference manual". It seems quite incomplete to me but I2C chapter should be covered fine.

I am implementing OS-free drivers for ESP32 but I have not even started working on I2C driver yet.

koobest, 2020-05-27:
Hi, @FoxKeys 

The data must be copied to fifo in order to send it out and there is no copy you said elsewhere.
If you want to send the same data, you can reuse the link by calling ` i2c_master_cmd_begin` directly.

Alvin1Zhang, 2020-06-02:
@FoxKeys Thanks for reporting, would you please help share if any updates for the issue? Thanks.

FoxKeys, 2020-06-02:
@Alvin1Zhang Unfortunately, i have no active projects with I2C right now. Will write, if get something in future...

Alvin1Zhang, 2020-06-02:
@FoxKeys Thanks for sharing the updates, would you mind closing the issue for now? Feel free to reopen if any updates. Thanks.

o-marshmallow, 2021-11-17:
In newer versions of IDF, it is now possible to re-use the I2C buffer, reducing a lot the overhead. Moreover, it is also possible to pass a buffer that will be used by the driver. In that case, no malloc/free will be performed anymore underneath.
Closing this issue. Please open a new issue if you still encounter this problem.

closed 2021-11-17 by o-marshmallow