espressif/esp-idf #2439: Registers functionality notes

michprev, 2018-09-19:
All registers should be rechecked and notes should be added to registers that are unfunctional or partly functional. Unfunctional registers might be even removed.

For example take a look at SPI peripheral.
`flash_read` does not work at all
https://github.com/espressif/esp-idf/blob/3276a1316f66a923ee2e75b9bd5c7f1006d160f5/components/soc/esp32/include/soc/spi_struct.h#L40

`*_hold` do not work
https://github.com/espressif/esp-idf/blob/3276a1316f66a923ee2e75b9bd5c7f1006d160f5/components/soc/esp32/include/soc/spi_struct.h#L124-L129

`cs_setup / setup_time` does not work in full-duplex
https://github.com/espressif/esp-idf/blob/3276a1316f66a923ee2e75b9bd5c7f1006d160f5/components/soc/esp32/include/soc/spi_struct.h#L112
https://github.com/espressif/esp-idf/blob/3276a1316f66a923ee2e75b9bd5c7f1006d160f5/components/soc/esp32/include/soc/spi_struct.h#L84

and there are probably more and more.

michprev, 2018-11-17:
After two years on the market still no info about what is working and what is not. Awesome.

Will this ever be done?

ginkgm, 2018-11-19:
@michprev for flash registers, they are functional, just not for the driver, it's used in the ROM or spi_flash.

For other registers, will check it more.

michprev, 2018-11-19:
@ginkgm so you mean that flash registers are working only on SPI1? I have tried a few of them on HSPI and they seem to work the same way as on SPI1.

However they are certainly not fully functional:
https://github.com/espressif/esp-idf/blob/960c240578d359a1c5cf8cb3cbd9ce3150d5a370/components/spi_flash/spi_flash_rom_patch.c#L599-L603

Here SPI user command is issued (setting SPI_USR bit of SPI_CMD_REG) instead of  using `flash_read` which does not work.
https://github.com/espressif/esp-idf/blob/960c240578d359a1c5cf8cb3cbd9ce3150d5a370/components/spi_flash/spi_flash_rom_patch.c#L198-L245