espressif/esp-idf #2442: UDP Client Connection

MHANE, 2018-09-19:
Hello,

i try to implement UDP Client and have an issue, if the udp server on PC Side is not started.
I expect that the function connect () returns an error if no udp server with a assigned port is started
 
    Code:
     .....
     S = socket (AF_INET, SOCK_DGRAM, IPPROTO_UDP); 
     if(connect(S, (struct sockaddr *)&sa_udpclient, sizeof(sa_udpclient)) <0)
     {
            close(S);
     }
     ......
     if( write(S ,&Buffer[0] ,10) < 0)
     {
            close(S);
     }

Normally he should get an error and close the Socket, but the condition is evertime false, so the esp32 send upd data. With wireshark i see that the esp32 transmit everytime new upd packets. This happens also if the Server ip is not available, in this case i use the terminal, and this shows also that data is transmitted, so both conditions evertimes are false. 

By using TCP Protocol this lines working finely and the connect() returns evertime an error.

Now, what is the difference in the UDP Protocol? 
Thanks for help.
Best regards

rinripper, 2018-09-19:
Same problem here

zhangyanjiaoesp, 2018-09-29:
@MHANE @rinripper 
The purpose of calling the `connect()`  function in the UDP is just to specify the  remote IP address and port where the UDP data will be sent to. It will not create a connection between the two targets as TCP does. 

So it doesn't make sense for you to use the `connect()`  function to detect whether the remote udp port is open or closed.

Here is the udp_connect() in the lwip, you can use it as a reference:

> udp_connect(struct udp_pcb *pcb, const ip_addr_t *ipaddr, u16_t port)
   {
      struct udp_pcb *ipcb;
      if ((pcb == NULL) || (ipaddr == NULL)) {
        return ERR_VAL;
      }
      if (pcb->local_port == 0) {
        err_t err = udp_bind(pcb, &pcb->local_ip, pcb->local_port);
        if (err != ERR_OK) {
          return err;
        }
     }
     **ip_addr_set_ipaddr(&pcb->remote_ip, ipaddr);
     pcb->remote_port = port;**
     pcb->flags |= UDP_FLAGS_CONNECTED;
     for (ipcb = udp_pcbs; ipcb != NULL; ipcb = ipcb->next) {
        if (pcb == ipcb) {
           return ERR_OK;
        }
    }
    pcb->next = udp_pcbs;
    udp_pcbs = pcb;
    return ERR_OK;
}
>

Alvin1Zhang, 2019-02-14:
@MHANE Feel free to reopen this issue if your issue still exists. Thanks.

closed 2019-02-14 by Alvin1Zhang