espressif/esp-idf #2443: How to set Service UUID in advertisement data with CONFIG_SET_RAW_ADV_DATA?

hopkinskong, 2018-09-20:
I have tried setting the following `adv_data`, same with `scan_rsp_data` but with `.set_scap_resp=true`:
```
static esp_ble_adv_data_t adv_data = {
	.set_scan_rsp = false,
	.include_name = true,
	.include_txpower = false,
	.min_interval = 0x20,
	.max_interval = 0x40,
	.appearance = 0x00,
	.manufacturer_len = 0,
	.p_manufacturer_data =  NULL, 
	.service_data_len = 0,
	.p_service_data = NULL,
	.service_uuid_len = 16,
	.p_service_uuid = my_uuid,
	.flag = (ESP_BLE_ADV_FLAG_GEN_DISC | ESP_BLE_ADV_FLAG_BREDR_NOT_SPT),
};
```
`my_uuid[16]`:
```
0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x53, 0xE5, 0x00, 0x00, 0x00,
```

As you can see, my service UUID is `0xE553`.

However, when I scan the ESP32, there is no UUID in the advertisement packet:
`0201061009494f54736149664774785a4d6f367504ff454945051220004000`

Taking a simple BLE device as an example (with UUID `0x1810`), it should be like this:
`02011a030310180f09424c55205065726970686572616c`

dhrishi, 2018-09-20:
@hopkinskong If your service UUID is 0xE553 then they should be present at index 12 and 13 (starting from index 0) and not index 11 and 12 as used above.

Can you try the same with:
0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0x53, 0xE5, 0x00, 0x00,

hopkinskong, 2018-09-20:
@dhrishi Done, the current scan response is:
`0201061009494f547252504d5379364c4d505130051220004000`

Still no service UUID in advertisement packet...

dhrishi, 2018-09-20:
@hopkinskong  It is working fine at my end

Can you try the example in IDF - examples/bluetooth/gatt_server_service_table/main/gatts_table_creat_demo.c with a couple of modifications:
1. Comment this line: #define CONFIG_SET_RAW_ADV_DATA
2. Replace service_uuid contents with your UUID:
0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0x53, 0xE5, 0x00, 0x00,

hopkinskong, 2018-09-20:
@dhrishi You are using `CONFIG_SET_RAW_ADV_DATA`?

dhrishi, 2018-09-20:
No. I have commented the "#define CONFIG_SET_RAW_ADV_DATA" line in the example

hopkinskong, 2018-09-20:
You have commented out the `#define` line, which means you are using RAW ADV DATA?

hopkinskong, 2018-09-20:
Oh, never mind, I thought you meant `#ifdef` instead of `#define`.

chegewara, 2018-09-25:
@hopkinskong You have set `.include_name = true`, how long in chars is your device name? It is possible that you exceeded 31 bytes in advertising packet and this is the reason you cant see service UUID.

Alvin1Zhang, 2019-03-21:
@hopkinskong Hi, would you share if any updates for the issue? Thanks.

closed 2019-05-16 by Alvin1Zhang