espressif/esp-idf #2447: ARP request of 42 bytes crash the app

bricewge, 2018-09-20:
## Environment
- Core: ESP-WROOM32
- IDF version: v3.1
- Development Env: Mongoose OS
- Operating System: NixOS
- Power Supply:  external 3.3V

## Problem Description
When an ARP request of 42 byte is broadcasted on the network it crash the app. It happen just after`dhcps: send_offer>>udp_sendto result 0` is logged
This problem appear in v3.1; in v3.0 I don't have this issue.

### Expected Behavior
Does not crash when an other device connect to the network

### Actual Behavior
Crash when a device connect to the same network.

### Steps to repropduce
1. Connect your device via WiFi and for it to get it's IP.
2. Connect an other device (esp32, linux or android) to the same network. It should send an ARP request of 42 bytes.

## Debug Logs

```
[Sep 20 16:11:22.013] dhcps: send_offer>>udp_sendto result 0
[Sep 20 16:11:22.013] Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
[Sep 20 16:11:22.013] Debug exception reason: Stack canary watchpoint triggered (tiT)
[Sep 20 16:11:22.029] Core 0 register dump:
[Sep 20 16:11:22.029] PC      : 0x4008d93b  PS      : 0x00060436  A0      : 0x8008dc42  A1      : 0x3ffbc060
[Sep 20 16:11:22.029] A2      : 0x3ffb8c08  A3      : 0x3ffb9b34  A4      : 0x00000010  A5      : 0x3ffb99ec
[Sep 20 16:11:22.045] A6      : 0x3ffb9b4c  A7      : 0x00000014  A8      : 0x8008d906  A9      : 0x3ffbc040
[Sep 20 16:11:22.045] A10     : 0x3ffb9b34  A11     : 0xfffffffc  A12     : 0x3ffb9b34  A13     : 0x3ffbc424
[Sep 20 16:11:22.061] A14     : 0x00000250  A15     : 0x00000003  SAR     : 0x00000004  EXCCAUSE: 0x00000001
[Sep 20 16:11:22.061] EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc
[Sep 20 16:11:22.077]
[Sep 20 16:11:22.077] Backtrace: 0x4008d93b 0x4008dc3f 0x40081944 0x400819a8 0x40081e31 0x4000beaf 0x400dbcb4 0x400d39ed 0x4015419e 0x401541da 0x40169376 0x40169415 0x400f3972 0x400e2195 0x400e1560 0x400e34fd 0x400e1081 0x4000bd83 0x4000117d 0x400592fe 0x4005937a 0x40058bbf 0x400d6229 0x400dba12 0x400d24b1 0x4014cd29 0x40144357 0x401483dd 0x4014bffa 0x4013f60d
[Sep 20 16:11:22.109] --- BEGIN CORE DUMP ---
[Sep 20 16:11:22.109] mos: catching core dump
[Sep 20 16:11:24.957] ..[Sep 20 16:11:28.604] mos: core dump aborted
[Sep 20 16:11:28.604]
[Sep 20 16:11:28.604] rst:0x9 (RTCWDT_SYS_RESET),boot:0x32 (SPI_FAST_FLASH_BOOT)
[Sep 20 16:11:28.604] configsip: 0, SPIWP:0xee
[Sep 20 16:11:28.604] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[Sep 20 16:11:28.604] mode:DIO, clock div:2
[Sep 20 16:11:28.620] load:0x3fff0018,len:4
[Sep 20 16:11:28.620] load:0x3fff001c,len:5792
[Sep 20 16:11:28.620] ho 0 tail 12 room 4
[Sep 20 16:11:28.620] load:0x40078000,len:8252
[Sep 20 16:11:28.620] load:0x40080400,len:6716
[Sep 20 16:11:28.620] entry 0x40080760
[Sep 20 16:11:28.620] W (61) boot: PRO CPU has been reset by WDT.
[Sep 20 16:11:28.636] W (61) boot: WDT reset info: PRO CPU PC=0x4008e0e0
[Sep 20 16:11:28.636] W (61) boot: WDT reset info: APP CPU PC=0x22dff448
[Sep 20 16:11:28.636] I (68) boot: ESP-IDF v3.1-r1 2nd stage bootloader
```
Note that I wasn't able to get the core dump from this crash.

IIM-Slide, 2018-10-12:
@liuzfesp Any advice on what the size of the Task Stack should be? What should we expect if we set it too large?

IIM-Slide, 2018-10-23:
Thank you for the answer. Will follow up on this if the stack overflows again.

bricewge, 2018-10-25:
I don't have the same environment anymore and I can't manage to reproduce this bug...
Note that the default LWIP task stack size is 3072 since https://github.com/espressif/esp-idf/commit/61835faed1d62d060b9d81b6de115d9abfb1b91b.

Alvin1Zhang, 2018-11-19:
@bricewge Feel free to reopen this issue if it is reproduced. Thanks.

closed 2018-11-19 by Alvin1Zhang