espressif/esp-idf #2448: uart_set_line_inverse() *must* be called first in UART init sequence

FoxKeys, 2018-09-20:
This is documetation issue.
There *must* be notice in uart_set_line_inverse() description that it *must* be called before uart_set_pin() (and, maybe, other UART setup functions). Because otherwise you will get unexpected pulse/glitch on TX pin that can (in worst case) damage some hardware.
In my case - it just completely mess few first bytes sent.

Oscilloscope screenshot attached.
[image]

koobest, 2018-09-25:
Hi,@FoxKeys, Thanks  for your reminder, we will update the documentation soon.

FoxKeys, 2018-09-25:
One more related issue found.

uart_set_pin() must have at least notice about pull mode it set inside.
And, maybe? By default pull mode must be inverted for inverted line?
Maybe, add pull mode as required parameter into uart_set_pin()?

Because, same problem, even i call gpio_set_pull_mode() immedeately after uart_set_pin() - it is few microseconds while in my case internal pullup set by uart_set_pin() "conflicted" with external "hardware" pulldown. That is not good.

FoxKeys, 2018-09-29:
Yes, i know. But not GPIO16/17 (RX2/TX2 on my board).

In this case i'm talking about implicit (not documented)
gpio_set_pull_mode(rx_io_num, GPIO_PULLUP_ONLY);
inside of uart_set_pin() function

There is two problems here. First - it is not described in docs for uart_set_pin() (not all programmers, especially beginers, read ESP-IDF sources). And second - PULLUP is good for "normal" UART mode. But, not correct for inverted mode.

Alvin1Zhang, 2019-03-14:
@koobest Hi, could you help to take a relook at this issue? Thanks.

Alvin1Zhang, 2020-05-27:
@FoxKeys Thanks for reporting. Will close this ticket now, feel free to reopen if you still have the issue. Thanks.

closed 2020-05-27 by Alvin1Zhang