espressif/esp-idf #2449: C++ version too old - update to a gcc version that supports c++17 or at least c++14

owebeeone, 2018-09-21:
I would like to see a more recent version of GCC used so that I can make it work correctly with a new library I'm working on that uses at least C++14 features.

Notably, the Arduino distribution AVR compiler does support C++14.

But hey, if you're going to update the compiler, you may as well go all the way to c++17 since the ABI changes are probably not too big an issue (I don't know, just suspecting).

projectgus, 2018-09-24:
Can you be more specific about what support you need?

ESP-IDF uses xtensa-esp32-elf-gcc version 5.2.0, which [AFAIK supports all of C++14](https://gcc.gnu.org/projects/cxx-status.html#cxx14).

ESP-IDF build system builds IDF source code with `-std=gnu++11`, but for your own components you can modify this in component.mk, ie:

```
CFLAGS += std=c++14
```
([Docs](https://docs.espressif.com/projects/esp-idf/en/stable/api-guides/build-system.html#optional-component-specific-variables))

Or if you using CMake, modify CMakeLists.txt with

```
component_compile_options(-std=c++14)
```
([Docs](https://docs.espressif.com/projects/esp-idf/en/stable/api-guides/build-system-cmake.html#controlling-component-compilation)).

> you may as well go all the way to c++17

We plan to switch to GCC 8 (which supports all of C++17) for IDF V4.0.

GCC 8 support may be available as an option in v3.x as well, depending on how testing goes.

projectgus, 2018-09-25:
> GCC 8 support may be available as an option in v3.x as well, depending on how testing goes.

This happened sooner than I expected! https://esp32.com/viewtopic.php?f=10&t=7400

v1993, 2019-02-22:
So when will toolchan be released from beta? Is it still WIP?

Alvin1Zhang, 2019-03-22:
@igrr Hi, any updates? Thanks.

igrr, 2019-03-22:
Still WIP, but we plan to release an update to GCC 8 based toolchain soon, and switch to GCC 8 as the default toolchain in IDF v4.0.

igrr, 2019-04-12:
New preview release of the toolchain with GCC 8.2 is now available, see https://www.esp32.com/viewtopic.php?f=10&t=7400&p=31257#p41667.

Trass3r, 2019-04-12:
Nice please make sure LTO and objcopy is working, there were some problems in the original arm-gcc 8.2 release: https://gnu-mcu-eclipse.github.io/blog/2019/02/14/arm-none-eabi-gcc-v8-2-1-1-4-released/#changes

ppisljar, 2019-05-09:
how to enable c++ 17 ? i tried adding
```
COMPONENT_COMPILE_OPTIONS := -std=c++17
```
to my component.mk but it doesn't seem to work

Timmmm, 2019-05-12:
Ideally there would be a `CONFIG_CXX_STANDARD` in the `sdkconfig` but that doesn't exist yet. Instead I just manually edited `esp-idf/make/project.mk` (or `esp-idf/tools/cmake/idf_functions.cmake` for CMake) and changed `gnu++11` to `gnu++17`. Worked perfectly as far as I can tell.

v1993, 2019-05-12:
@Timmmm Why do you want to compile IDF itself in C++17 mode? You can set it using CMake commands for your project/component target I think. As for old build, I was using c++14 before by setting CXX flags without any problems, shouldn't be different for C++17.

PerMalmberg, 2019-05-12:
Is std::filesystem supposed to be available with this gcc 8?

Using my system default compiler (`gcc (Ubuntu 8.3.0-6ubuntu1) 8.3.0`) and `target_link_libraries(${PROJECT_NAME} stdc++fs)` everything compiles. 

Using `xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1-rc2) 8.2.0` I get

> xtensa-esp32-elf/bin/ld: cannot find -lstdc++fs

Timmmm, 2019-05-12:
> @Timmmm Why do you want to compile IDF itself in C++17 mode?

I don't. The `-std=gnu++11` flag is added in IDF for all source files. See the files that I mentioned.

> Also usage of global (non-target) variables like `CONFIG_CXX_STANDARD` is really deprecated in modern CMake and should be avoided.

I wasn't talking about CMake.

v1993, 2019-05-12:
@Timmmm I use this for C++14 (in `component.mk`):

```
CXXFLAGS += -std=gnu++14
```

Try something like this for C++17?

Timmmm, 2019-05-12:
@v1993 That didn't work when I tried it for some reason, and won't that give you *both* flags in the command line?

Also, it won't matter in practice in this case but it's worth pointing out that linking different versions of C++ is technically not allowed, so you really should compile everything as C++17.

v1993, 2019-05-12:
@Timmmm It worth pointing out that it **is** safe in GCC: https://stackoverflow.com/q/46746878/5697743

igrr, 2019-05-12:
@PerMalmberg At the moment, std::filesystem code is not available. It depends on several functions which aren't provided in our newlib builds (ref GLIBCXX_CHECK_FILESYSTEM_DEPS in libstdc++ source). It will likely be a larger task to provide some stubs for all the filesystem functions...

PerMalmberg, 2019-05-12:
@igrr Ok, thanks for the quick reply.

Perhaps make a note of what major things that are not included in the post of the forum?

closed 2019-06-10 by igrr

rustyx, 2021-07-12:
The correct way to specify required C++ standard in the project's `CMakeLists.txt`:

```
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
```

MarkJeronimus, 2021-09-08:
Is there a way to specify the language level in the Eclipse IDE also? Right now, IDF is building my code just fine, but every c++17 feature in the source code shows up as an error in the IDE and marks the file with a red box.

franz-ms-muc, 2022-03-31:
Tested, works well. 

Example: 

[https://github.com/Meisterschulen-am-Ostbahnhof-Munchen/nvs_shadow_cxx/blob/feature/Map_of_any/Map_of_any/main/main.cpp](https://github.com/Meisterschulen-am-Ostbahnhof-Munchen/nvs_shadow_cxx/blob/feature/Map_of_any/Map_of_any/main/main.cpp)

using std::any to make a map of std::map\<std::string, std::any>

franz-ms-muc, 2022-03-31:
component_compile_options(-std=c++17)

MarkJeronimus, 2022-04-04:
I tried adding `component_compile_options` to my project-wide CMakeLists.txt, which didn't work (Unknown CMake command error). I changed it to `add_compile_options`, which complains that it's not valid for C++. I tried changing it to `add_cxx_compile_options` which gave the same error. I tried adding it to the list of `set(CMAKE_CXX_FLAGS "...")` which didn't work (Compiles fine, Eclipse still has red squigglies under C++17 features.

MarkJeronimus, 2023-07-31:
> set(CMAKE_CXX_STANDARD 17)
> set(CMAKE_CXX_STANDARD_REQUIRED ON)

This suddenly doesn't work anymore, and now I can't compile any of the projects at work.
I didn't update ESP-IDF, so does someone know if it could have been caused by updates of other dependencies on Fedora? The compiler it detects after a clean build is GNU 8.4.0

GreyCon, 2023-11-26:
Hi, I also wish to use C++17, and have this same issue. However the fix of commenting out:
#    list(APPEND cxx_compile_options "-std=gnu++11")
doesn't work for me, as this line doesn't exist in my version of build.cmake. (esp-idf-5.1)
Anyone else managed to get C++17 to work?

igrr, 2023-11-27:
@GreyCon  IDF v5.1 already uses C++23 by default, so no extra changes should be necessary: https://docs.espressif.com/projects/esp-idf/en/v5.1.2/esp32/api-guides/cplusplus.html#c-language-standard

jdoubleu, 2023-11-27:
I've used the recommended CMake way before:
```cpp
target_compile_features(${COMPONENT_LIB} PRIVATE cxx_std_23)
```

However, linking against some (esp) libraries overwrote my preferred language standard. Best you check your error output, the compiler invocation command should be displayed there. If you see multiple `-std=` that might be an issue. Usually the last one is taken.