espressif/esp-idf #2450: Bluedroid library storage

ntremble-sst, 2018-09-21:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): []
- Core (if using chip or module): [ESP32-Wrover]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.):
   d7079b6 (v3.2-dev-755-gd7079b6-dirty)
- Development Env:		[Eclipse]
- Operating System:		[Ubuntu]
- Power Supply:         [external 3.3V]

## Problem Description
We are developing an embedded telemetry application based on the ESP-WROVER module. One of the features we would like to include is the ability to scan for Bluetooth beacons. I have this scanning working so I know it is possible, but the Bluedroid library is huge, so including this feature would mean we would lose the ability to perform over the air updates.

We are wondering if it is possible to resolve this by one of the following means:
1. Can Espressif provide a cut down version of the Bluedroid library that includes only the features required for beacon scanning? For this to be useful the library would need to be reduced from the current 316kB to less than 100kB.

2. Can Espressif provide a version of the ESP-WROVER with more flash? We are using the 4MB part, so we would need at least 8MB and perhaps 16MB to ensure we have no similar issues in the future.

If neither of these is possible, can Espressif suggest another approach that may solve our problem?

I contemplated using a common partition to hold just the Bluedroid library, and using the MMU to map that partition into addressable space when running from either of the OTA partitions, but that presents issues of its own e.g.: we could not readily OTA the Bluedroid library itself.

### Expected Behavior
We require OTA_0 and OTA_1 partitions large enough to hold our application now and in the future while having full speed run time access to the Bluedroid library. Ideally the factory partition would be the same size as the OTA partitions.

### Actual Behavior
We currently have a 1024KB factory partition and a pair of 1400KB OTA partitions plus a few smaller partitions for holding configuration and production test data. These partitions are too small to hold our application when the Bluedroid library is linked; the partitions must be at least 1500KB to hold that code.
And we still have quite a lot of functionality to add, so we will end up needing something closer to 2000KB per partition for a total exceeding 6MB.

### Steps to repropduce

1. Link our application with Bluedroid.
2. Attempt to flash it and see it doesn't fit.

// It helps if you attach a picture of your setup/wiring here.
[image]

### Code to reproduce this issue

N/A

## Debug Logs

N/A

## Other items if possible

- [[sdkconfig.zip](https://github.com/espressif/esp-idf/files/2403951/sdkconfig.zip)] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.) 

- custom partition table
  nvs,          data,    nvs,         0x9000,    16K
  otadata,      data,    ota,         0xd000,    8K
  phy_init,     data,    phy,         0xf000,    4K
  factory,      app,     factory,     0x10000,   1024K
  ota_0,        app,     ota_0,       0x110000,	 1400K
  ota_1,        app,     ota_1,       0x270000,  1400K
  index,        data,    0x40,        0x3ce000,  4K
  production,   data,    0x41,        0x3cf000,  4K
  config1,      data,    0x42,        0x3d0000,  64K
  config2,      data,    0x43,        0x3e0000,  64K
  coredump,     data,    coredump,    0x3f0000,  64K

Thank you

TianHao-Espressif, 2018-09-21:
@ntremble-sst , you seem to care about the instruction size of the project, especially bluedroid, right?
If think the bluedroid it too large, I suggest you write a tiny host with some HCI command end HCI event handler to just work on BLE scanning . You can take a look at the esp-idf/example/bluetooth/ble_adv/, this example work without bluedroid. 

As far as now, bluedroid cannot provide a function of BLE scanning only. But when do Bluedroid optimize in the future, we will consider your suggestion.

chegewara, 2018-09-21:
Official wrover module with 8 and 16 MB flash:
https://esp32.com/viewtopic.php?f=10&t=6912

ntremble-sst, 2018-09-23:
@TianHao-Espressif Yes, it is the code size that is causing us issues. I will look at the example you mention, thank you

ntremble-sst, 2018-09-23:
@chegewara Thank you, we had not realised these modules had been released. Now to see if we can actually get any...

Alvin1Zhang, 2018-10-16:
@ntremble-sst Could you help share if any updates for this issue? Thanks.

ntremble-sst, 2018-10-16:
We have obtained samples of the 16MB part, so that will resolve the immediate issue.
I have not at this stage investigated the HCI approach, but will do so when time permits.
Thank you.

closed 2018-10-16 by ntremble-sst