espressif/esp-idf #2452: rmt_write_items - duplication / mutexes

dirkx, 2018-09-21:
There seems to be some duplication in rmt_write_items() that escapes the mux scrutiny

1) Rather than call rmt_set_tx_wrap_en() -   RMT.apb_conf.mem_tx_wrap_en = 1; is set unprotected. (807)

2) both rmt_set_tx_thr_intr_en() and         RMT.tx_lim_ch[channel].limit = item_sub_len (line 807) is set; the latter unprotected. And this one then too calls rmt_set_tx_wrap_en() (see above.

3) rmt_set_tx_loop_mode() is not called - tx_conti_mode is set unprotected.

Etc. May be nice to clean up (I am seeing RMT occasionally glitch when used intensively).

dirkx, 2018-09-22:
See https://github.com/espressif/esp-idf/pull/2456

koobest, 2018-10-24:
hi, @dirkx, thanks for your code, we will update the c file and it will be merged soon.

closed 2018-11-29 by igrr