espressif/esp-idf #2453: Mounting Error

nickypeh, 2018-09-21:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit): [v3]
- Core (if using chip or module): [ESP32-Wrover-I]
- IDF version (latest commit): 
- Development Env:		[Make]
- Operating System:		[MacOS]
- Power Supply:         [USB]

## Problem Description

Able to use mkspiffs to flash into the spiffs but **could not mount** spiffs to read the files.  it will try to format the storage if it is turned on the config.

### Expected Behavior
mkspiffs the file and upload it to the flash.
Mount and read the flashed file content from esp32.

### Actual Behavior
successfully flash the files using mkspiffs but **could not** mount the spiffs and read the file. 

### Steps to reproduce

1. mkspiffs using the command below : 
- mkspiffs -c $(pwd)/spiffs_image/ -b 4096 -p 256 -s 0x100000 spiffs.bin
- esptool.py --chip esp32 --port /dev/tty.usbserial-141101 --baud 115200 write_flash -z 0x180000 spiffs.bin
2. partition table csv
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 1M,
storage,  data, spiffs,  0x180000,0xF0000, 
3. make flash monitor the codes as shown below to mount the spiffs and to read the file after that.

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main(void)
{
    ESP_LOGI(TAG, "Initializing SPIFFS");
    
    esp_vfs_spiffs_conf_t conf = {
      .base_path = "/spiffs",
      .partition_label = NULL,
      .max_files = 5,
      .format_if_mount_failed = false
    };
    
    // Use settings defined above to initialize and mount SPIFFS filesystem.
    // Note: esp_vfs_spiffs_register is an all-in-one convenience function.
    esp_err_t ret = esp_vfs_spiffs_register(&conf);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount or format filesystem");
        } else if (ret == ESP_ERR_NOT_FOUND) {
            ESP_LOGE(TAG, "Failed to find SPIFFS partition");
        } else {
            ESP_LOGE(TAG, "Failed to initialize SPIFFS (%s)", esp_err_to_name(ret));
        }
        return;
    }
    
    size_t total = 0, used = 0;
    ret = esp_spiffs_info(NULL, &total, &used);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "Failed to get SPIFFS partition information (%s)", esp_err_to_name(ret));
    } else {
        ESP_LOGI(TAG, "Partition size: total: %d, used: %d", total, used);
    }
}

```

## Debug Logs

```
I (0) cpu_start: App cpu up.
I (106) heap_init: Initializing. RAM available for dynamic allocation:
I (113) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (119) heap_init: At 3FFB3170 len 0002CE90 (179 KiB): DRAM
I (125) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (132) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (138) heap_init: At 40088F30 len 000170D0 (92 KiB): IRAM
I (144) cpu_start: Pro cpu start user code
I (162) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (164) example: Initializing SPIFFS
E (168) SPIFFS: mount failed, -10025
E (171) example: Failed to mount or format filesystem
```

nickypeh, 2018-09-21:
1. sdkconfig 

```
# SPIFFS Configuration
CONFIG_SPIFFS_MAX_PARTITIONS=3
# SPIFFS Cache Configuration
CONFIG_SPIFFS_CACHE=y
CONFIG_SPIFFS_CACHE_WR=y
CONFIG_SPIFFS_CACHE_STATS=
CONFIG_SPIFFS_PAGE_CHECK=y
CONFIG_SPIFFS_GC_MAX_RUNS=10
CONFIG_SPIFFS_GC_STATS=
CONFIG_SPIFFS_PAGE_SIZE=256
CONFIG_SPIFFS_OBJ_NAME_LEN=32
CONFIG_SPIFFS_USE_MAGIC=y
CONFIG_SPIFFS_USE_MAGIC_LENGTH=y
CONFIG_SPIFFS_META_LENGTH=4
CONFIG_SPIFFS_USE_MTIME=y
CONFIG_SPIFFS_DBG=
CONFIG_SPIFFS_API_DBG=
CONFIG_SPIFFS_GC_DBG=
CONFIG_SPIFFS_CACHE_DBG=
CONFIG_SPIFFS_CHECK_DBG=
CONFIG_SPIFFS_TEST_VISUALISATION=
```

2. mkspiffs --version 
```
mkspiffs ver. 0.2.3-2-g4eae236
Build configuration name: custom
SPIFFS ver. 0.3.7-5-gf5e26c4
Extra build flags: -DSPIFFS_OBJ_META_LEN=4
SPIFFS configuration:
  SPIFFS_OBJ_NAME_LEN: 32
  SPIFFS_OBJ_META_LEN: 4
  SPIFFS_USE_MAGIC: 1
  SPIFFS_USE_MAGIC_LENGTH: 1
  SPIFFS_ALIGNED_OBJECT_INDEX_TABLES: 0
```

igrr, 2018-09-22:
The rest of configuration looks consistent, so once you correct the size argument in mkspiffs call, it should work.

nickypeh, 2018-09-22:
yea it worked!! thanks a lot!

closed 2018-09-22 by igrr

PRitu21, 2019-05-23:
In my case, it is not working. My partition table is below:

# Name | Type | SubType | Offset | Size | Flags
nvs | data | nvs | 0x9000 | 16K |  
otadata | data | ota | 0xd000 | 8K |  
phy_init | data | phy | 0xf000 | 4K |  
factory | app | factory | 0x10000 | 1M |  
storage | data | spiffs |   | 16K |  
ota_0 | 0 | ota_0 |   | 1200K |  
ota_1 | 0 | ota_1 |   | 1200K |  

Below are commands, I ran after git cloning mkspiffs - git clone https://github.com/igrr/mkspiffs
./mkspiffs -c ./data -b 4096 -p 256 -s 0x4000 spiffs.bin
python esptool.py --port COM3 -b 115200 write_flash 0x110000 ./spiffs.bin
 and I am getting formatting error when I try to initialize spiffs.

Could you please help me understand this?
For your information, I am able to initialize, read, and write using spiffs example in esp-idf. It creates problem when I am using spiff in my application. I dont understand what is lacking in my application.

Looking forward for your response.

Thanks
Ritu.

renzbagaporo, 2019-05-23:
Hi @PRitu21,

Have you also confirmed the consistency between the sdkconfig values and mkspiffs --version output? Mind sharing your config here as @nickypeh did?

On another note, if you're willing to use a newer version of ESP-IDF, we introduced a new tool called `spiffsgen.py` which should be easier to use. See https://github.com/espressif/esp-idf/issues/3470.

PRitu21, 2019-05-23:
Hi @renzbagaporo 

Please refer below as required.
./mkspiffs --version
mkspiffs ver. 0.2.3-6-g983970e
Build configuration name: generic
SPIFFS ver. 0.3.7-5-gf5e26c4
Extra build flags: (none)
SPIFFS configuration:
  SPIFFS_OBJ_NAME_LEN: 32
  SPIFFS_OBJ_META_LEN: 0
  SPIFFS_USE_MAGIC: 1
  SPIFFS_USE_MAGIC_LENGTH: 1
  SPIFFS_ALIGNED_OBJECT_INDEX_TABLES: 0

 cat sdkconfig | grep SPIFFS
# SPIFFS Configuration
CONFIG_SPIFFS_MAX_PARTITIONS=3
# SPIFFS Cache Configuration
CONFIG_SPIFFS_CACHE=y
CONFIG_SPIFFS_CACHE_WR=y
CONFIG_SPIFFS_CACHE_STATS=
CONFIG_SPIFFS_PAGE_CHECK=y
CONFIG_SPIFFS_GC_MAX_RUNS=10
CONFIG_SPIFFS_GC_STATS=
CONFIG_SPIFFS_PAGE_SIZE=256
CONFIG_SPIFFS_OBJ_NAME_LEN=32
CONFIG_SPIFFS_USE_MAGIC=y
CONFIG_SPIFFS_USE_MAGIC_LENGTH=y
CONFIG_SPIFFS_META_LENGTH=0
CONFIG_SPIFFS_DBG=
CONFIG_SPIFFS_API_DBG=
CONFIG_SPIFFS_GC_DBG=
CONFIG_SPIFFS_CACHE_DBG=
CONFIG_SPIFFS_CHECK_DBG=
CONFIG_SPIFFS_TEST_VISUALISATION=

$ git describe
v4.0-dev-217-g5b2247df2

Is there a new version to this?? I can use spiffgen.py if there is a proper use case and documentation.

Thanks.
Ritu

PRitu21, 2019-05-24:
Hi,

I changed the size of spiff partition from 16K to 1M and changed fread() to fgets() and I could read the spiff file.
Is there any limitation of size for using spiff in esp-idf?? If yes, Can I switch between factory app and ota_0 when upgrading software over ota ?? My board has 4MB flash, and I have to include all this in my project.

Please suggest the better way.

Waiting for your early response.

Thanks
Ritu.

renzbagaporo, 2019-05-27:
Hi @PRitu21,

Using the `storage/spiffs` example, reducing the size to 16K of the spiffs partition still succeeds, so I guess in your case it should work (you can try it this out yourself). The thing that's easier with spiffsgen.py is that it is integrated into the build system, automatically creating the image during build and optionally flashing it for you when you invoke `idf.py flash`. You can find more about it on here: https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/spiffs.html#spiffsgen-py

Renz