espressif/esp-idf #2454: lwip_accept failed after a while with error 23 (file table full)

Reiner1210, 2018-09-21:
After a while lwip_accept faul with error code 23.
I know that it had worked some month ago, so I think it's a new bug.

zhangyanjiaoesp, 2018-10-16:
@Reiner1210 Can you show me the IDF version and commit ID when it works well and when the failure occured? Thanks.

Reiner1210, 2018-10-16:
Now I use v3.2-dev-1133-g4d50427e8 and it works. The older version was ~ one week before

zhangyanjiaoesp, 2018-10-16:
@Reiner1210 you means the latest IDF works well, but the version a week ago has this problem?

Reiner1210, 2018-10-16:
I do NOT mean the latest - I mean v3.2-dev-1133-g4d50427e8.
The version I tried before v3.2-dev-1133-g4d50427e8 has not worked.
So look at the release date for v3.2-dev-1133-g4d50427e8 and then go back some (1 or 2) weeks.
Why are so many (untested) releases for the esp32 ??

zhangyanjiaoesp, 2018-10-17:
I got it. This problem has been solved in commit: https://github.com/espressif/esp-idf/commit/f2d3efd875dc01cf5e34ec7e79d2a11cf9b8f94b. You can close this issue.

closed 2018-10-17 by Reiner1210

projectgus, 2018-10-17:
> Why are so many (untested) releases for the esp32 ??

Something to take note of, these aren't ESP-IDF releases - commits on the ESP-IDF master branch have passed automated testing but they haven't had any additional manual testing done (beyond developer testing).

You can read more about IDF versions here:
https://docs.espressif.com/projects/esp-idf/en/latest/versions.html#which-version-should-i-start-with

and find stable releases here:
https://github.com/espressif/esp-idf/releases