espressif/esp-idf #2455: spicommon_bus_initialize_io() does not accept SCLK on input pins when slave

bu5hm4nn, 2018-09-22:
## Environment

- Development Kit:     [WiPy 3.0]
- Core:                        [ESP32D0WDQ6 revision 1]
- IDF version               [Don't know how to get from platformio]
- Development Env:   [PlatformIO]
- Operating System:  [Any]
- Power Supply:         [USB]

## Problem Description

When using SPI as slave, the function `spicommon_bus_initialize_io` does not allow
SCLK to be on an input-only pin.

### Expected Behavior

When the devices is configured as SPI Slave, then SCLK is an input-only signal and should be allowed on input-only pins.

### Actual Behavior

Assert is thrown:
```
E (1554) spi: spicommon_bus_initialize_io(136): sclk not valid
```

### Steps to repropduce

Try to initialize the SPI as slave with SCK on an input-only pin with the code. No wiring needed.

### Code to reproduce this issue

```cpp
// Configuration for the SPI bus
spi_bus_config_t buscfg =
{
    .mosi_io_num = GPIO_NUM_36,
    .miso_io_num = GPIO_NUM_22,
    .sclk_io_num = GPIO_NUM_37,
    .quadwp_io_num = -1,
    .quadhd_io_num = -1,
    .max_transfer_sz = SPI_SLAVE_BUFFER_SIZE,
    .flags = 0
};

// Configuration for the SPI slave interface
spi_slave_interface_config_t slvcfg = 
{
    .spics_io_num =     GPIO_NUM_38,
    .flags =            0,
    .queue_size =       1,
    .mode =             0,
    .post_setup_cb =    post_setup_callback,
    .post_trans_cb =    post_transport_callback
};

// Initialize SPI slave interface
assert( spi_slave_initialize( VSPI_HOST, &buscfg, &slvcfg, 1 ) == ESP_OK);
```
## Debug Logs

```
E (1554) spi: spicommon_bus_initialize_io(136): sclk not valid
```

##  Proposed solution:

Check the SCLK pin in the same way as the MOSI pin:

```cpp
//the SCLK should be output capble in master mode, or in DIO/QIO mode.
bool sclk_output = (flags&SPICOMMON_BUSFLAG_MASTER)!=0 || flags&SPICOMMON_BUSFLAG_DUAL;
```

and further down replace with this:

```cpp
if (bus_config->sclk_io_num>=0) {
    temp_flag |= SPICOMMON_BUSFLAG_SCLK;
    if( sclk_output ) {
        SPI_CHECK(GPIO_IS_VALID_OUTPUT_GPIO(bus_config->sclk_io_num), "sclk not valid", ESP_ERR_INVALID_ARG);
    } else {
        SPI_CHECK(GPIO_IS_VALID_GPIO(bus_config->sclk_io_num), "sclk not valid", ESP_ERR_INVALID_ARG);
    }
    if (bus_config->sclk_io_num != spi_periph_signal[host].spiclk_iomux_pin) use_iomux = false;
} else {
    SPI_CHECK((flags&SPICOMMON_BUSFLAG_SCLK)==0, "sclk pin required.", ESP_ERR_INVALID_ARG);
}
```

trancefreak77, 2019-02-03:
When will this issue be fixed? I'm currently working on a schematic design where I have to know if CLK in slave mode can be assigned to an input only pin.

ginkgm, 2019-02-11:
Hi @trancefreak77 .

It is now under internal review. And may appear on master soon.

trancefreak77, 2019-02-12:
Thanks for the update.

closed 2019-02-22 by igrr