espressif/esp-idf #2459: Automatic light sleep crashes randomly

erkki, 2018-09-22:
## Environment

- Development Kit:      [none]
- Core (if using chip or module): [ESP-WROOM32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 3276a13
- Development Env:		[Make]
- Operating System:		[MacOS]
- Power Supply:         [USB|Battery]

## Problem Description

Auto light sleep seems to crash randomly, both in a minimal test case and the wifi power save example. Tested on 2 different esp-wroom-32 kits on both usb power and battery power. In the minimal sample case, reducing vTaskDelay to 30ms seems to work fine while anything up from 40ms and larger constantly crashes, with differing error results, either WDT reboots or memory load/store errors.

Testing with release/v3.1, the wifi power save example works. Git bisect got to commit ff81dc16e066bb5f76d3650144bffc23ccd33208 as the first bad commit.

### Code to reproduce this issue
The wifi power save example or the following:
```cpp
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <esp_pm.h>
#include <esp_log.h>

void app_main() {
        esp_pm_config_esp32_t pm_config = {
            .max_freq_mhz = 160,
            .min_freq_mhz = 40,
            .light_sleep_enable = true
    };
    ESP_ERROR_CHECK(esp_pm_configure(&pm_config));
    while(1) {
        vTaskDelay(40 / portTICK_PERIOD_MS);
        ESP_LOGI("auto_sleep", "Wakeup");
    }
}
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
I (16) boot: ESP-IDF v3.2-dev-1055-g3276a13 2nd stage bootloader
I (23) boot: compile time 16:52:57
I (27) boot: Enabling RNG early entropy source...
I (32) boot: SPI Speed      : 40MHz
I (36) boot: SPI Mode       : DIO
I (40) boot: SPI Flash Size : 4MB
I (44) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (55) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (63) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (70) boot:  2 factory          factory app      00 00 00010000 00100000
I (78) boot: End of partition table
I (82) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x08830 ( 34864) map
I (103) esp_image: segment 1: paddr=0x00018858 vaddr=0x3ffb0000 size=0x021f0 (  8688) load
I (107) esp_image: segment 2: paddr=0x0001aa50 vaddr=0x3ffb21f0 size=0x00000 (     0) load
I (111) esp_image: segment 3: paddr=0x0001aa58 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at auto_sleep/esp-idf/components/freertos/xtensa_vectors.S:1685

I (120) esp_image: segment 4: paddr=0x0001ae60 vaddr=0x40080400 size=0x051b0 ( 20912) load
I (137) esp_image: segment 5: paddr=0x00020018 vaddr=0x400d0018 size=0x13118 ( 78104) map
0x400d0018: _flash_cache_start at ??:?

I (165) esp_image: segment 6: paddr=0x00033138 vaddr=0x400855b0 size=0x04ee4 ( 20196) load
0x400855b0: xTaskGetIdleTaskHandleForCPU at auto_sleep/esp-idf/components/freertos/tasks.c:4617

I (174) esp_image: segment 7: paddr=0x00038024 vaddr=0x400c0000 size=0x00064 (   100) load
I (174) esp_image: segment 8: paddr=0x00038090 vaddr=0x50000000 size=0x00000 (     0) load
I (187) boot: Loaded app from partition at offset 0x10000
I (187) boot: Disabling RNG early entropy source...
[... 199 lines cut ...]
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400870dd  PS      : 0x00060a33  A0      : 0x8008625f  A1      : 0x3ffb05a0  
0x400870dd: uxListRemove at auto_sleep/esp-idf/components/freertos/list.c:214

A2      : 0x3ffb5238  A3      : 0x00000000  A4      : 0x80082290  A5      : 0x3ffb0590  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x00000005  A9      : 0x00000001  
A10     : 0x3ffb54c4  A11     : 0x00000000  A12     : 0x00060a21  A13     : 0x00000001  
A14     : 0x00060a21  A15     : 0x00060023  SAR     : 0x00000006  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000009  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
Core 0 was running in ISR context:
EPC1    : 0x400870dd  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000
0x400870dd: uxListRemove at auto_sleep/esp-idf/components/freertos/list.c:214

Backtrace: 0x400870dd:0x3ffb05a0 0x4008625c:0x3ffb05c0 0x400848b7:0x3ffb05e0 0x40086dd1:0x3ffb0600 0x40082df2:0x3ffb0610 0x4000bfed:0x00000000
0x400870dd: uxListRemove at auto_sleep/esp-idf/components/freertos/list.c:214

0x4008625c: xTaskIncrementTick at auto_sleep/esp-idf/components/freertos/tasks.c:4617

0x400848b7: xPortSysTickHandler at auto_sleep/esp-idf/components/freertos/port.c:401

0x40086dd1: _frxt_timer_int at auto_sleep/esp-idf/components/freertos/portasm.S:303

0x40082df2: _xt_lowint1 at auto_sleep/esp-idf/components/freertos/xtensa_vectors.S:1105
```

dereks, 2018-09-27:
@igrr @mahavirj   Hi! I think I have a reliable reproduction of this issue here on my bench.

I will look through the bisected commit to see if I can find a patch. If you have any new information about the root cause, or if you have a bug fix branch I could test for you, please let me know. Thanks!

dereks, 2018-09-27:
I tried to revert that commit one small bit at a time, to find the error, but I could not get the `abort()` error to disappear. I think the problem may be a more complex hardware issue than just an extra `#if` or a missing ` asm("waiti 0")` that I saw in this patch.

My next step would be reverting the entire commit properly, with git (instead of piecemeal). And maybe confirm the git bisect result. I'll hold off on that for a bit, to see where you guys are at.

mahavirj, 2018-09-28:
@erkki 
Thanks for reporting this. Fix will be available soon in ESP-IDF.

@dereks 
If you are really blocked on this then you may try patch at [here](https://gist.github.com/mahavirj/31922fe41710a52622e6c89f37ce4f80)

dereks, 2018-09-29:
@mahavirj   

**It works!**  The linked patch above resolved my `abort()` crash.

My project is doing a deployment, and we need this patch. This is a vote to merge this to master and do a tagged minor release (3.1.1 maybe?). Which is to say, this seems worthy of a bugfix release...?

Thank you!

igrr, 2018-09-29:
@dereks This fix will be in master branch soon. My understanding is that the regression is only in master branch, not in v3.1. Please let me know if you see this in v3.1 as well.

erkki, 2018-09-29:
Okay I spoke too soon, the test snippet is still broken for me on master(11b444b) + patch

erkki, 2018-09-29:
```
I (11) boot: ESP-IDF v3.2-dev-1175-g11b444b 2nd stage bootloader
I (11) boot: compile time 14:13:30
I (12) boot: Enabling RNG early entropy source...
I (16) boot: SPI Speed      : 40MHz
I (20) boot: SPI Mode       : DIO
I (24) boot: SPI Flash Size : 4MB
I (28) boot: Partition Table:
I (32) boot: ## Label            Usage          Type ST Offset   Length
I (39) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (47) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (54) boot:  2 factory          factory app      00 00 00010000 00100000
I (62) boot: End of partition table
I (66) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x08a48 ( 35400) map
I (87) esp_image: segment 1: paddr=0x00018a70 vaddr=0x3ffb0000 size=0x021f0 (  8688) load
I (91) esp_image: segment 2: paddr=0x0001ac68 vaddr=0x3ffb21f0 size=0x00000 (     0) load
I (95) esp_image: segment 3: paddr=0x0001ac70 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/erkkieilonen/projects/learning/esp32/ttgo/auto_sleep/esp-idf/components/freertos/xtensa_vectors.S:1685

I (104) esp_image: segment 4: paddr=0x0001b078 vaddr=0x40080400 size=0x04f98 ( 20376) load
I (121) esp_image: segment 5: paddr=0x00020018 vaddr=0x400d0018 size=0x13154 ( 78164) map
0x400d0018: _flash_cache_start at ??:?

I (149) esp_image: segment 6: paddr=0x00033174 vaddr=0x40085398 size=0x05024 ( 20516) load
0x40085398: prvInitialiseNewTask at /Users/erkkieilonen/projects/learning/esp32/ttgo/auto_sleep/esp-idf/components/freertos/tasks.c:4617

I (158) esp_image: segment 7: paddr=0x000381a0 vaddr=0x400c0000 size=0x00064 (   100) load
I (158) esp_image: segment 8: paddr=0x0003820c vaddr=0x50000000 size=0x00000 (     0) load
I (171) boot: Loaded app from partition at offset 0x10000
I (171) boot: Disabling RNG early entropy source...
I (176) cpu_start: Pro cpu up.
I (180) cpu_start: Starting app cpu, entry point is 0x40080ff4
0x40080ff4: call_start_cpu1 at /Users/erkkieilonen/projects/learning/esp32/ttgo/auto_sleep/esp-idf/components/esp32/cpu_start.c:234

I (180) cpu_start: App cpu up.
D (190) memory_layout: Checking 7 reserved memory ranges:
D (195) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (202) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb32b8
D (208) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (215) memory_layout: Reserved memory range 0x3ffe4000 - 0x3ffe4350
D (221) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (227) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at /Users/erkkieilonen/projects/learning/esp32/ttgo/auto_sleep/esp-idf/components/freertos/xtensa_vectors.S:1685

D (234) memory_layout: Reserved memory range 0x40080000 - 0x4008a3bc
0x40080000: _WindowOverflow4 at /Users/erkkieilonen/projects/learning/esp32/ttgo/auto_sleep/esp-idf/components/freertos/xtensa_vectors.S:1685

D (240) memory_layout: Building list of available memory regions:
D (247) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
D (253) memory_layout: Available memory region 0x3ffb32b8 - 0x3ffb8000
D (260) memory_layout: Available memory region 0x3ffb8000 - 0x3ffc0000
D (266) memory_layout: Available memory region 0x3ffc0000 - 0x3ffc2000
D (273) memory_layout: Available memory region 0x3ffc2000 - 0x3ffc4000
D (279) memory_layout: Available memory region 0x3ffc4000 - 0x3ffc6000
D (286) memory_layout: Available memory region 0x3ffc6000 - 0x3ffc8000
D (293) memory_layout: Available memory region 0x3ffc8000 - 0x3ffca000
D (299) memory_layout: Available memory region 0x3ffca000 - 0x3ffcc000
D (306) memory_layout: Available memory region 0x3ffcc000 - 0x3ffce000
D (312) memory_layout: Available memory region 0x3ffce000 - 0x3ffd0000
D (319) memory_layout: Available memory region 0x3ffd0000 - 0x3ffd2000
D (326) memory_layout: Available memory region 0x3ffd2000 - 0x3ffd4000
D (332) memory_layout: Available memory region 0x3ffd4000 - 0x3ffd6000
D (339) memory_layout: Available memory region 0x3ffd6000 - 0x3ffd8000
D (345) memory_layout: Available memory region 0x3ffd8000 - 0x3ffda000
D (352) memory_layout: Available memory region 0x3ffda000 - 0x3ffdc000
D (359) memory_layout: Available memory region 0x3ffdc000 - 0x3ffde000
D (365) memory_layout: Available memory region 0x3ffde000 - 0x3ffe0000
D (372) memory_layout: Available memory region 0x3ffe0440 - 0x3ffe4000
D (378) memory_layout: Available memory region 0x3ffe4350 - 0x3ffe8000
D (385) memory_layout: Available memory region 0x3ffe8000 - 0x3fff0000
[... 32 lines cut ...]
D (86) heap_init: New heap initialised at 0x3ffe0440
D (96) heap_init: New heap initialised at 0x3ffe4350
D (96) intr_alloc: Connected src 16 to int 12 (cpu 0)
I (106) pm_esp32: Frequency switching config: CPU_MAX: 160, APB_MAX: 80, APB_MIN: 40, Light sleep: ENABLED
D (116) sleep: RTC_PERIPH: AUTO(OFF), RTC_SLOW_MEM: AUTO(OFF), RTC_FAST_MEM: ON
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400856ed  PS      : 0x00060033  A0      : 0x40086d87  A1      : 0x3ffb5aa0  
0x400856ed: vTaskSwitchContext at /Users/erkkieilonen/projects/learning/esp32/ttgo/auto_sleep/esp-idf/components/freertos/tasks.c:4617

0x40086d87: _frxt_dispatch at /Users/erkkieilonen/projects/learning/esp32/ttgo/auto_sleep/esp-idf/components/freertos/portasm.S:407

A2      : 0x00050023  A3      : 0x00000001  A4      : 0x3ffb2c38  A5      : 0x3ffb2c38  
[... cut ...]
EXCVADDR: 0x00000049  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400856ed:0x3ffb5aa0 0x40086d84:0x3ffb5ac0 0x40086d3a:0x0000000c
0x400856ed: vTaskSwitchContext at /Users/erkkieilonen/projects/learning/esp32/ttgo/auto_sleep/esp-idf/components/freertos/tasks.c:4617

0x40086d84: _frxt_dispatch at /Users/erkkieilonen/projects/learning/esp32/ttgo/auto_sleep/esp-idf/components/freertos/portasm.S:406

0x40086d3a: _frxt_int_exit at /Users/erkkieilonen/projects/learning/esp32/ttgo/auto_sleep/esp-idf/components/freertos/portasm.S:206
```

mahavirj, 2018-09-29:
@erkki 
It could be possible that there are multiple independent issues here. Could you please help to share your application (archive including sdkconfig) that is reproducing this crash (on both master and release/v3.1)?

erkki, 2018-09-29:
> @erkki
> It could be possible that there are multiple independent issues here. Could you please help to share your application (archive including sdkconfig) that is reproducing this crash (on both master and release/v3.1)?

Sure thing, here's a minimal example (needs to be modified a bit for v3.1 for the pm parts).
[auto_sleep.zip](https://github.com/espressif/esp-idf/files/2430867/auto_sleep.zip)

mahavirj, 2018-09-29:
@erkki 
Probably with log verbosity changed to debug level, IDLE task stack size also needs to be increased. Could you please try changing `Component Config -> FreeRTOS-> Idle Task Stack size` to say 3184? Let me know if that fixes the problem or not.

erkki, 2018-09-29:
Okay I'm a dummy, the new minimal example I generated used a blank sdkconfig with power management turned off. The patched version works now.

dereks, 2018-10-01:
@igrr 
> Please let me know if you see this in v3.1 as well.

I will try to test it sometime this week. The app I inherited was using an old master HEAD (pre 3.1), so I just did a git pull.

I know it's bad to use master in a deployment, but you guys have added an amazing number of bug fixes and feature improvements in just the last few days -- I think we will want those in our deployed image. We have OTA available so there is some risk mitigation (not counting a bricking bug).

But of course tagged official releases are what one should usually use.