espressif/esp-idf #2462: DLE negotiation fails if LL_LENGTH_RSP has 2120+ TX/RX time

jakkra, 2018-09-25:
## Environment

- Development Kit:      [own]
- Core (if using chip or module): [ESP32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    93b588a0cf7dca19ea02c08e3773ecf8e908ed45
- Development Env:		[Make]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description

DLE negotiation fails when LL_LENGTH_RSP from other device contains 2MB PHY parameters (tx/rx time). I initialy thought this was a Nordic NRF issue, however it seems as it's an error in the ESP32 Bluetooth controller. 

### Expected Behavior
DLE negotiation successful with 251 TX octets and 2120 TX/RX time.

According to the BT spec. the ESP32 should start using those parameters.
### Actual Behavior
LL_REJECT_IND sent and connection times out.

### Steps to repropduce

1. I'm sending a LL_LENGTH_REQ (251 octets and 2120ms time) to nRF52840 from an ESP32.
2. The nRF52840 responds with LL_LENGTH_RSP (251, 2704ms) (2704 is 2MB PHY parameters)
3. ESP32 now sends LL_REJECT_IND and DLE negotiation fails and the link times out.

More information can be found in the NRF issue I created [https://devzone.nordicsemi.com/f/nordic-q-a/38738/dle-negotiation-ll_length_request-failure](https://devzone.nordicsemi.com/f/nordic-q-a/38738/dle-negotiation-ll_length_request-failure)

In that issue the air traces can be found also and comments from Nordic about the issue, and what exactly is wrong.

jakkra, 2018-09-26:
Apparently this was a NRF Stack issue after all.

closed 2018-09-26 by jakkra

Emill, 2019-05-15:
I think this case should be re-opened since it is prohibited to respond with LL_REJECT_IND to an LL_LENGTH_RSP packet.

A packet which contains invalid parameters should be replied to using LL_UNKNOWN_RSP. LL_REJECT_IND is only used in the specification where it explicitly states that this packet should be used, for example by the encryption and connection parameter negotiation procedures.

> If an LL Control PDU is received that is not supported or reserved for future use, the Link Layer shall respond with an LL_UNKNOWN_RSP PDU. The UnknownType field of the LL_UNKNOWN_RSP PDU shall be set to the value of the Opcode in the received PDU.

> If an LL Control PDU is received with the wrong length or with invalid CtrData fields, the Link Layer may continue with the relevant Link Layer procedure with an implementation-specific interpretation of the data (e.g., if the PDU is too long it can ignore the extra data while if a field is out of range it can use the nearest permitted value). If it does not continue the procedure, it shall respond with an LL_UNKNOWN_RSP PDU or, if the relevant procedure allows it, an LL_REJECT_IND or LL_REJECT_EXT_IND PDU. The UnknownType field of the LL_UNKNOWN_RSP PDU or the RejectOpcode field of the LL_REJECT_EXT_IND PDU shall be set to the value of the Opcode in the received PDU.

The correct behaviour would be to follow the corrected 4.2 version, which removes the requirements of the maximum value of the parameters:

[image]

jakkra, 2019-05-16:
@Emill Feel free to open this issue. NRF have fixed it in the latest Softdevice, but I agree ESP32 should also fix it. I don't know, maybe ESP already fixed it, I have not tested in a while. Feel free to do it.

mpvader, 2021-01-29:
Hi @jakkra / @Emill or any others:

@Emill wrote:
> I think this case should be re-opened since it is prohibited to respond with LL_REJECT_IND to an LL_LENGTH_RSP packet.

We have been testing an issue between an ESP32 and a few phones, amongst which a Huawei P30 Lite running Android 10. In our situation the phone is the central, and the GlobalLink is the peripheral.

Below a few screenshots of the airtrace we made. What it looks like is that the phone sends a few parameters that according to some 4.2 BLE spec version are allowed, and some other 4.2 version not. And that the GlobalLink then responds with with wrong message (LL_REJECT_EXT_IND). While it should respond with something else.

Any ideas from someone? Is this fixed upstream? Or perhaps at least a pointer to where this handled in the espressif-code? (hoping its in the open source part).

Our device with the ESP32 on a Pycom G01 running ESP-IDF 3.3.1. Obviously I know thats not the latest version, we're looking into upgrading it - even if only for a test - but thats not a very small effort unfortunately.

[image]

[image]

Emill, 2021-01-29:
As mentioned previously, this is a spec violation by the ESP32. The relevant code is in the closed-source link layer. Since nobody reopened this issue, I think you should open a new issue.

jakkra, 2021-01-29:
@Emill @mpvader Create a new issue and link to this one, Espressif don't like when you reopen old issues. Also often faster support on new ones.

mpvader, 2021-01-29:
Thanks both, noted - we'll make a new issue - thanks!

mpvader, 2021-01-29:
ps. before opening a new issue, we'll first make 100% sure its the same on the latest ESP-IDF.

mpvader, 2021-02-16:
Hi, for the records: we tested this with ESP-IDF 4.1, and the problem is solved. case closed for us.

Emill, 2021-02-16:
Did you analyze the packets transferred over the air, to verify that the same protocol packets are sent up to the point where it originally did send LL_REJECT_EXT_IND, and that that packet has now been changed to something else?

mpvader, 2021-02-16:
Yes. Airtrace.

LL_REJECT_EXT_IND is gone and replaced by LL_LENGTH_RSP. All looking good

mpvader, 2021-02-16:
For completeness: we had different issues on different phones, but all leading back to this problem being the root cause. And besides that the airtrace now looks good wrt this pcket length negotiation, also all problems are gone on the various phone.