espressif/esp-idf #2463: Where is ETH_CLOCK_GPIO0_OUT = 1 clock mode

smacyas, 2018-09-25:
## Environment

- Development Kit:      ESP32-WROOM32 module
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Core ESP-WROOM32
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:		Arduino IDE
- Operating System:		Windows
- Power Supply:         external 3.3V

## Problem Description

In the newest idf there is no ETH_CLOCK_GPIO0_OUT (1) clock mode, but in all my devices i uses it. What is the reason?

suda-morris, 2018-09-25:
The ETH_CLOCK_GPIO0_OUT mode usually cannot get stable in our test. Typically the GMAC should work under 50MHz clock, at which situation, GPIO0 will output around 100MHz clock. But GPIO16 and GPIO17 will process the clock from PLLA and produce a much more stable 50MHz clock to these two GPIOs. Actually GPIO0 is not controlled by GMAC, it just routes the clock from PLLA. So if you have no choice but use GPIO0 output, you may have a try to add external divider outside GPIO0.

smacyas, 2018-09-25:
I see, thanks.

closed 2018-09-25 by smacyas