espressif/esp-idf #2466: Problem with including esp_fill_random()

ussserrr, 2018-09-25:
I use default Make environment and get `error: implicit declaration of function 'esp_fill_random' [-Werror=implicit-function-declaration]` at the same time when `esp_random()` works just fine.

Steps to reproduce:
  1. Clone `esp-idf-template` project
  2. Add `esp_random()` at the bottom of `main.c`, everything compiles OK
  3. Add `esp_fill_random()` and get `implicit declaration` error

ESP-IDF version: 
```
$ git rev-parse --short HEAD
22da5f6de
```
(installed about a week ago)

fri-sch, 2018-11-09:
esp-idf 22da5f6de is v3.1-rc2 and this version simply does not have esp_fill_random(). It was added afterwards. See https://github.com/espressif/esp-idf/blob/v3.1-rc2/components/esp32/include/esp_system.h.

ussserrr, 2018-12-09:
OK, maybe I was referring to 'latest' documentation instead of 'stable'

closed 2018-12-09 by ussserrr