espressif/esp-idf #2467: LE Connection fail rate 10-15% after enabling BT Modem sleep

jakkra, 2018-09-25:
## Environment

- Development Kit:      [none]
- Core (if using chip or module): [ESP32]
- IDF version: 
    93b588a0cf7dca19ea02c08e3773ecf8e908ed45
- Development Env:		[Make]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description
After enabling Bluetooth Modem sleep we see a lot more LE Connect failures between ESP32's. 10-15% of LE create connection fails. We are not using bluedroid. I have attached the HCI log which you can see all sent/received HCI commands.

**The scenario looks like this:**
ESP32 Central: (Full log here [acl_connect_fail.zip](https://github.com/espressif/esp-idf/files/2415469/acl_connect_fail.zip))
```
Received LE Create Connection (status: pending)
Received LE Connection Complete (status: success)
Received LE Disconnect Complete  (status: 0x3e Failed to be established)
```

ESP32 Peripheral:
`EMPTY - No Connection Events at all received from the Bluetooth controller`

So the question here is first, why do we get a Disconnect event stright after connect with error code "Failed to be established" when we just before got an event which said Connection Complete "success"?

Second, why does the other ESP32 (Peripheral) not get a LE Connection Complete event at all?

All connection parameters can be seen in the HCI LOG I have attached.
### Expected Behavior
I know that it's normal for BLE to every once in a while get a quick disconnect after a connection. However 10-15% seems a little much? Also the HCI Events does not seem logical i.e wierd error code + No HCI event on the peripheral. Is this expected?

### Actual Behavior
Less failed LE Create Connection.
### Steps to repropduce
I have not tried this with example applications yet, but our application does not do much other than below.

1. Normal LE Connect from ESP32 central to ESP32 Peripheral (Both with BT Modem sleep enabled)
2. Repeat until my scenario explained above is reached. 10+ tries required. Sometimes more are needed.

## Debug Logs
[ESP32 Central acl_connect_fail.zip](https://github.com/espressif/esp-idf/files/2415469/acl_connect_fail.zip)

## Other items if possible

Dual mode BLE/Classic
The application and Bluetooth stack is running on core 1
BT Controller is running on 0

mywang-espressif, 2018-09-29:
@jakkra,

Let me try to answer your first two questions. Here I think there is a difference in the definition between "connection creation" and "connection establishment", according to Bluetooth Core Spec(v5.0).

As to the connection setup procedure, the spec says,
```
After the initiator sends a CONNECTION_IND_PDU on the primary advertising channel ..., the
Link Layer is in the Connecion State in the Master Role. The master shall reset the Link
Layer connection supervision timer T_LLconnSupervision. The Link Layer shall notify the
Host that the connection has been created...
```

```
The Link Layer enters the Connection State when an initiator sends a CONNECT_REQ PDU on
the primary advertising channel from an initiator ...

After entering the Connection State, the connection is considered to be created. The
connection is not considered to be established at this point. A connection is only
considered to be established once a data channel packet has been received from the
peer device.
```

```
If the Link Layer connection supervision timer reaches 6 * connInterval before the connection
is established, the connection shall be considered lost. This enables fast termination of
connections that fail to establish
```

As to the HCI event LE_Connection_Complete, the core spec says:
```
The LE Connection Complete event indicates to both of the Hosts forming the
connection that a new connection has been created. Upon the creation of the
connection a Connection_Handle shall be assigned by the Controller, and
passed to the Host in this event. If the connection creation fails this event shall
be provided to the Host that had issued the LE_Create_Connection command.

This event indicates to the Host which issued a LE_Create_Connection
command and received a Command Status event if the connection creation
failed or was successful.
```

Coming back to the HCI error code "CONNECTION FAILED TO BE ESTABLISHED(0x3E):
```
The Connection Failed to be Established error code indicates that the LL intiated
a connection but the connection has failed to established.
```

My explanation for this case is, for the initiator, the connection
succeeds in "connection creation" but fails in "connection establishment", therefore
results in a Disconnect event with error code (0x3e) following the LE_Connection_Complete
event. The process may be that the CONN_REQ PDU has been sent by the initiator, but the
advertiser fails to receive the packet, and then link supervision timer on the initiator
expires. This also explains why the periperhal side does not get an LE_Connection_Complete
event at all.

mywang-espressif, 2018-09-29:
I am sorry that I confounded CONNECT_IND PDU and CONNECT_REQ PDU in last comment. The former is defined in core spec v5.0 and the latter in v4.2. But the two has similar functions in connection creation procedure.

jakkra, 2018-09-29:
@mywang-espressif Hi, thanks alot for the information! That explains the behaviour. 

However this is something new after enabling BT Modem sleep. We have not seen this behaviour before, and disabling modem sleep it goes away. So it seems as this has to do with modem sleep. 10-15% failed connections now, compared to maybe ~1% without modem sleep, which is a big difference.

Is this normal?

jakkra, 2018-09-29:
Thank you, I will do more testing. We have a fairly noisy environment at the office. But I can easily see a difference in our test system of the number of failed tests due to connection establishment failing (scenario described above) when I have modem sleep enabled or not. 

I will do some different tests, and come back with more information. And try to narrow down what more specific scenario causes the failures.

Thanks for the info before, I can now see that this is probably the case, which is what you described[image]

jakkra, 2018-10-05:
Hi @mywang-espressif I have now been running tests in a radio shielded box, to remove any external radio environment factors. I still see the exact same problem. I did each test 1000 times with different setups, see table below:

|                                          | Shielded Box Modem Sleep enabled | Shielded Box Modem Sleep disabled |
|------------------------------------------|-------------------------|--------------------------|
| Reset both every 10 connects             | 10%                     | 0%                       |
| Reset both between between every connect | 12%                     | 0%                       |
| No reset between connects                | 12%                     | 0%                       |

If I do the exact same tests against other BLE manufacter peripherals, I see no connection problem. Only when I connect to ESP32 peripheral.

mywang-espressif, 2018-10-09:
@jakkra, many thanks for your test results and detailed information. I am doing more tests and studying  more thoroughly on this issue.

mywang-espressif, 2018-10-09:
@jakkra, I would like to ask two questions on your test:
1. Can you tell me the LE  connection interval and slave latency for the connection?
2. I wonder what the results would be if you only enable bluetooth modem sleep on one side(i.e. master or slave) and disable modem sleep on another(i.e. slave or master). I think the result can be helpful in my analysis on this issue.

My previous test was carried on without a Shielded Box and no reset is performed between connects. So according to your test results there seems to be little influence of "reset" on the connection fail rate, right? I will carry on more tests under different conditions.

jakkra, 2018-10-09:
Hi @mywang-espressif, thanks for looking into this issue more. The fix for #2385 did not do any difference for this issue. 

Yes, from my testing it did not make much difference if I reset between every connect.

1. If you open the file I attached to the issue in Wireshark you can see the exact parameters on the HCI commands sent/received. Here are create connection and connection complete. For more info check the attached .pklg file.
LE Create Connection
[image]
Le Connection Complete
[image]

2. I will do that and come back with the results.

mywang-espressif, 2018-10-10:
@jakkra, thanks for the information. I have been testing the connection rate with the same connection parameters and set up the test environment in a shielded box, but cannot reproduce the problem. I will come back later after reviewing the whole procedure.

mywang-espressif, 2018-10-10:
@jakkra, I would also ask the advertising parameters, i.e. parameters in command HCI_LE_Set_Advertising_Parameters on the peripheral side

jakkra, 2018-10-10:
@mywang-espressif That is unfortunate that you have not been able to eproduce it. Ill try to think of what else could be different. I also attached the sdkconfig.
[image]
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/2464703/sdkconfig.zip)

mywang-espressif, 2018-10-11:
Hi @jakkra, I have compared the sdkconfig file and there are no important differences, I think.
I have tested again with some of the Kconfig options modified as in the attachment, and use the same advertising parameters and connection parameters, but still cannot reproduce the issue.

I don't know where the difference is, perhaps the it is due to the behaviour of bluetooth host stack?
In your test, do you use other functionalities of ESP32 except bluetooth, for example, WiFi, other peripherals, etc. I think to keep the application simple enough is helpful in our analysis. In my test with bluedroid on ESP32, I don't rather limited modules, and bluedroid only uses the NVS for link key storage. Use of NVS may be a guess as the cause of this issue but I tried both bluedroid and btstack with NVS disabled, but tests results show no difference here.
Could you please review your logic and check the functional modules used?
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/2468330/sdkconfig.zip)

jakkra, 2018-10-11:
Hi  @mywang-espressif Thanks for trying to reproduce this issue. We are not using any other peripherals than Bluetooth. We do use NVS fairly much, but when connections are established, not much else is going on. I will try to strip away parts of our application, and see if the issue goes away at some point.

I will try to come back with more info.

Alvin1Zhang, 2018-10-16:
@jakkra Could you help share if any more updates for this issue? Thanks.

jakkra, 2018-11-30:
@Alvin1Zhang Hi, this issue is hard to debug. Could it help if you provided me with a bt lib with some clever printouts that could be relevant? And in that way get some more information about what could be the root cause (except BT modem sleep).

jakkra, 2018-11-30:
I tested without enabling NVS, but no difference, about 10% of the connections still fails.

chegewara, 2018-12-03:
Sorry, i dont want to interrupt this topic because i dont know as much about bluetooth as you guys, and most likely my test procedure is wrong.
To make c++ (arduino) BLE class better i am trying to follow all issues about ble and if i can i am trying to perform tests to see how good is my work.
 Now i am trying to do test to reproduce this issue (i dont have no-echo chamber) this way:
- 1 esp32 as peripheral wit 40ms advertising interval,
- 1 esp32 as central with scan interval 1533 and window 97,
- scan,
- connect,
- perform write to peripheral characteristic, 
- peripheral responds with notification on each write request,
- disconnect,
- wait 2000 ms,
- repeat.
This is result:
```
W (4054904) my_gattc_event_handler: total disconnect count: 1366, disconnect reason 0x16: 1355, disconnect reason0x3e: 11
```

jakkra, 2018-12-09:
@chegewara Hi, thanks for testing. Could you also try with modem sleep enabled on peripheral, as that is the case where we are having issues? Having modem sleep enbled/disabled on central does not make a difference, see https://github.com/espressif/esp-idf/issues/2467#issuecomment-428141561

chegewara, 2018-12-09:
There is for sure something with modem sleep, because my earlier tests with modem sleep of on both, peripheral and central, gave me result about 0.2% disconnection with result 0x3e (about 36 on over 17400 tries). With modem on on central its about 10 times more often. 
I have no time right now to perform test with peripheral sleep modem, but i will try to do it.

mywang-espressif, 2019-01-02:
@chegewara, @jakkra, we have two bugfix recently which are related to bluetooth modem sleep. I hope they can help resolve the increasing disconnection rate issue, I am not sure because I still did not  reproduce the issue. The two bugfixes are:
1.) 6cb37ecc5
2.) 561021e7f
You can use the latest commit on master branch. The bluetooth modem sleep procedure code is somehow modified in these bug fixes.

jakkra, 2019-01-14:
Hi @mywang-espressif we have upgraded to ESP-IDF 3.2 beta which includes those two patches and now we see big improvements when connecting to EPS32 peripheral with modem sleep. I think we can finally close this issue 🥇

closed 2019-02-04 by jakkra