espressif/esp-idf #2468: promiscuous : wifi_pkt_rx_ctrl_t field "timestamp" contains same timestamp multiple times

Dpunkt, 2018-09-25:
## Environment

- Development Kit:     ESP32-DevKitC
- Kit version v1
- Core (if using chip or module): ESP-WROOM32
- IDF version  3276a1316
- Development Env:		Make
- Operating System:		Windows
- Power Supply:         USB

## Problem Description

When using the wifi_promiscuous_callback to receive beacons from one access point, I sometimes receive the same timestamp in the wifi_pkt_rx_ctrl_t  field in multiple packets.

According to the source code there are known issues with the accuracy:
 unsigned timestamp:32;        /**< timestamp. The local time when this packet is received. It is precise only if modem sleep or light sleep is not enabled. unit: microsecond */

But since i disabled powermanagement in make menuconfig I thought this should not be an issue.
Are there any more undescribed reasons for the timestamp to be off? Is powersave somehow enabled even when disabled in configuration?

### Expected Behavior
Timestamps in wifi_pkt_rx_ctrl_t should increase in the same steps (after unit conversation) as the timestamp field in a beacon. (Timestamp field in beacon and timing of beacons verified using wireshark)
### Actual Behavior
ESP32 writes same timestamp-wifi_pkt_rx_ctrl_t to some packets. Starting with a later packet the time is correct again

### Steps to repropduce

1. Flash Code using example snippets.
2. compare beacon and wifi_pkt_rx_ctrl_t -timestamp

### Code to reproduce this issue
```
uint32_t oldTimestamp;
void wifi_promiscuous_cb(void *buf, wifi_promiscuous_pkt_type_t type)
{

	wifi_promiscuous_pkt_t *rxBuffer = (wifi_promiscuous_pkt_t *)buf;
			uint64_t zeit=0;
			memcpy(&zeit,&rxBuffer->payload[24],8);
			printf("BTS: %llu ",zeit);

	uint32_t newTimestamp= rxBuffer->rx_ctrl.timestamp;
	uint32_t diffTimestamp;
	if (newTimestamp<oldTimestamp)
		diffTimestamp=(0xFFFFFFFF-oldTimestamp)+ newTimestamp;
	else
		diffTimestamp=newTimestamp-oldTimestamp;
	printf(" TDfiff %d %d %d",diffTimestamp/1000,oldTimestamp/1000,newTimestamp/1000);
	if ((diffTimestamp)< TIME_THREASHOLD)
	oldTimestamp=newTimestamp;
}
main:
	ESP_ERROR_CHECK(esp_wifi_set_promiscuous_rx_cb(wifi_promiscuous_cb));
	ESP_ERROR_CHECK(esp_wifi_set_promiscuous(true));
```

## Debug Logs

```
Beacon Timestamp        Diff oldTS   newTS 
BTS: 791675019371 TDfiff 102 132902 133005
BTS: 791685254440 TDfiff 0 133005 133005
BTS: 791695501183 TDfiff 204 133005 133209
BTS: 791705741486 TDfiff 102 133209 133312
BTS: 7917159854ae TDfiff 0 133312 133312
BTS: 79172622379d TDfiff 204 133312 133516
BTS: 79173645262e TDfiff 102 133516 133619
BTS: 791746693638 TDfiff 102 133619 133721
BTS: 7917569467bb TDfiff 0 133721 133721
BTS: 791767176454 TDfiff 0 133721 133721
BTS: 791777418468 TDfiff 0 133721 133721
BTS: 79178766228e TDfiff 0 133721 133721
BTS: 79179789071b TDfiff 512 133721 134233
BTS: 79180813091d TDfiff 0 134233 134233
BTS: 791818374743 TDfiff 0 134233 134233
BTS: 79182861594f TDfiff 307 134233 134541
BTS: 79183885906e TDfiff 0 134541 134541
BTS: 791849088606 TDfiff 0 134541 134541
BTS: 791859329812 TDfiff 307 134541 134848
```
Expected was a change in newTimestamp, never the same timestamp multiple times.

igrr, 2018-09-25:
Can you please add a call to esp_wifi_set_ps to disable modem sleep? Power management (setting in menuconfig) is independent from modem sleep.

Dpunkt, 2018-09-25:
I now tried with ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_NONE)), now the timestamps are accurate.
Unfortunally i didn't found any documentation about this.
It would be a good thing to explan this a little more in the comment for timestamp.

closed 2018-09-25 by Dpunkt