espressif/esp-idf #2469: [question] ble adv_direct_ind advertising settings

chegewara, 2018-09-25:
@Weijian-Espressif Is `ADV_TYPE_DIRECT_IND_HIGH` supported and if it is how we can set target device address? According to bluetooth specs in this case payload should have only advertiser address and target device address. I cant find in any advertising settings to set target address, so its my question. Only thing i can think of is to setup address with raw data advertising. Thanks in advance.

Weijian-Espressif, 2018-09-30:
@chegewara 
you can set ble adv_direct_ind advertising  by esp_ble_gap_start_advertising(esp_ble_adv_params_t *adv_params)

typedef struct {
    uint16_t                adv_int_min;        /*!< Minimum advertising interval for
                                                  undirected and low duty cycle directed advertising.
                                                  Range: 0x0020 to 0x4000 Default: N = 0x0800 (1.28 second)
                                                  Time = N * 0.625 msec Time Range: 20 ms to 10.24 sec */
    uint16_t                adv_int_max;        /*!< Maximum advertising interval for
                                                  undirected and low duty cycle directed advertising.
                                                  Range: 0x0020 to 0x4000 Default: N = 0x0800 (1.28 second)
                                                  Time = N * 0.625 msec Time Range: 20 ms to 10.24 sec Advertising max interval */
    esp_ble_adv_type_t      adv_type;           /*!< Advertising type */
    esp_ble_addr_type_t     own_addr_type;      /*!< Owner bluetooth device address type */
    esp_bd_addr_t           peer_addr;          /*!< Peer device bluetooth device address */
    esp_ble_addr_type_t     peer_addr_type;     /*!< Peer device bluetooth device address type */
    esp_ble_adv_channel_t   channel_map;        /*!< Advertising channel map */
    esp_ble_adv_filter_t    adv_filter_policy;  /*!< Advertising filter policy */
} esp_ble_adv_params_t;

chegewara, 2018-09-30:
I must be blind, ive been reading thru this structure few days (adv intervals) and i missed that.

Thanks.

chegewara, 2018-10-01:
@Weijian-Espressif Sorry, its me again. When i set `esp_ble_addr_type_t peer_addr_type` to any RPA type (BLE_ADDR_TYPE_RPA_RANDOM or BLE_ADDR_TYPE_RPA_PUBLIC) then im getting this error:
`E (2733) BT_HCI: hci write adv params error 0x30`.

Also with adv_type = ADV_TYPE_DIRECT_IND_LOW im getting this error:
```
E (2767) BT_BTM: state requested not supported: 5
E (2772) BT_APPL: bta_dm_ble_set_adv_params_all(), fail to start ble adv.
```

Weijian-Espressif, 2018-10-11:
hi @chegewara  i test  ble adv_direct_ind advertising  in latest ESP-IDF master, it works ok , please update to the latest IDF and try, thanks.

chegewara, 2018-10-11:
@Weijian-Espressif 
Thanks, you are great.

chegewara, 2018-10-11:
Thanks, i will try to check it. Since it works i have better question. Is it possible with this stack to create more than one advertising at the same time? With my smartphone i can (as for now i tried up to 6). Im asking, because i would like for example to have one advertising type ADV_TYPE_IND and few ADV_TYPE_DIRECT_IND to few whitelisted peer devices.

Weijian-Espressif, 2018-10-15:
@chegewara Sorry, ESP32 do not support multiple advertising now.

closed 2018-10-15 by Alvin1Zhang

chegewara, 2019-02-09:
Hi, its me again. I am doing something wrong because `ADV_TYPE_DIRECT_IND_HIGH` is not working to me. Here is example code i am using:
```
/*
   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "esp_bt.h"

#include "esp_gap_ble_api.h"
#include "esp_gatts_api.h"
#include "esp_bt_defs.h"
#include "esp_bt_main.h"
#include "example_ble_sec_gatts_demo.h"

#define GATTS_TABLE_TAG "SEC_GATTS_DEMO"

#define HEART_PROFILE_NUM                         1
#define HEART_PROFILE_APP_IDX                     0
#define ESP_HEART_RATE_APP_ID                     0x55
#define EXCAMPLE_DEVICE_NAME                      "ESP_BLE_SECURITY"
#define HEART_RATE_SVC_INST_ID                    0

#define GATTS_DEMO_CHAR_VAL_LEN_MAX               0x40

#define ADV_CONFIG_FLAG                           (1 << 0)
#define SCAN_RSP_CONFIG_FLAG                      (1 << 1)

static uint8_t adv_config_done = 0;

static uint16_t heart_rate_handle_table[HRS_IDX_NB];

static uint8_t test_manufacturer[3]={'E', 'S', 'P'};

static uint8_t sec_service_uuid[16] = {
    /* LSB <--------------------------------------------------------------------------------> MSB */
    //first uuid, 16bit, [12],[13] is the value
    0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0x18, 0x0D, 0x00, 0x00,
};

// config adv data
static esp_ble_adv_data_t heart_rate_adv_config = {
    .set_scan_rsp = false,
    .include_txpower = true,
    .include_name = false,
    .min_interval = 0x0006, //slave connection min interval, Time = min_interval * 1.25 msec
    .max_interval = 0x0010, //slave connection max interval, Time = max_interval * 1.25 msec
    .appearance = 0x00,
    .manufacturer_len = 0, //TEST_MANUFACTURER_DATA_LEN,
    .p_manufacturer_data =  NULL, //&test_manufacturer[0],
    .service_data_len = 0,
    .p_service_data = NULL,
    .service_uuid_len = sizeof(sec_service_uuid),
    .p_service_uuid = sec_service_uuid,
    .flag = (ESP_BLE_ADV_FLAG_GEN_DISC | ESP_BLE_ADV_FLAG_BREDR_NOT_SPT),
};
// config scan response data
static esp_ble_adv_data_t heart_rate_scan_rsp_config = {
    .set_scan_rsp = true,
    .include_name = true,
    .include_txpower = true,
[... 539 lines cut ...]
    uint8_t oob_support = ESP_BLE_OOB_DISABLE;
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_STATIC_PASSKEY, &passkey, sizeof(uint32_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_AUTHEN_REQ_MODE, &auth_req, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_IOCAP_MODE, &iocap, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_MAX_KEY_SIZE, &key_size, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_ONLY_ACCEPT_SPECIFIED_SEC_AUTH, &auth_option, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_OOB_SUPPORT, &oob_support, sizeof(uint8_t));
    /* If your BLE device act as a Slave, the init_key means you hope which types of key of the master should distribut to you,
    and the response key means which key you can distribut to the Master;
    If your BLE device act as a master, the response key means you hope which types of key of the slave should distribut to you,
    and the init key means which key you can distribut to the slave. */
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_INIT_KEY, &init_key, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_RSP_KEY, &rsp_key, sizeof(uint8_t));

    /* Just show how to clear all the bonded devices
     * Delay 30s, clear all the bonded devices
     *
     * vTaskDelay(30000 / portTICK_PERIOD_MS);
     * remove_all_bonded_devices();
     */
}
```

With `ADV_TYPE_DIRECT_IND_HIGH` i cant scan esp32 neither connect it, im getting `GATT ERROR 0x85`. 
Of course its possible that android still have issues with `GATT ERROR 0x85`.