espressif/esp-idf #2472: NVS Partition Generator Utility problem

karawin, 2018-09-25:
## Environment

- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
   3276a1316
- Development Env:		[other]
- Operating System:		[Windows]
- Power Supply:         [USB|external 5V]

## Problem Description
I created a nvs partition hardware of 0x1000
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,16K,
otadata,data,ota,0xd000,8K,
phy_init,data,phy,0xf000,4K,
ota_0,app,ota_0,0x10000,1792K,
ota_1,app,ota_1,0x1d0000,1792K,
device,64,0,0x390000,4K,
stations,65,0,0x391000,64K,
device1,66,0,0x3a1000,4K,
hardware,data,nvs,0x3a2000,4K,
*******************************************************************************
I created a csv file with 
key,type,encoding,value
gpio_space,namespace,,
PIN_NUM_MISO,data,u8,19
PIN_NUM_MOSI,data,u8,23
PIN_NUM_CLK,data,u8,18

When i want to create a bin i have this error:
$ python C:/msys32/home/jp/esp/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py standard.csv standard.bin 4KB
Size parameter is insufficient.

With 8KB partition (0x2000) it works.
Why a 4KB size is not accepted?

negativekelvin, 2018-09-25:
Nvs needs at least 2 sectors

karawin, 2018-09-27:
Not documented?

closed 2018-11-28 by igrr