espressif/esp-idf #2473: eea49d250 breaks wifi in STA (wpa2) mode, reverting to 66fd7e9cc restores functionality

Jegeva, 2018-09-25:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

## Environment

- Development Kit:     none
- Core (if using chip or module): ESP-WROOM32
- IDF version                              eea49d250
- Development Env:		Make
- Operating System:		Debian
- Power Supply:         [USB+Battery]

## Problem Description

//Detailed problem description goes here.

when pulling eea49d250, the connection stay to :
I (138) wifi: wifi driver task: 3ffdc7bc, prio:23, stack:3584, core=0
I (228) wifi: wifi firmware version: e27cdcf
I (238) wifi: config NVS flash: enabled
I (238) wifi: config nano formating: disabled
I (238) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (248) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (278) wifi: Init dynamic tx buffer num: 32
I (278) wifi: Init data frame dynamic rx buffer num: 32
I (278) wifi: Init management frame dynamic rx buffer num: 32
I (278) wifi: Init static rx buffer size: 1600
I (288) wifi: Init static rx buffer num: 10
I (288) wifi: Init dynamic rx buffer num: 32
I (358) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 0
I (358) wifi: mode : sta (24:0a:c4:0c:a1:b8)
sta start

and hangs.....
....forever.......

same code when i revert to 66fd7e9cc

I (136) wifi: wifi driver task: 3ffdc86c, prio:23, stack:3584, core=0
I (226) wifi: wifi firmware version: 7e3a527
I (226) wifi: config NVS flash: enabled
I (236) wifi: config nano formating: disabled
I (236) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (246) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (276) wifi: Init dynamic tx buffer num: 32
I (276) wifi: Init data frame dynamic rx buffer num: 32
I (276) wifi: Init management frame dynamic rx buffer num: 32
I (276) wifi: Init static rx buffer size: 1600
I (286) wifi: Init static rx buffer num: 10
I (286) wifi: Init dynamic rx buffer num: 32
I (356) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 0
I (356) wifi: mode : sta (24:0a:c4:0c:a1:b8)
sta start
I (1686) wifi: n:11 0, o:1 0, ap:255 255, sta:11 0, prof:1
I (2336) wifi: state: init -> auth (b0)
I (2346) wifi: state: auth -> assoc (0)
I (2366) wifi: state: assoc -> run (10)
I (2506) wifi: connected with telenet-xxxxxxx, channel 11
I (2506) wifi: pm start, type: 1

### Expected Behavior
connect to wifi
### Actual Behavior
doesn't
### Steps to repropduce
use the wifi demo code

negativekelvin, 2018-09-26:
Did you swap the commit IDs?  eea49d2 is older.

Jegeva, 2018-09-26:
@negativekelvin indeed i think you are right, 66fd7e9cc breaks, eea49d250 restore 

git reflog
eea49d250 (HEAD -> master) HEAD@{0}: reset: moving to eea49d250
66fd7e9cc (origin/master, origin/HEAD) HEAD@{1}: pull: Fast-forward
eea49d250 (HEAD -> master) HEAD@{2}: pull: Fast-forward
be81d2c16 HEAD@{3}: pull: Fast-forward
a3c43251b HEAD@{4}: pull: Fast-forward
0e501e5ed HEAD@{5}: clone: from https://github.com/espressif/esp-idf.git

negativekelvin, 2018-09-26:
Are you using examples/wifi/getting_started/station? This is my output

```
I (174) wifi station: ESP_WIFI_MODE_STA
I (174) wifi: wifi driver task: 3ffc1110, prio:23, stack:3584, core=0
I (174) wifi: wifi firmware version: e27cdcf
I (174) wifi: config NVS flash: enabled
I (174) wifi: config nano formating: disabled
I (184) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (194) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (224) wifi: Init dynamic tx buffer num: 32
I (224) wifi: Init data frame dynamic rx buffer num: 32
I (224) wifi: Init management frame dynamic rx buffer num: 32
I (224) wifi: Init static rx buffer size: 1600
I (234) wifi: Init static rx buffer num: 10
I (234) wifi: Init dynamic rx buffer num: 32
I (324) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 0
I (324) wifi: mode : sta (18:fe:34:6a:92:f2)
I (324) wifi station: wifi_init_sta finished.
I (324) wifi station: connect to ap SSID:****** password:******
I (1534) wifi: n:10 0, o:1 0, ap:255 255, sta:10 0, prof:1
I (2514) wifi: state: init -> auth (b0)
I (2524) wifi: state: auth -> assoc (0)
I (2524) wifi: state: assoc -> run (10)
I (3534) wifi: connected with ******, channel 10
I (3604) wifi: pm start, type: 1
```

Alvin1Zhang, 2018-10-16:
@Jegeva Could you help share if any updates for this issue? Thanks.

liuzfesp, 2019-07-24:
@Jegeva Could you help share if any updates for this issue? Thanks.

Alvin1Zhang, 2019-08-21:
@Jegeva  Thanks for reporting the issue, will close this due to short of feedback, feel free to reopen if you still have the issue. Thanks.

closed 2019-08-21 by Alvin1Zhang