espressif/esp-idf #2474: Toolchain Refresh - Command "python setup.py egg_info" failed with error code 1

Si03, 2018-09-26:
## Environment
- IDF version: 66fd7e9cc
- Operating System:	Windows 10

## Problem Description
I have been working fine on the toolchain for smonths, but noticed a build warning appear about 'crosstool' after a recent pull and decided to do a fresh install by deleting the existing c:\msys32 folder and installing from scratch, following the instructions [here](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#get-started-get-esp-idf) for installing the master branch environment on Windows. during the install process I get an error when running the command:  
`python -m pip install --user -r $IDF_PATH/requirements.txt`
Error: 
`Command "python setup.py egg_info" failed with error code 1 in :/users/simon/appdata/local/temp/pip-build-8acuzf/cffi/`

subsequent builds produce the build output folders but fail to produce any app.bin file, for instance, a make on hello_world:
```
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-80-g6c4433a5
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.

```
... ...
```
esptool.py v2.5.0
The following Python requirements are not satisfied:
future>=0.16.0
cryptography
Please run "C:/msys32/mingw32/bin/python.exe -m pip install --user -r C:/msys32/home/Simon/esp/esp-idf/requirements.txt" for resolving the issue.
make: *** [C:/msys32/home/Simon/esp/esp-idf/make/project.mk:454: check_python_dependencies] Error 1

```
running `C:/msys32/mingw32/bin/python.exe -m pip install --user -r C:/msys32/home/Simon/esp/esp-idf/requirements.txt` does not resolve the issue and errors repeat as above.

Any help appreciated.
cheers,
Si.

negativekelvin, 2018-09-26:
Maybe you got an old link? This tells you how to update just the toolchain https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup-scratch.html#alternative-setup-just-download-a-toolchain

igrr, 2018-09-26:
Could you please attach the output you get when you run `C:/msys32/mingw32/bin/python.exe -m pip install --user -r C:/msys32/home/Simon/esp/esp-idf/requirements.txt`?

projectgus, 2018-09-26:
I think the solution to this may be the same as given here:
https://esp32.com/viewtopic.php?f=13&t=7419#p31342

Can you please try that command and see if you're able to continue?

We'll update the MSYS2 package ASAP.

Si03, 2018-09-26:
Thanks for the replies.
@projectgus : This is the correct solution. All working fine after running the 'pacman...'  command followed by the 'python...pip...' command.
@igrr : for completeness, here is the output -[output.txt](https://github.com/espressif/esp-idf/files/2418932/output.txt)
The warning relating to 'crosstool-ng-1.22.0-80-g6c4433a5' as dignosed by @negativekelvin obviously remains but is not causing a problem (apart from being what prompted me to refresh the tools in the first place)

Super support, many thanks,
Si.

closed 2018-10-05 by igrr