espressif/esp-idf #2475: Build and Flash error

alexgithub56, 2018-09-26:
Test@DESKTOP-IGAAV3E MINGW32 ~/esp/hello_world
$ make flash
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-80-g6c4433a5
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Compiler version: 5.2.0
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-80-g6c4433a5
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
ImportError: No module named site
make[1]: *** [/home/Test/esp/esp-idf/components/esptool_py/Makefile.projbuild:59: /home/Test/esp/hello_world/build/bootloader/bootloader.bin] Error 1
make: *** [/home/Test/esp/esp-idf/components/bootloader/Makefile.projbuild:41: /home/Test/esp/hello_world/build/bootloader/bootloader.bin] Error 2

negativekelvin, 2018-09-26:
Same as https://github.com/espressif/esp-idf/issues/2474

negativekelvin, 2018-09-26:
https://www.esp32.com/viewtopic.php?t=4703

projectgus, 2018-09-27:
Hi @alexgithub56 ,

Could you please run `make V=1` and post the full output from that command?

Are you using the prebuilt MSYS2 environment downloaded from as a zip file?

Thanks,

Angus

alexgithub56, 2018-09-27:
Hi Angus,

I used MSYS2 emvironment from:
https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20180110.zip

And this is result from make V=1 command.
Test@DESKTOP-IGAAV3E MINGW32 ~/esp/hello_world
$ make V=1
including /home/Test/esp/esp-idf/components/app_update/Makefile.projbuild...
including /home/Test/esp/esp-idf/components/bootloader/Makefile.projbuild...
including
/home/Test/esp/esp-idf/components/bootloader_support/Makefile.projbuild...
including /home/Test/esp/esp-idf/components/coap/Makefile.projbuild...
including /home/Test/esp/esp-idf/components/esp32/Makefile.projbuild...
including /home/Test/esp/esp-idf/components/esptool_py/Makefile.projbuild...
including /home/Test/esp/esp-idf/components/mbedtls/Makefile.projbuild...
including /home/Test/esp/esp-idf/components/nghttp/Makefile.projbuild...
including
/home/Test/esp/esp-idf/components/partition_table/Makefile.projbuild...
including /home/Test/esp/esp-idf/components/ulp/Makefile.projbuild...
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported:
crosstool-ng-1.22.0-80-g6c4433a5
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or
proceed at your own risk.
Compiler version: 5.2.0
PROJECT_PATH= COMPONENT_DIRS= make -C
/home/Test/esp/esp-idf/components/bootloader/subproject V=1
BUILD_DIR_BASE=/home/Test/esp/hello_world/build/bootloader TEST_COMPONENTS=
TESTS_ALL= EXCLUDE_COMPONENTS=
/home/Test/esp/hello_world/build/bootloader/bootloader.bin
make[1]: Entering directory
'/home/Test/esp/esp-idf/components/bootloader/subproject'
including /home/Test/esp/esp-idf/components/esptool_py/Makefile.projbuild...
including
/home/Test/esp/esp-idf/components/bootloader_support/Makefile.projbuild...
including
/home/Test/esp/esp-idf/components/bootloader/subproject/main/Makefile.projbuild...
WARNING: Toolchain version is not supported:
crosstool-ng-1.22.0-80-g6c4433a5
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or
proceed at your own risk.
make -C /home/Test/esp/hello_world/build/bootloader/bootloader_support -f
/home/Test/esp/esp-idf/make/component_wrapper.mk
COMPONENT_MAKEFILE=/home/Test/esp/esp-idf/components/bootloader_support/
component.mk COMPONENT_NAME=bootloader_support build
make[2]: Entering directory
'/home/Test/esp/hello_world/build/bootloader/bootloader_support'
make[2]: Nothing to be done for 'build'.
make[2]: Leaving directory
'/home/Test/esp/hello_world/build/bootloader/bootloader_support'
Target 'component-bootloader_support-build' responsible for
'/home/Test/esp/hello_world/build/bootloader/bootloader_support/libbootloader_support.a'
make -C /home/Test/esp/hello_world/build/bootloader/log -f
/home/Test/esp/esp-idf/make/component_wrapper.mk
COMPONENT_MAKEFILE=/home/Test/esp/esp-idf/components/log/component.mk
COMPONENT_NAME=log build
make[2]: Entering directory
'/home/Test/esp/hello_world/build/bootloader/log'
make[2]: Nothing to be done for 'build'.
make[2]: Leaving directory '/home/Test/esp/hello_world/build/bootloader/log'
Target 'component-log-build' responsible for
'/home/Test/esp/hello_world/build/bootloader/log/liblog.a'
make -C /home/Test/esp/hello_world/build/bootloader/spi_flash -f
/home/Test/esp/esp-idf/make/component_wrapper.mk
COMPONENT_MAKEFILE=/home/Test/esp/esp-idf/components/spi_flash/component.mk
COMPONENT_NAME=spi_flash build
make[2]: Entering directory
[... 31 lines cut ...]
make[2]: Entering directory
'/home/Test/esp/hello_world/build/bootloader/main'
make[2]: Nothing to be done for 'build'.
make[2]: Leaving directory
'/home/Test/esp/hello_world/build/bootloader/main'
Target 'component-main-build' responsible for
'/home/Test/esp/hello_world/build/bootloader/main/libmain.a'
python /home/Test/esp/esp-idf/components/esptool_py/esptool/esptool.py
--chip esp32 elf2image --flash_mode "dio" --flash_freq "40m" --flash_size
"2MB"  -o /home/Test/esp/hello_world/build/bootloader/bootloader.bin
/home/Test/esp/hello_world/build/bootloader/bootloader.elf
ImportError: No module named site
make[1]: ***
[/home/Test/esp/esp-idf/components/esptool_py/Makefile.projbuild:59:
/home/Test/esp/hello_world/build/bootloader/bootloader.bin] Error 1
make[1]: Leaving directory
'/home/Test/esp/esp-idf/components/bootloader/subproject'
make: ***
[/home/Test/esp/esp-idf/components/bootloader/Makefile.projbuild:41:
/home/Test/esp/hello_world/build/bootloader/bootloader.bin] Error 2

Thank you for your respond and sorry I'm just try ESP-IDF first time.

Regards,
Alex

alexgithub56, 2018-10-07:
Thank you very much. I have fixed my problem yet.

closed 2018-10-07 by alexgithub56