espressif/esp-idf #2478: ESP32 SD card example is not working when SDMMC interface is used.

mohinishnmrinq, 2018-09-26:
Hi
I am trying to interface SD card to ESP32 over SDMMC hardware interface.
To get started I ran the sample code given in example but I am getting some error as shown below.

I (1143) cpu_start: Pro cpu start user code
I (389) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (391) example: Initializing SD card
I (391) example: Using SDMMC peripheral
I (391) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (451) sdmmc_sd: sdmmc_check_scr: send_scr returned 0x109
E (451) example: Failed to initialize the card (ESP_ERR_INVALID_CRC). Make sure SD card lines have pull-up resistors in place.

When I tried to communicate to SD card over SPI hardware interface it works fine.

I have made all the connections as mentioned in README.md.
I am using ESP32 DevKit to run this example.
I have flashed voltage selection fuse of flash to 3.3V. And I have connected the GPIO2 to GPIO0.

Please help me resolve this issue.

igrr, 2018-09-27:
ESP_ERR_INVALID_CRC often indicates that there is a signal integrity issue between ESP and SD card. Please try lowering the bus speed to 400kHz (default is 20MHz) and see if it starts working.

Also perhaps you can describe what the hardware connections between the chip and the SD card are. Are they on the same board, what is the distance/length of traces, anything special about routing (multiple parallel slots, jumpers, etc)...

closed 2018-09-27 by Alvin1Zhang

olegantonyan, 2018-11-28:
ESP32 DevKit C4 has a capacitor from GPIO0 to GND (C15 on schematic, shown as NC, btw, but it actually exists on the board). I removed it and helped.
Now the board can be flashed (thanks to GPIO0 connected to GPIO2) and SD card works with these pins connected. No need to disconnect jumpers or whatever

rebeliousconformist, 2020-02-04:
I like open this issue again.

 example: Failed to initialize the card (ESP_ERR_INVALID_CRC). Make sure SD card lines have pull-up resistors in place

JasonFevang, 2020-07-13:
I had this problem with ESP_ERR_INVALID_CRC when using 4-bit mode only. In my case, the clock signal was the issue, and changing the jumper wire I was using to carry the clock signal was enough to improve the signal quality and not get an invalid crc error. When the design is moved to a pcb, hopefully there won't be any more issues. As a first step, check the clock and data signals with an oscilloscope or logic analyzer

rglissmann, 2020-08-31:
I finally got this program to work by changing SDMMC_FREQ_DEFAULT from 20000 to 5000. It is defined in sdmmc_types.h. I was also able to remove the pull up resistors.

thingAS, 2020-09-24:
rglissmann 
I had the same error! <send_scr returned 0x109>
with me it only works with 5000 too! 
Thank you!

wakamohle, 2021-09-11:
> I finally got this program to work by changing SDMMC_FREQ_DEFAULT from 20000 to 5000. It is defined in sdmmc_types.h. I was also able to remove the pull up resistors.

Thank you! This totally worked. The pull-up resistors were necessary, though as my SD Card module uses 5V, so that is something to take in consideration. I used an approach that didn't need to modify a *.h file, I added this line after the `sdmmc_host_t` object was initialized

` 
host.max_freq_khz = 5000;
`

idea--list, 2021-10-05:
I am on a quest to find a new framework instead of Mbed OS, just touched Arduino where the ESP32 Arduino core works just perfectly fine in SDIO4 mode. The only thing that driven me to try ESP IDF was the fact i found some SD card benchmarks suggesting ESP IDF can achieve 5-6 MB/sec read speed on the SD card i have.

So began to play around with the sd card example shipped with ESP IDF. I was keep facing the issue mentioned in the title.
Found this ticket by googling for the issue. In my case `host.max_freq_khz = 5000;` did not solve the issue.
So next i just tried `host.max_freq_khz = 400;` which still did not yield success.
With some trial&error i found that if i also comment out lines 110-115 (pullups) and allow only 400KHz for the bus then the example works about 33% of the time.

I am using ca. 15cm long bread board wires between a Wemos Lolin D32 board and a Polulu 2597 sd break out... the same setup works 100% in Arduino. Still did not compare the implementation of Arduino and ESP IDF, but based on my findings i guess this might be rather a software bug that can be improved. Hope this info helps until we get there.

LucasEstevo, 2021-11-19:
Good morning, I was having the same problem in **ESP-IDF**, as shown below:
```C
I (326) SD_CARD: Initializing SD card
I (326) SD_CARD: Usando periférico SPI
I (336) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (346) sdspi_transaction: cmd=52, R1 response: command not supported
I (396) sdspi_transaction: cmd=5, R1 response: command not supported
E (426) sdmmc_sd: sdmmc_check_scr: send_scr returned 0x109
E (426) vfs_fat_sdmmc: sdmmc_card_init failed (0x109).
I (426) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (436) SD_CARD: Failed to initialize the card (ESP_ERR_INVALID_CRC). Make sure SD card lines have pull-up resistors in place.
```

I followed **rglissmann**'s directions and it worked.

> I finally got this program to work by changing SDMMC_FREQ_DEFAULT from 20000 to 5000. It is defined in sdmmc_types.h. I was also able to remove the pull up resistors.

1º Add this to your code
```C
#include "driver/sdmmc_types.h"
```
    
2º Open it 
```sdmmc_types.h```

3º To locate and change the value 20000 to 5000
```SDMMC_FREQ_DEFAULT```

4º Save the file and compile the code

viditembrill, 2022-02-11:
Yes. It works for me too by changing SDMMC_FREQ_DEFAULT 5000.

kifelmuhammed, 2022-06-16:
thanks @rglissmann for the solution, it worked for me.

BeancurdPachelbel, 2022-07-29:
**Thanks @rglissmann** Your solution worked for me!

amolrwagh, 2022-09-13:
> Good morning, I was having the same problem in **ESP-IDF 4.3**, as shown below:
> 
> ```c
> I (326) SD_CARD: Initializing SD card
> I (326) SD_CARD: Usando periférico SPI
> I (336) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
> I (346) sdspi_transaction: cmd=52, R1 response: command not supported
> I (396) sdspi_transaction: cmd=5, R1 response: command not supported
> E (426) sdmmc_sd: sdmmc_check_scr: send_scr returned 0x109
> E (426) vfs_fat_sdmmc: sdmmc_card_init failed (0x109).
> I (426) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
> E (436) SD_CARD: Failed to initialize the card (ESP_ERR_INVALID_CRC). Make sure SD card lines have pull-up resistors in place.
> ```
> 
> I followed **rglissmann**'s directions and it worked.
> 
> > I finally got this program to work by changing SDMMC_FREQ_DEFAULT from 20000 to 5000. It is defined in sdmmc_types.h. I was also able to remove the pull up resistors.
> 
> 1º Add this to your code
> 
> ```c
> #include "driver/sdmmc_types.h"
> ```
> 
> 2º Open it `sdmmc_types.h`
> 
> 3º To locate and change the value 20000 to 5000 `SDMMC_FREQ_DEFAULT`
> 
> 4º Save the file and compile the code

Thank you very much. Its perfectly working for me.

amolrwagh, 2022-09-13:
#define PIN_NUM_MISO 2     // 10 K pull-up required
#define PIN_NUM_MOSI 15   // 10 K pull-up required
#define PIN_NUM_CLK  14    // 10 K pull-up required
#define PIN_NUM_CS   13     // 10 K pull-up required

Additionally,
#include "driver/sdmmc_types.h"  // Add this line in your sd_card esp-idf code
and change  this line 
#define SDMMC_FREQ_DEFAULT  20000    to    #define SDMMC_FREQ_DEFAULT 5000 
Example works fine.

neolu, 2023-03-07:
pinMode(15, INPUT_PULLUP);
  pinMode(2, INPUT_PULLUP);
  pinMode(4, INPUT_PULLUP);
  pinMode(12, INPUT_PULLUP);
  pinMode(13, INPUT_PULLUP);

raj-mkris, 2023-09-14:
> 5000

This thing is working fine with latest version of esp-idf,
```
I (27) boot: ESP-IDF v4.4.5-dirty 2nd stage bootloader
I (27) boot: compile time 14:21:12       
I (27) boot: chip revision: v1.0
I (31) boot.esp32: SPI Speed      : 40MHz
I (35) boot.esp32: SPI Mode       : DIO  
I (40) boot.esp32: SPI Flash Size : 4MB
I (44) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0c6ech ( 50924) map
I (114) esp_image: segment 1: paddr=0001c714 vaddr=3ffb0000 size=01d0ch (  7436) load
I (117) esp_image: segment 2: paddr=0001e428 vaddr=40080000 size=01bf0h (  7152) load
I (123) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=20460h (132192) map
I (176) esp_image: segment 4: paddr=00040488 vaddr=40081bf0 size=0b614h ( 46612) load
I (202) boot: Loaded app from partition at offset 0x10000
I (202) boot: Disabling RNG early entropy source...
I (214) cpu_start: Pro cpu up.
I (214) cpu_start: Starting app cpu, entry point is 0x40081104
0x40081104: call_start_cpu1 at C:/Users/dkp02/esp/esp-idf/components/esp_system/port/cpu_start.c:147

I (0) cpu_start: App cpu up.
I (228) cpu_start: Pro cpu start user code
I (228) cpu_start: cpu freq: 160000000
I (228) cpu_start: Application information:
I (233) cpu_start: Project name:     SD-Card
I (237) cpu_start: App version:      1
I (242) cpu_start: Compile time:     Sep 14 2023 14:19:58
I (248) cpu_start: ELF file SHA256:  13df5a9c32ec9b47...
I (254) cpu_start: ESP-IDF:          v4.4.5-dirty
I (259) cpu_start: Min chip rev:     v0.0
I (264) cpu_start: Max chip rev:     v3.99 
I (269) cpu_start: Chip rev:         v1.0
I (274) heap_init: Initializing. RAM available for dynamic allocation:
I (281) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (287) heap_init: At 3FFB2668 len 0002D998 (182 KiB): DRAM
I (293) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (299) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (306) heap_init: At 4008D204 len 00012DFC (75 KiB): IRAM
I (313) spi_flash: detected chip: generic
I (317) spi_flash: flash io: dio
I (322) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (332) example: Initializing SD card
I (332) example: Using SPI peripheral
I (342) example: Mounting filesystem
I (342) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0   
I (392) sdspi_transaction: cmd=5, R1 response: command not supported
I (552) example: Filesystem mounted
Name: SB16G
Type: SDHC/SDXC
Speed: 5 MHz
Size: 15193MB
I (552) example: Opening file /sdcard/hello.txt
I (592) example: File written
I (602) example: Renaming file /sdcard/hello.txt to /sdcard/foo.txt
I (612) example: Reading file /sdcard/foo.txt
I (612) example: Read from file: 'Hello SB16G!'
I (612) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0   
I (622) example: Card unmounted
I (622) gpio: GPIO[23]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (632) gpio: GPIO[19]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0  
I (642) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
```

zhaoyu-ma, 2024-08-23:
[image]
it works well, it solves my problem

jronaldi, 2025-01-16:
In my case, after a few DAYS of trying to understand my issues, I found out that the wires between two breadboards were too long. Shortening the path solved the issue.
In another case, it was a case of my power source not delivering enough power...
So in both cases, it was not a software problem but rather due to me being a rookie at electronics design...