espressif/esp-idf #2480: python -m pip install --user -r $IDF_PATH/requirements.txt alway error

warman2000, 2018-09-27:
Using cached https://files.pythonhosted.org/packages/e7/a7/4cd50e57cc6f436f1cc3a7e8fa700ff9b8b4d471620629074913e3735fb2/cffi-1.11.5.tar.gz
    Complete output from command python setup.py egg_info:

        No working compiler found, or bogus compiler options passed to
        the compiler from Python's standard "distutils" module.  See
        the error messages above.  Likely, the problem is not related
        to CFFI but generic to the setup.py of any Python package that
        tries to compile C code.  (Hints: on OS/X 10.8, for errors about
        -mno-fused-madd see http://stackoverflow.com/questions/22313407/
        Otherwise, see https://wiki.python.org/moin/CompLangPython or
        the IRC channel #python on irc.freenode.net.)

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in c:/users/wander/appdata/local/temp/pip-install-tskyym/cffi/

projectgus, 2018-09-27:
Hi @warman2000 ,

Are you using the MSYS2 environment from the Windows Getting Started? Please try the fix mentioned in this post:
https://esp32.com/viewtopic.php?f=13&t=7419&sid=5c7c0aedfaccb2e59fb90521b084bfd7#p31342

The environment and instructions will be updated ASAP.

Angus

warman2000, 2018-09-27:
hi @projectgu
1.  I using MSYS2 form windows.
2 . I operated according to the website's instructions, but still failed

warman2000, 2018-09-27:
I have solved the problem，Just need to execute @projectgus 
1.python -m pip install --upgrade pip
2.pacman -S mingw-w64-i686-python2-cryptography

don't need execute :pip install --upgrade setuptools

FadiBunni, 2018-09-27:
```
$ python2.7 -m pip install --user -r requirements.txt
Requirement already satisfied: setuptools in /usr/lib/python2.7/site-packages (from -r requirements.txt (line 4)) (40.4.3)
Requirement already satisfied: pyserial>=3.0 in /home/fadib/.local/lib/python2.7/site-packages (from -r requirements.txt (line 8)) (3.4)
Requirement already satisfied: future>=0.16.0 in /home/fadib/.local/lib/python2.7/site-packages (from -r requirements.txt (line 9)) (0.16.0)
Collecting cryptography (from -r requirements.txt (line 10))
  Using cached https://files.pythonhosted.org/packages/22/21/233e38f74188db94e8451ef6385754a98f3cad9b59bedf3a8e8b14988be4/cryptography-2.3.1.tar.gz
Requirement already satisfied: idna>=2.1 in /home/fadib/.local/lib/python2.7/site-packages (from cryptography->-r requirements.txt (line 10)) (2.7)
Requirement already satisfied: asn1crypto>=0.21.0 in /home/fadib/.local/lib/python2.7/site-packages (from cryptography->-r requirements.txt (line 10)) (0.24.0)
Requirement already satisfied: six>=1.4.1 in /home/fadib/.local/lib/python2.7/site-packages (from cryptography->-r requirements.txt (line 10)) (1.11.0)
Collecting cffi!=1.11.3,>=1.7 (from cryptography->-r requirements.txt (line 10))
  Using cached https://files.pythonhosted.org/packages/e7/a7/4cd50e57cc6f436f1cc3a7e8fa700ff9b8b4d471620629074913e3735fb2/cffi-1.11.5.tar.gz
Collecting enum34 (from cryptography->-r requirements.txt (line 10))
  Using cached https://files.pythonhosted.org/packages/c5/db/e56e6b4bbac7c4a06de1c50de6fe1ef3810018ae11732a50f15f62c7d050/enum34-1.1.6-py2-none-any.whl
Collecting ipaddress (from cryptography->-r requirements.txt (line 10))
  Using cached https://files.pythonhosted.org/packages/fc/d0/7fc3a811e011d4b388be48a0e381db8d990042df54aa4ef4599a31d39853/ipaddress-1.0.22-py2.py3-none-any.whl
Requirement already satisfied: pycparser in /home/fadib/.local/lib/python2.7/site-packages (from cffi!=1.11.3,>=1.7->cryptography->-r requirements.txt (line 10)) (2.19)
Building wheels for collected packages: cryptography, cffi
  Running setup.py bdist_wheel for cryptography ... error
  Complete output from command /usr/bin/python2.7 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-YFdS4S/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-0Dp8y3 --python-tag cp27:
  c/_cffi_backend.c:15:17: fatal error: ffi.h: No such file or directory
   #include <ffi.h>
                   ^
  compilation terminated.
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-install-YFdS4S/cryptography/setup.py", line 318, in <module>
      **keywords_with_side_effects(sys.argv)
    File "/usr/lib/python2.7/site-packages/setuptools/__init__.py", line 139, in setup
      _install_setup_requires(attrs)
    File "/usr/lib/python2.7/site-packages/setuptools/__init__.py", line 134, in _install_setup_requires
      dist.fetch_build_eggs(dist.setup_requires)
    File "/usr/lib/python2.7/site-packages/setuptools/dist.py", line 514, in fetch_build_eggs
      replace_conflicting=True,
    File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 777, in resolve
      replace_conflicting=replace_conflicting
    File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1060, in best_match
      return self.obtain(req, installer)
    File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1072, in obtain
      return installer(requirement)
    File "/usr/lib/python2.7/site-packages/setuptools/dist.py", line 581, in fetch_build_egg
      return cmd.easy_install(req)
    File "/usr/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 676, in easy_install
      return self.install_item(spec, dist.location, tmpdir, deps)
    File "/usr/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 702, in install_item
      dists = self.install_eggs(spec, download, tmpdir)
    File "/usr/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 887, in install_eggs
      return self.build_and_install(setup_script, setup_base)
    File "/usr/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1155, in build_and_install
      self.run_setup(setup_script, setup_base, args)
    File "/usr/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1143, in run_setup
      raise DistutilsError("Setup script exited with %s" % (v.args[0],))
  distutils.errors.DistutilsError: Setup script exited with error: command 'gcc' failed with exit status 1

  ----------------------------------------
  Failed building wheel for cryptography
  Running setup.py clean for cryptography
  Complete output from command /usr/bin/python2.7 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-YFdS4S/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" clean --all:
  c/_cffi_backend.c:15:17: fatal error: ffi.h: No such file or directory
   #include <ffi.h>
                   ^
  compilation terminated.
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-install-YFdS4S/cryptography/setup.py", line 318, in <module>
      **keywords_with_side_effects(sys.argv)
    File "/usr/lib/python2.7/site-packages/setuptools/__init__.py", line 139, in setup
[... cut ...]
    ----------------------------------------
Command "/usr/bin/python2.7 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-YFdS4S/cffi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-yIx_Vq/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-install-YFdS4S/cffi/
```

I tried the same but i got this error? How to proceed from here?

projectgus, 2018-09-27:
@FadiBunni What do you get if you try the same `pacman -S mingw-w64-i686-python2-cryptography` that was mentioned above?

FadiBunni, 2018-09-27:
I already ran the mentioned command (I will try again)
It seems that setuptools, pyserial and future is installing correctly. but cryptography and cffi is not. 

```
$ pacman -S mingw-w64-i686-python2-cryptography
warning: mingw-w64-i686-python2-cryptography-1.9-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (1) mingw-w64-i686-python2-cryptography-1.9-1

Total Installed Size:  2.19 MiB
Net Upgrade Size:      0.00 MiB

:: Proceed with installation? [Y/n] Y
(1/1) checking keys in keyring                                                                                                                                     [####################################################################################################] 100%
(1/1) checking package integrity                                                                                                                                   [####################################################################################################] 100%
(1/1) loading package files                                                                                                                                        [####################################################################################################] 100%
(1/1) checking for file conflicts                                                                                                                                  [####################################################################################################] 100%
(1/1) checking available disk space                                                                                                                                [####################################################################################################] 100%
:: Processing package changes...
(1/1) reinstalling mingw-w64-i686-python2-cryptography                                                                                                             [####################################################################################################] 100%

fadib@DESKTOP-DAA503K MSYS ~/esp-idf
$ python2.7 -m pip install --user -r requirements.txt
Requirement already satisfied: setuptools in /usr/lib/python2.7/site-packages (from -r requirements.txt (line 4)) (40.4.3)
Requirement already satisfied: pyserial>=3.0 in /home/fadib/.local/lib/python2.7/site-packages (from -r requirements.txt (line 8)) (3.4)
Requirement already satisfied: future>=0.16.0 in /home/fadib/.local/lib/python2.7/site-packages (from -r requirements.txt (line 9)) (0.16.0)
Collecting cryptography (from -r requirements.txt (line 10))
  Using cached https://files.pythonhosted.org/packages/22/21/233e38f74188db94e8451ef6385754a98f3cad9b59bedf3a8e8b14988be4/cryptography-2.3.1.tar.gz
Requirement already satisfied: idna>=2.1 in /home/fadib/.local/lib/python2.7/site-packages (from cryptography->-r requirements.txt (line 10)) (2.7)
Requirement already satisfied: asn1crypto>=0.21.0 in /home/fadib/.local/lib/python2.7/site-packages (from cryptography->-r requirements.txt (line 10)) (0.24.0)
Requirement already satisfied: six>=1.4.1 in /home/fadib/.local/lib/python2.7/site-packages (from cryptography->-r requirements.txt (line 10)) (1.11.0)
Collecting cffi!=1.11.3,>=1.7 (from cryptography->-r requirements.txt (line 10))
  Using cached https://files.pythonhosted.org/packages/e7/a7/4cd50e57cc6f436f1cc3a7e8fa700ff9b8b4d471620629074913e3735fb2/cffi-1.11.5.tar.gz
Collecting enum34 (from cryptography->-r requirements.txt (line 10))
  Using cached https://files.pythonhosted.org/packages/c5/db/e56e6b4bbac7c4a06de1c50de6fe1ef3810018ae11732a50f15f62c7d050/enum34-1.1.6-py2-none-any.whl
Collecting ipaddress (from cryptography->-r requirements.txt (line 10))
  Using cached https://files.pythonhosted.org/packages/fc/d0/7fc3a811e011d4b388be48a0e381db8d990042df54aa4ef4599a31d39853/ipaddress-1.0.22-py2.py3-none-any.whl
Requirement already satisfied: pycparser in /home/fadib/.local/lib/python2.7/site-packages (from cffi!=1.11.3,>=1.7->cryptography->-r requirements.txt (line 10)) (2.19)
Building wheels for collected packages: cryptography, cffi
  Running setup.py bdist_wheel for cryptography ... error
  Complete output from command /usr/bin/python2.7 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-q5BtRW/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-0gHQfp --python-tag cp27:
  c/_cffi_backend.c:15:17: fatal error: ffi.h: No such file or directory
   #include <ffi.h>
                   ^
  compilation terminated.
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-install-q5BtRW/cryptography/setup.py", line 318, in <module>
      **keywords_with_side_effects(sys.argv)
    File "/usr/lib/python2.7/site-packages/setuptools/__init__.py", line 139, in setup
      _install_setup_requires(attrs)
    File "/usr/lib/python2.7/site-packages/setuptools/__init__.py", line 134, in _install_setup_requires
      dist.fetch_build_eggs(dist.setup_requires)
    File "/usr/lib/python2.7/site-packages/setuptools/dist.py", line 514, in fetch_build_eggs
      replace_conflicting=True,
    File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 777, in resolve
[... cut ...]
    error: command 'gcc' failed with exit status 1

    ----------------------------------------
Command "/usr/bin/python2.7 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-q5BtRW/cffi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-byVk_h/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-install-q5BtRW/cffi/

```

dobairoland, 2018-09-27:
The output of `which python2.7` is `/mingw32/bin/python2.7`? For example I don't have `/usr/bin/python2.7` in my MSYS. 

If this passes:
`/mingw32/bin/python2.7 -m pip install --user -r requirements.txt`
then you can try to specify in menuconfig `/mingw32/bin/python2.7` as your python interpreter.

FadiBunni, 2018-09-27:
The output is of which python2.7 is:

```
$ which python2.7
/usr/bin/python2.7
```

I ran your code:

`/mingw32/bin/python2.7 -m pip install --user -r requirements.txt`

It gave me this:

`$ /mingw32/bin/python2.7 -m pip install --user -r requirements.txt
Requirement already satisfied: setuptools in c:/msys32/mingw32/lib/python2.7/site-packages (from -r requirements.txt (line 4))
Requirement already satisfied: pyserial>=3.0 in c:/msys32/home/fadib/.local/lib/python2.7/site-packages (from -r requirements.txt (line 8))
Requirement already satisfied: future>=0.16.0 in c:/msys32/home/fadib/.local/lib/python2.7/site-packages (from -r requirements.txt (line 9))
Requirement already satisfied: cryptography in c:/msys32/mingw32/lib/python2.7/site-packages (from -r requirements.txt (line 10))
Requirement already satisfied: idna>=2.1 in c:/msys32/home/fadib/.local/lib/python2.7/site-packages (from cryptography->-r requirements.txt (line 10))
Requirement already satisfied: asn1crypto>=0.21.0 in c:/msys32/home/fadib/.local/lib/python2.7/site-packages (from cryptography->-r requirements.txt (line 10))
Requirement already satisfied: six>=1.4.1 in c:/msys32/home/fadib/.local/lib/python2.7/site-packages (from cryptography->-r requirements.txt (line 10))
Requirement already satisfied: enum34 in c:/msys32/home/fadib/.local/lib/python2.7/site-packages (from cryptography->-r requirements.txt (line 10))
Requirement already satisfied: ipaddress in c:/msys32/home/fadib/.local/lib/python2.7/site-packages (from cryptography->-r requirements.txt (line 10))
Requirement already satisfied: cffi>=1.7 in c:/msys32/mingw32/lib/python2.7/site-packages (from cryptography->-r requirements.txt (line 10))
Requirement already satisfied: pycparser in c:/msys32/home/fadib/.local/lib/python2.7/site-packages (from cffi>=1.7->cryptography->-r requirements.txt (line 10))
You are using pip version 9.0.1, however version 18.0 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.`

Which means that it is now working? So why do I need to use "mingw32/bin/python2.7" and not just python2.7?

dobairoland, 2018-09-27:
Yes, it will work if you specify `/mingw32/bin/python2.7` as your interpreter in the IDF menuconfig. I guess that `/usr/bin/python2.7` is another Python installed in your system for which the cryptography package cannot be installed by pip.

You installed `mingw-w64-i686-python2-cryptography`, therefore, you have to use the mingw Python interpreter because the cryptography package is installed for that one.

oclyke, 2018-09-27:
Hello everyone, I am having a similar problem. 

To recreate the problem I've taken these steps:

- Freshly unzipped msys32 into (C:)
- create 'esp' directory in the 'home/user' directory
- clone idf repo into 'esp' directory with 'git clone --recursive https://github.com/espressif/esp-idf.git'
- create script to export path variable "export IDF_PATH="C:/msys32/home/owen.lyke/esp/esp-idf"
- close msys32 and reopen, verify path variable existence with printenv IDF_PATH
- check python version and 'which python2.7' (it is /mingw32/bin/python2.7 )
- upgraded pip with 'python -m pip install --upgrade pip'
- upgrade setuptools with 'pip install --upgrade setuptools'
- used 'pacman -S mingw-w64-i686-python2-cryptography' per above
- tried to install requirements with 'python2.7 -m pip install --user -r $IDF_PATH/requirements.txt' 

After trying to install the required files I was surprised to see that setuptools and pyserial were already up to date (first time through all this they weren't, and I thought I had been careful to completely remove the old msys32 instance.)

Once again I found an error occurring when trying to install future. I thought I might as well try to install cryptography without future using 'python2.7 -m pip install --user cryptography' and it seems to have completed successfully. 

Tried to install future again for good luck but had the same problem. 

From my level of experience with python and operating systems (little to none!) I'm not even sure what question I need to ask to help sort this out. Hopefully the extra data point can be useful though, and I'll be watching this play out. I'm really excited to use the IDF!  Thank you for taking the time to look into this everyone. 

Below is all my output from after re-starting msys32 for the environment path.

```
owen.lyke@SX-OWENL MINGW32 ~
$ printenv IDF_PATH
C:/msys32/home/owen.lyke/esp/esp-idf

owen.lyke@SX-OWENL MINGW32 ~
$ python --version
Python 2.7.14

owen.lyke@SX-OWENL MINGW32 ~
$ which python 2.7
/mingw32/bin/python
which: no 2.7 in (/mingw32/bin:/usr/local/bin:/usr/bin:/bin:/c/Windows/System32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0/:/opt/xtensa-esp32-elf/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl)

owen.lyke@SX-OWENL MINGW32 ~
$ which python2.7
/mingw32/bin/python2.7

owen.lyke@SX-OWENL MINGW32 ~
$ python -m pip install --upgrade pip
Collecting pip
  Using cached https://files.pythonhosted.org/packages/5f/25/e52d3f31441505a5f3af41213346e5b6c221c9e086a166f3703d2ddaf940/pip-18.0-py2.py3-none-any.whl
Installing collected packages: pip
  Found existing installation: pip 9.0.1
    Uninstalling pip-9.0.1:
      Successfully uninstalled pip-9.0.1
Successfully installed pip-18.0

owen.lyke@SX-OWENL MINGW32 ~
$ pip install --upgrade setuptools
Collecting setuptools
  Using cached https://files.pythonhosted.org/packages/96/06/c8ee69628191285ddddffb277bd5abdf769166e7a14b867c2a172f0175b1/setuptools-40.4.3-py2.py3-none-any.whl
Installing collected packages: setuptools
  Found existing installation: setuptools 36.6.0.post20171017
    Uninstalling setuptools-36.6.0.post20171017:
      Successfully uninstalled setuptools-36.6.0.post20171017
Successfully installed setuptools-40.4.3

owen.lyke@SX-OWENL MINGW32 ~
$ python2.7 -m pip install --upgrade pip
Requirement already up-to-date: pip in c:/msys32/mingw32/lib/python2.7/site-packages (18.0)

owen.lyke@SX-OWENL MINGW32 ~
$ pacman -S mingw-w64-i686-python2-cryptography
resolving dependencies...
[... 50 lines cut ...]
    distutils.errors.DistutilsPlatformError: VC 6.0 is not supported by this module

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in c:/users/owen~1.lyk/appdata/local/temp/pip-install-p_eygt/future/

owen.lyke@SX-OWENL MINGW32 ~
$ python2.7 -m pip install --user cryptography
Requirement already satisfied: cryptography in c:/msys32/mingw32/lib/python2.7/site-packages (1.9)
Requirement already satisfied: idna>=2.1 in c:/msys32/mingw32/lib/python2.7/site-packages (from cryptography) (2.6)
Requirement already satisfied: asn1crypto>=0.21.0 in c:/msys32/mingw32/lib/python2.7/site-packages (from cryptography) (0.22.0)
Collecting six>=1.4.1 (from cryptography)
  Using cached https://files.pythonhosted.org/packages/67/4b/141a581104b1f6397bfa78ac9d43d8ad29a7ca43ea90a2d863fe3056e86a/six-1.11.0-py2.py3-none-any.whl
Collecting enum34 (from cryptography)
  Downloading https://files.pythonhosted.org/packages/c5/db/e56e6b4bbac7c4a06de1c50de6fe1ef3810018ae11732a50f15f62c7d050/enum34-1.1.6-py2-none-any.whl
Collecting ipaddress (from cryptography)
  Downloading https://files.pythonhosted.org/packages/fc/d0/7fc3a811e011d4b388be48a0e381db8d990042df54aa4ef4599a31d39853/ipaddress-1.0.22-py2.py3-none-any.whl
Requirement already satisfied: cffi>=1.7 in c:/msys32/mingw32/lib/python2.7/site-packages (from cryptography) (1.10.0)
Requirement already satisfied: pycparser in c:/msys32/mingw32/lib/python2.7/site-packages (from cffi>=1.7->cryptography) (2.17)
Installing collected packages: six, enum34, ipaddress
Successfully installed enum34-1.1.6 ipaddress-1.0.22 six-1.11.0

owen.lyke@SX-OWENL MINGW32 ~
$ python2.7 -m pip install --user future>=0.16.0
Command "python setup.py egg_info" failed with error code 1 in c:/users/owen~1.lyk/appdata/local/temp/pip-install-v9mfoo/future/
```

dobairoland, 2018-09-27:
@oclyke  Please try the following commands in this order:

```bash
/mingw32/bin/python -m pip install --user --upgrade pip
/mingw32/bin/python -m pip install --user --upgrade setuptools
pacman -S mingw-w64-i686-python2-cryptography
/mingw32/bin/python -m pip install --user -r $IDF_PATH/requirements.txt
```

projectgus, 2018-10-01:
It seems like MSYS2/MINGW32 needs a MSYS-installed setuptools, installing (or upgrading) setuptools via pip actually breaks pip (causes stack traces involving `monkey.py`).

If you already upgraded setuptools via pip (or are getting errors with stack traces including `monkey.py`, try:

```
python -m pip uninstall setuptools
pacman -S mingw-w64-i686-python2-setuptools
```

chegewara, 2018-10-01:
I am assuming it works, but for some reason i am getting those errors:
```
error: failed retrieving file 'mingw-w64-i686-python2-setuptools-36.6.0-1-any.pkg.tar.xz' from repo.msys2.org : The requested URL returned error: 404
error: failed retrieving file 'mingw-w64-i686-python2-setuptools-36.6.0-1-any.pkg.tar.xz' from downloads.sourceforge.net : The requested URL returned error: 404
error: failed retrieving file 'mingw-w64-i686-python2-setuptools-36.6.0-1-any.pkg.tar.xz' from www2.futureware.at : The requested URL returned error: 404
```
and for cryptography (`pacman -S mingw-w64-i686-python2-cryptography`):
```
error: failed retrieving file 'mingw-w64-i686-python2-pycparser-2.17-1-any.pkg.tar.xz' from repo.msys2.org : The requested URL returned error: 404
error: failed retrieving file 'mingw-w64-i686-python2-pycparser-2.17-1-any.pkg.tar.xz' from downloads.sourceforge.net : The requested URL returned error: 404
error: failed retrieving file 'mingw-w64-i686-python2-pycparser-2.17-1-any.pkg.tar.xz' from www2.futureware.at : The requested URL returned error: 404
warning: failed to retrieve some files
error: failed retrieving file 'mingw-w64-i686-python2-cffi-1.10.0-1-any.pkg.tar.xz' from repo.msys2.org : The requested URL returned error: 404
error: failed retrieving file 'mingw-w64-i686-python2-cffi-1.10.0-1-any.pkg.tar.xz' from downloads.sourceforge.net : The requested URL returned error: 404
error: failed retrieving file 'mingw-w64-i686-python2-cffi-1.10.0-1-any.pkg.tar.xz' from www2.futureware.at : The requested URL returned error: 404
warning: failed to retrieve some files
error: failed retrieving file 'mingw-w64-i686-python2-pyasn1-0.2.3-1-any.pkg.tar.xz' from repo.msys2.org : The requested URL returned error: 404
error: failed retrieving file 'mingw-w64-i686-python2-pyasn1-0.2.3-1-any.pkg.tar.xz' from downloads.sourceforge.net : The requested URL returned error: 404
error: failed retrieving file 'mingw-w64-i686-python2-pyasn1-0.2.3-1-any.pkg.tar.xz' from www2.futureware.at : The requested URL returned error: 404
warning: failed to retrieve some files
error: failed retrieving file 'mingw-w64-i686-python2-idna-2.6-1-any.pkg.tar.xz' from repo.msys2.org : The requested URL returned error: 404
error: failed retrieving file 'mingw-w64-i686-python2-idna-2.6-1-any.pkg.tar.xz' from downloads.sourceforge.net : The requested URL returned error: 404
error: failed retrieving file 'mingw-w64-i686-python2-idna-2.6-1-any.pkg.tar.xz' from www2.futureware.at : The requested URL returned error: 404
warning: failed to retrieve some files
error: failed retrieving file 'mingw-w64-i686-python2-asn1crypto-0.22.0-1-any.pkg.tar.xz' from repo.msys2.org : The requested URL returned error: 404
error: failed retrieving file 'mingw-w64-i686-python2-asn1crypto-0.22.0-1-any.pkg.tar.xz' from downloads.sourceforge.net : The requested URL returned error: 404
error: failed retrieving file 'mingw-w64-i686-python2-asn1crypto-0.22.0-1-any.pkg.tar.xz' from www2.futureware.at : The requested URL returned error: 404
warning: failed to retrieve some files
error: failed retrieving file 'mingw-w64-i686-python2-cryptography-1.9-1-any.pkg.tar.xz' from repo.msys2.org : The requested URL returned error: 404
error: failed retrieving file 'mingw-w64-i686-python2-cryptography-1.9-1-any.pkg.tar.xz' from downloads.sourceforge.net : The requested URL returned error: 404
error: failed retrieving file 'mingw-w64-i686-python2-cryptography-1.9-1-any.pkg.tar.xz' from www2.futureware.at : The requested URL returned error: 404
```

FadiBunni, 2018-10-03:
Hello, I chose to start from the begning using the https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup-scratch.html walk-through, that actually worked.

closed 2018-10-05 by igrr

chegewara, 2018-10-11:
after fresh git pull i get this issue with requirements:
```
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Traceback (most recent call last):
  File "D:/msys32/home/imper/esp/esp-idf/tools/check_python_dependencies.py", line 50, in <module>
    import setuptools
  File "D:/msys32/home/imper/.local/lib/python2.7/site-packages/setuptools/__init__.py", line 191, in <module>
    monkey.patch_all()
  File "D:/msys32/home/imper/.local/lib/python2.7/site-packages/setuptools/monkey.py", line 101, in patch_all
    patch_for_msvc_specialized_compiler()
  File "D:/msys32/home/imper/.local/lib/python2.7/site-packages/setuptools/monkey.py", line 164, in patch_for_msvc_specialized_compiler
    patch_func(*msvc9('find_vcvarsall'))
  File "D:/msys32/home/imper/.local/lib/python2.7/site-packages/setuptools/monkey.py", line 151, in patch_params
    mod = import_module(mod_name)
  File "D:/msys32/mingw32/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "D:/msys32/mingw32/lib/python2.7/distutils/msvc9compiler.py", line 306, in <module>
    raise DistutilsPlatformError("VC %0.1f is not supported by this module" % VERSION)
distutils.errors.DistutilsPlatformError: VC 6.0 is not supported by this module
make: *** [/d/msys32/home/imper/esp/esp-idf/make/project.mk:454: check_python_dependencies] Error 1
```
Last time `git pull`-ed before today about week ago and all was good (except issue from OP, but got resolved).

projectgus, 2018-10-11:
@chegewara This happens if a version of setuptools is installed on MSYS2 via pip.

You should be able to `pip uninstall setuptools` and then re-run `make`, it will probably give you another error but it will work after that.

If you get really stuck then the Windows Getting Started guide has an updated version of the MSYS2 zip (20181001).

chegewara, 2018-10-11:
@projectgus Like i said, it was weird because we all got this issue described in OP about 2 weeks ago. I managed to fix it and all was ok until another git pull yesterday. The same issue happened even if i didnt install anything with pip this time. But again i found solution, this time i had to use:
`python uninstall setuptool `

projectgus, 2018-10-11:
Ah, I had this tab open from yesterday and I'd missed the edit. Glad you got it resolved.

It does seem strange that it suddenly started failing, but it should be vOK from here forward.

chegewara, 2018-11-30:
@projectgus Again the same issue with master branch today:

```
$ pip install future
Collecting future
  Using cached https://files.pythonhosted.org/packages/90/52/e20466b85000a181e1e144fd8305caf2cf475e2f9674e797b222f8105f5f/future-0.17.1.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "D:/msys32/mingw32/lib/python2.7/site-packages/setuptools/__init__.py", line 195, in <module>
        monkey.patch_all()
      File "D:/msys32/mingw32/lib/python2.7/site-packages/setuptools/monkey.py", line 101, in patch_all
        patch_for_msvc_specialized_compiler()
      File "D:/msys32/mingw32/lib/python2.7/site-packages/setuptools/monkey.py", line 164, in patch_for_msvc_specialized_compiler
        patch_func(*msvc9('find_vcvarsall'))
      File "D:/msys32/mingw32/lib/python2.7/site-packages/setuptools/monkey.py", line 151, in patch_params
        mod = import_module(mod_name)
      File "D:/msys32/mingw32/lib/python2.7/importlib/__init__.py", line 37, in import_module
        __import__(name)
      File "D:/msys32/mingw32/lib/python2.7/distutils/msvc9compiler.py", line 306, in <module>
        raise DistutilsPlatformError("VC %0.1f is not supported by this module" % VERSION)
    distutils.errors.DistutilsPlatformError: VC 6.0 is not supported by this module

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in c:/users/imper/appdata/local/temp/pip-install-gabu7h/future/
```

sravanth005, 2019-03-01:
@projectgus 

admin-unizen@adminunizen-Lenovo-ideapad-520-15IKB:~/esp-who/examples/single_chip/camera_web_server$ make all
Toolchain path: /home/admin-unizen/esp32/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Compiler version: 5.2.0
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
The following Python requirements are not satisfied:
cryptography>=2.1.4
Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
Alternatively, you can run "/usr/bin/python -m pip install --user -r /home/admin-unizen/esp/esp-idf/requirements.txt" for resolving the issue.
/home/admin-unizen/esp/esp-idf/make/project.mk:500: recipe for target 'check_python_dependencies' failed
**make: *** [check_python_dependencies] Error 1**

please help me out to clear this error as iam a fresher i dont know to correct if anything is to be installed means just send me the link or commands for ubuntu

dobairoland, 2019-03-01:
@sravanth005 

- `sudo apt-get install python-cryptography` for Ubuntu 18.04 or newer.
- `python -m pip install --user cryptography>=2.1.4` for older Ubuntu.

sravanth005, 2019-03-04:
@dobairoland ,
**hi i have UBUNTU 16.04 version.i have tried both the commands which you have suggested, i have did both the commands copied that and i have pasted it even i have pasted the problem which iam facing after flashing**

sudo apt-get install python-cryptography
[sudo] password for admin-unizen: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python-cryptography is already the newest version (1.2.3-1ubuntu0.2).
0 upgraded, 0 newly installed, 0 to remove and 22 not upgraded.

admin-unizen@adminunizen-Lenovo-ideapad-520-15IKB:~/esp-who/examples/single_chip/camera_web_server$ make flash
make: execvp: xtensa-esp32-elf-gcc: Permission denied
expr: syntax error
/bin/sh: 1: xtensa-esp32-elf-gcc: Permission denied
/bin/sh: 1: xtensa-esp32-elf-gcc: Permission denied
Toolchain path: 
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
make[1]: execvp: xtensa-esp32-elf-gcc: Permission denied
expr: syntax error
/bin/sh: 1: xtensa-esp32-elf-gcc: Permission denied
/bin/sh: 1: xtensa-esp32-elf-gcc: Permission denied
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
The following Python requirements are not satisfied:
cryptography>=2.1.4
Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
Alternatively, you can run "/usr/bin/python -m pip install --user -r /home/admin-unizen/esp32/esp-idf/requirements.txt" for resolving the issue.
/home/admin-unizen/esp32/esp-idf/make/project.mk:500: recipe for target 'check_python_dependencies' failed
make: *** [check_python_dependencies] Error 1

admin-unizen@adminunizen-Lenovo-ideapad-520-15IKB:~/esp32/xtensa-esp32-elf/bin$ python -m pip install --user cryptography>=2.1.4
You are using pip version 8.1.1, however version 19.0.3 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
admin-unizen@adminunizen-Lenovo-ideapad-520-15IKB:~/esp32/xtensa-esp32-elf/bin$ pip install --upgrade pip
Traceback (most recent call last):
  File "/home/admin-unizen/.local/bin/pip", line 7, in <module>
    from pip._internal import main
ImportError: No module named _internal

same error is displayed after flashing.

is there any other way to fix this problem.please do send the options to solve this error for the os UBUNTU 16.04 LTS

dobairoland, 2019-03-04:
@sravanth005 

You have two problems:
1. You are getting permission denied error for the toolchain. Probably you unpacked the toolchain with `sudo` and therefore, your regular user doesn't have access to it. You should remove the installed toolchain with `sudo rm -r <path-to-toolchain>`. After that please reinstall it following the steps in the documentation (without `sudo` or being logged-in as `root`).
2. Your `pip` is broken. Do a forced reinstall of `python-pip` package from the Ubuntu repository. Please see the documentation of `apt-get` to see how to do it.  Don't use `pip install --upgrade pip` because that is breaking your `pip`. After that you can resolve your issue with  `python -m pip install --user cryptography>=2.1.4`.

sravanth005, 2019-03-04:
@dobairoland 
i tried it,again its not happening.Can you please send all the commands from first to last to download the toolchain in ubuntu 16.04 LTS.So that i can do it very carefully.

krzychb, 2019-03-04:
Hi @sravanth005,

> Can you please send all the commands from first to last to download the toolchain in ubuntu 16.04 LTS.So that i can do it very carefully.

Here are [all the commands from first to last to download the toolchain in ubuntu 16.04](
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/linux-setup.html#toolchain-setup)

sravanth005, 2019-03-05:
hi @krzychb 
i face this how to continue this

$ pip install --upgrade pip
Traceback (most recent call last):
  File "/home/admin-unizen/.local/bin/pip", line 7, in <module>
    from pip._internal import main
ImportError: No module named _internal

does any one have any video clip how the how the  set up of tool chain is being run for ubuntu 16.04 please do share it may assists me.

sravanth005, 2019-03-05:
https://robotzero.one/esp32-camera-module/#comment-2917

i trying to run the program from the above link which is all about ESP32 CAM. i followed each and every steps according to that link but at last i face error so anyone can go through that link and check and say that if i go through that link can flash at last if yes please comment because im using UBUNTU 16.04 os.

especially this part (see below) i get struck with so how to do this step for Ubuntu os.for more info check the link above.

ESP32 Development Environment

Download the 32bit version (-i686-) of MYSYS from here: http://www.msys2.org/
Install following the steps on the page above but uncheck the “Run MSYS2 32-bit now” checkbox at the end.
Navigate to C:\msys32\ and double-click mingw32.exe to run the application.
Copy the contents of the file here https://raw.githubusercontent.com/espressif/esp-idf/master/tools/windows/windows_install_prerequisites.sh to a new file in C:\msys32\
Copy or paste the command below and press enter:

sravanth005, 2019-03-05:
hi @krzychb 
i installed the tool chain as per the instruction given in that link thanks for sharing the link @krzychb.
But i face one error while making menu configuration.i will paste the error which i face please try to help me out guys

~/esp-who/examples/single_chip/camera_web_server$ make menuconfig
Makefile:11: /home/admin-unizen/esp32/esp-idf/make/project.mk: No such file or directory
make: *** No rule to make target '/home/admin-unizen/esp32/esp-idf/make/project.mk'.  Stop.

  ~/esp/hello_world$ make menuconfig
Makefile:8: /home/admin-unizen/esp32/esp-idf/make/project.mk: No such file or directory
make: *** No rule to make target '/home/admin-unizen/esp32/esp-idf/make/project.mk'.  Stop.

i tried in two ways but i fall with same error

sravanth005, 2019-03-05:
hi @dobairoland ,

admin-unizen@adminunizen-Lenovo-ideapad-520-15IKB:~/esp-who/examples/single_chip/camera_web_server$ make flash
make: xtensa-esp32-elf-gcc: Command not found
expr: syntax error
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
Toolchain path: 
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
make[1]: xtensa-esp32-elf-gcc: Command not found
expr: syntax error
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
make[1]: xtensa-esp32-elf-gcc: Command not found
expr: syntax error
CC build/bootloader/bootloader_support/src/bootloader_flash.o
make[2]: xtensa-esp32-elf-gcc: Command not found
/home/admin-unizen/esp32/esp-idf/make/component_wrapper.mk:289: recipe for target 'src/bootloader_flash.o' failed
make[2]: *** [src/bootloader_flash.o] Error 127
/home/admin-unizen/esp32/esp-idf/make/project.mk:551: recipe for target 'component-bootloader_support-build' failed
make[1]: *** [component-bootloader_support-build] Error 2
/home/admin-unizen/esp32/esp-idf/components/bootloader/Makefile.projbuild:41: recipe for target '/home/admin-unizen/esp-who/examples/single_chip/camera_web_server/build/bootloader/bootloader.bin' failed
make: *** [/home/admin-unizen/esp-who/examples/single_chip/camera_web_server/build/bootloader/bootloader.bin] Error 2

how to solve this any idea?anyone please suggest

krzychb, 2019-03-05:
Hi @sravanth005,

MYSYS is applicable for Windows, not for Ubuntu.

If you are using Ubuntu, you first need to follow [Getting Started Guide](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html), going the path specific for Linux and be able to successfully complete all the steps up to and including [Monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#monitorl) section.

Without getting this done I do not a reasonable way to help you load ESP-WHO application. 

Also this section is for tracking bugs in ESP-IDF. For general help with ESP-IDF installation please check [esp32.com](https://esp32.com/viewforum.php?f=13) forum.

sravanth005, 2019-03-06:
hi @krzychb 

    Download this file, then extract it in ~/esp directory:

        for 64-bit Linux:

        mkdir -p ~/esp
        cd ~/esp
        tar -xzf ~/Downloads/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz

        for 32-bit Linux:

        mkdir -p ~/esp
        cd ~/esp
        tar -xzf ~/Downloads/xtensa-esp32-elf-linux32-1.22.0-80-g6c4433a-5.2.0.tar.gz

    The toolchain will be extracted into ~/esp/xtensa-esp32-elf/ directory.

    To use it, you will need to update your PATH environment variable in ~/.profile file. To make xtensa-esp32-elf available for all terminal sessions, add the following line to your ~/.profile file:

    export PATH="$HOME/esp/xtensa-esp32-elf/bin:$PATH"

    Alternatively, you may create an alias for the above command. This way you can get the toolchain only when you need it. To do this, add different line to your ~/.profile file:

    alias get_esp32='export PATH="$HOME/esp/xtensa-esp32-elf/bin:$PATH"'

    Then when you need the toolchain you can type get_esp32 on the command line and the toolchain will be added to your PATH.

i have downloaded the 64 bit linux file. after that i could not able to move on .what is that $PATH how to set a path can anyone explain with example.

sravanth005, 2019-03-06:
hi @dobairoland ,
i got the path as in Getting Started Guide. but i dont know how to move on with:

If you have /bin/bash set as login shell, and both .bash_profile and .profile exist, then update .bash_profile instead. In CentOS, alias should set in .bashrc.

can anyone help me out how to  move on with that .bash command.please explain me with some example in step by step.

projectgus, 2019-03-06:
@sravanth005 Setting the PATH variable on Linux is not specific t ESP-IDF, it's part of configuring a your Linux system. There are many resources online which can help you, for example I googled "set PATH .bash_profile linux" and found this guide among others: https://stackabuse.com/how-to-permanently-set-path-in-linux/

sravanth005, 2019-03-07:
after i do make command i always fall up this error i tried in lot of ways but i cant find a way to clear this 

The following Python requirements are not satisfied:
cryptography>=2.1.4
Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
Alternatively, you can run "/usr/bin/python -m pip install --user -r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt" for resolving the issue.
/home/admin-unizen/Desktop/iotbits_esp32/esp-idf/make/project.mk:500: recipe for target 'check_python_dependencies' failed
make: *** [check_python_dependencies] Error 1

how to clear this error can any one suggest me the commands for Ubuntu 16.04.
i tried a lot cant make it out so please someone send the commands to correct that error.

krzychb, 2019-03-07:
Hi @sravanth005,

From you logs I see `esp-idf` installed in `Desktop/iotbits_esp32` directory instead of `esp` directory. People are more likely to help you find the issue if you follow default installation steps.

[image]

Do you know how to "adjust respective commands" and have you done that?

Were you getting any error messages during installation before you finally got to "make" command?

What output do you get when you run:

```
 /usr/bin/python -m pip install --user -r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt
```

sravanth005, 2019-03-08:
hi @krzychb 
how to adjust command?

not during installation after i do 'make' command.i fall up with that error as i mentioned in my last post.

the output i get after i install your command 
$  /usr/bin/python -m pip install --user -r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Requirement already satisfied: setuptools in /usr/lib/python2.7/dist-packages (from -r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 4)) (20.7.0)
Requirement already satisfied: pyserial>=3.0 in /usr/lib/python2.7/dist-packages (from -r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 8)) (3.0.1)
Requirement already satisfied: future>=0.15.2 in /usr/lib/python2.7/dist-packages (from -r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 9)) (0.15.2)
Collecting cryptography>=2.1.4 (from -r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 10))
  Using cached https://files.pythonhosted.org/packages/c3/c1/cf8665c955c9393e9ff0872ba6cd3dc6f46ef915e94afcf6e0410508ca69/cryptography-2.6.1-cp27-cp27mu-manylinux1_x86_64.whl
Requirement already satisfied: pyparsing>=2.0.3 in /usr/lib/python2.7/dist-packages (from -r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 11)) (2.0.3)
Requirement already satisfied: enum34; python_version < "3" in /usr/lib/python2.7/dist-packages (from cryptography>=2.1.4->-r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 10)) (1.1.2)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in ./.local/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 10)) (1.12.2)
Requirement already satisfied: six>=1.4.1 in /usr/lib/python2.7/dist-packages (from cryptography>=2.1.4->-r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 10)) (1.10.0)
Requirement already satisfied: asn1crypto>=0.21.0 in ./.local/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 10)) (0.24.0)
Requirement already satisfied: ipaddress; python_version < "3" in /usr/lib/python2.7/dist-packages (from cryptography>=2.1.4->-r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 10)) (1.0.16)
Requirement already satisfied: pycparser in ./.local/lib/python2.7/site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 10)) (2.19)
Installing collected packages: cryptography
Successfully installed cryptography-2.6.1

sravanth005, 2019-03-08:
hi @krz
how to adjust command?

not during installation after i do 'make' command.i fall up with that error as i mentioned in my last post.

the output i get after i install your command 
$  /usr/bin/python -m pip install --user -r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Requirement already satisfied: setuptools in /usr/lib/python2.7/dist-packages (from -r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 4)) (20.7.0)
Requirement already satisfied: pyserial>=3.0 in /usr/lib/python2.7/dist-packages (from -r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 8)) (3.0.1)
Requirement already satisfied: future>=0.15.2 in /usr/lib/python2.7/dist-packages (from -r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 9)) (0.15.2)
Collecting cryptography>=2.1.4 (from -r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 10))
  Using cached https://files.pythonhosted.org/packages/c3/c1/cf8665c955c9393e9ff0872ba6cd3dc6f46ef915e94afcf6e0410508ca69/cryptography-2.6.1-cp27-cp27mu-manylinux1_x86_64.whl
Requirement already satisfied: pyparsing>=2.0.3 in /usr/lib/python2.7/dist-packages (from -r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 11)) (2.0.3)
Requirement already satisfied: enum34; python_version < "3" in /usr/lib/python2.7/dist-packages (from cryptography>=2.1.4->-r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 10)) (1.1.2)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in ./.local/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 10)) (1.12.2)
Requirement already satisfied: six>=1.4.1 in /usr/lib/python2.7/dist-packages (from cryptography>=2.1.4->-r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 10)) (1.10.0)
Requirement already satisfied: asn1crypto>=0.21.0 in ./.local/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 10)) (0.24.0)
Requirement already satisfied: ipaddress; python_version < "3" in /usr/lib/python2.7/dist-packages (from cryptography>=2.1.4->-r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 10)) (1.0.16)
Requirement already satisfied: pycparser in ./.local/lib/python2.7/site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt (line 10)) (2.19)
Installing collected packages: cryptography
Successfully installed cryptography-2.6.1

krzychb, 2019-03-08:
I see now `Successfully installed cryptography-2.6.1` that apparently was missing.

What output do you get now once you run "make"?

sravanth005, 2019-03-08:
hi @krzychb 

**i was about to do project in this and same as old i end up with same error.**

admin-unizen@adminunizen-Lenovo-ideapad-520-15IKB:~$ cd ~
admin-unizen@adminunizen-Lenovo-ideapad-520-15IKB:~$ git clone --recursive https://github.com/espressif/esp-who.git
fatal: destination path 'esp-who' already exists and is not an empty directory.
admin-unizen@adminunizen-Lenovo-ideapad-520-15IKB:~$ cd ~/esp-who/examples/single_chip/camera_web_server/
admin-unizen@adminunizen-Lenovo-ideapad-520-15IKB:~/esp-who/examples/single_chip/camera_web_server$ make menuconfig
Toolchain path: /home/admin-unizen/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
MENUCONFIG

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

admin-unizen@adminunizen-Lenovo-ideapad-520-15IKB:~/esp-who/examples/single_chip/camera_web_server$ make flash
Toolchain path: /home/admin-unizen/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
The following Python requirements are not satisfied:
cryptography>=2.1.4
Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
Alternatively, you can run "/usr/bin/python -m pip install --user -r /home/admin-unizen/esp/esp-idf/requirements.txt" for resolving the issue.
/home/admin-unizen/esp/esp-idf/make/project.mk:500: recipe for target 'check_python_dependencies' failed
make: *** [check_python_dependencies] Error 1

**i even tried your command with some edit to install cryptography-2.6.1 but it was not working may be if im wrong with my command means please suggest the commands to install cryptography-2.6.1 for this location.**

admin-unizen@adminunizen-Lenovo-ideapad-520-15IKB:~$ /usr/bin/python -m pip install --user -r /home/admin-unizen/esp/esp-id/requirements.txt
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Could not open requirements file: [Errno 2] No such file or directory: '/home/admin-unizen/esp/esp-id/requirements.txt'

sravanth005, 2019-03-11:
hi @warman2000 

admin-unizen@adminunizen-Lenovo-ideapad-520-15IKB:~/esp-who/examples/single_chip/camera_web_server$ make flash
Toolchain path: /home/admin-unizen/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
GENCONFIG
App "camera_web_server" version: v0.9.1-12-g058c427
CC build/bootloader/bootloader_support/src/bootloader_flash.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/bootloader_random.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/soc/esp32/cpu_util.o
CC build/bootloader/soc/esp32/rtc_time.o
CC build/bootloader/soc/esp32/rtc_init.o
CC build/bootloader/soc/esp32/rtc_clk_init.o
CC build/bootloader/soc/esp32/rtc_sleep.o
CC build/bootloader/soc/esp32/rtc_clk.o
CC build/bootloader/soc/src//memory_layout_utils.o
AR build/bootloader/soc/libsoc.a
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
CC build/bootloader/efuse/esp32/esp_efuse_table.o
CC build/bootloader/efuse/src/esp_efuse_api.o
CC build/bootloader/efuse/src/esp_efuse_utility.o
CC build/bootloader/efuse/src/esp_efuse_fields.o
AR build/bootloader/efuse/libefuse.a
LD build/bootloader/bootloader.elf
esptool.py v2.6
The following Python requirements are not satisfied:
cryptography>=2.1.4
Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
Alternatively, you can run "/usr/bin/python -m pip install --user -r /home/admin-unizen/esp/esp-idf/requirements.txt" for resolving the issue.
/home/admin-unizen/esp/esp-idf/make/project.mk:500: recipe for target 'check_python_dependencies' failed
make: *** [check_python_dependencies] Error 1
admin-unizen@adminunizen-Lenovo-ideapad-520-15IKB:~/esp-who/examples/single_chip/camera_web_server$ /usr/bin/python -m pip install --user -r /home/admin-unizen/
esp/esp-idf/requirements.txt
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Requirement already satisfied: setuptools in /usr/lib/python2.7/dist-packages (from -r /home/admin-unizen/esp/esp-idf/requirements.txt (line 4)) (20.7.0)
Requirement already satisfied: pyserial>=3.0 in /usr/lib/python2.7/dist-packages (from -r /home/admin-unizen/esp/esp-idf/requirements.txt (line 8)) (3.0.1)
Requirement already satisfied: future>=0.15.2 in /usr/lib/python2.7/dist-packages (from -r /home/admin-unizen/esp/esp-idf/requirements.txt (line 9)) (0.15.2)
Collecting cryptography>=2.1.4 (from -r /home/admin-unizen/esp/esp-idf/requirements.txt (line 10))
  Using cached https://files.pythonhosted.org/packages/c3/c1/cf8665c955c9393e9ff0872ba6cd3dc6f46ef915e94afcf6e0410508ca69/cryptography-2.6.1-cp27-cp27mu-manylinux1_x86_64.whl
Requirement already satisfied: pyparsing>=2.0.3 in /usr/lib/python2.7/dist-packages (from -r /home/admin-unizen/esp/esp-idf/requirements.txt (line 11)) (2.0.3)
Requirement already satisfied: enum34; python_version < "3" in /usr/lib/python2.7/dist-packages (from cryptography>=2.1.4->-r /home/admin-unizen/esp/esp-idf/requirements.txt (line 10)) (1.1.2)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in /home/admin-unizen/.local/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/admin-unizen/esp/esp-idf/requirements.txt (line 10)) (1.12.2)
Requirement already satisfied: six>=1.4.1 in /usr/lib/python2.7/dist-packages (from cryptography>=2.1.4->-r /home/admin-unizen/esp/esp-idf/requirements.txt (line 10)) (1.10.0)
Requirement already satisfied: asn1crypto>=0.21.0 in /home/admin-unizen/.local/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/admin-unizen/esp/esp-idf/requirements.txt (line 10)) (0.24.0)
Requirement already satisfied: ipaddress; python_version < "3" in /usr/lib/python2.7/dist-packages (from cryptography>=2.1.4->-r /home/admin-unizen/esp/esp-idf/requirements.txt (line 10)) (1.0.16)
Requirement already satisfied: pycparser in /home/admin-unizen/.local/lib/python2.7/site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r /home/admin-unizen/esp/esp-idf/requirements.txt (line 10)) (2.19)
Installing collected packages: cryptography
Successfully installed cryptography-2.6.1
admin-unizen@adminunizen-Lenovo-ideapad-520-15IKB:~/esp-who/examples/single_chip/camera_web_server$ make flash
Toolchain path: /home/admin-unizen/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
The following Python requirements are not satisfied:
cryptography>=2.1.4
[... cut ...] rogramming Guide for setting up the required packages.
Alternatively, you can run "/usr/bin/python -m pip install --user -r /home/admin-unizen/esp/esp-idf/requirements.txt" for resolving the issue.
/home/admin-unizen/esp/esp-idf/make/project.mk:500: recipe for target 'check_python_dependencies' failed
make: *** [check_python_dependencies] Error 1

even after i update the cryptography i fall up with same error.if anyone knows how to solve it send all the commands so that i can follow accordingly.

sheldon123z, 2019-07-09:
Hi, 
I have similar problem while setup the python packages as the manual instructed. Here is what I got from the msys terminal:

shzheng@NB0013671 MINGW32 ~
$ python -m pip install --user -r $IDF_PATH/requirements.txt
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python                     2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Requirement already satisfied: setuptools in c:/msys32/mingw32/lib/python2.7/site-packages (from -r C:\msys32\home                    \shzheng\esp\esp-idf/requirements.txt (line 4)) (36.6.0.post20171017)
Requirement already satisfied: pyserial>=3.0 in c:/msys32/mingw32/lib/python2.7/site-packages (from -r C:\msys32\h                    ome\shzheng\esp\esp-idf/requirements.txt (line 8)) (3.4)
Requirement already satisfied: future>=0.15.2 in c:/msys32/home/shzheng/.local/lib/python2.7/site-packages (from -                    r C:\msys32\home\shzheng\esp\esp-idf/requirements.txt (line 9)) (0.17.1)
Collecting cryptography>=2.1.4 (from -r C:\msys32\home\shzheng\esp\esp-idf/requirements.txt (line 10))
  Using cached https://files.pythonhosted.org/packages/c2/95/f43d02315f4ec074219c6e3124a87eba1d2d12196c2767fadfdc0                    7a83884/cryptography-2.7.tar.gz
  Installing build dependencies ... error
  ERROR: Complete output from command C:/msys32/mingw32/bin/python.exe C:/msys32/mingw32/lib/python2.7/site-packages/pip install --ignore-installed --no-user --prefix c:/users/shzheng/appdata/local/temp/pip-build-env-cmbot3/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'setuptools>=40.6.0' wheel 'cffi>=1.8,!=1.11.3; python_implementation != '"'"'PyPy'"'"'':
  DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
  Collecting setuptools>=40.6.0
    Using cached https://files.pythonhosted.org/packages/ec/51/f45cea425fd5cb0b0380f5b0f048ebc1da5b417e48d304838c02d6288a1e/setuptools-41.0.1-py2.py3-none-any.whl
  Collecting wheel
    Using cached https://files.pythonhosted.org/packages/bb/10/44230dd6bf3563b8f227dbf344c908d412ad2ff48066476672f3a72e174e/wheel-0.33.4-py2.py3-none-any.whl
  Collecting cffi!=1.11.3,>=1.8
    Using cached https://files.pythonhosted.org/packages/93/1a/ab8c62b5838722f29f3daffcc8d4bd61844aa9b5f437341cc890ceee483b/cffi-1.12.3.tar.gz
  Collecting pycparser (from cffi!=1.11.3,>=1.8)
  Installing collected packages: setuptools, wheel, pycparser, cffi
    Running setup.py install for cffi: started
      Running setup.py install for cffi: finished with status 'error'
      ERROR: Complete output from command C:/msys32/mingw32/bin/python.exe -u -c 'import setuptools, tokenize;__file__='"'"'c:/users/shzheng/appdata/local/temp/pip-install-nzydnc/cffi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record c:/users/shzheng/appdata/local/temp/pip-record-gvaqxk/install-record.txt --single-version-externally-managed --prefix c:/users/shzheng/appdata/local/temp/pip-build-env-cmbot3/overlay --compile:
      ERROR:
          No working compiler found, or bogus compiler options passed to
          the compiler from Python's standard "distutils" module.  See
          the error messages above.  Likely, the problem is not related
          to CFFI but generic to the setup.py of any Python package that
          tries to compile C code.  (Hints: on OS/X 10.8, for errors about
          -mno-fused-madd see http://stackoverflow.com/questions/22313407/
          Otherwise, see https://wiki.python.org/moin/CompLangPython or
          the IRC channel #python on irc.freenode.net.)

          Trying to continue anyway.  If you are trying to install CFFI from
          a build done in a different context, you can ignore this warning.

      running install
      running build
      running build_py
      creating build
      creating build/lib.mingw-2.7
      creating build/lib.mingw-2.7/cffi
      copying cffi/api.py -> build/lib.mingw-2.7/cffi
      copying cffi/backend_ctypes.py -> build/lib.mingw-2.7/cffi
      copying cffi/cffi_opcode.py -> build/lib.mingw-2.7/cffi
      copying cffi/commontypes.py -> build/lib.mingw-2.7/cffi
      copying cffi/cparser.py -> build/lib.mingw-2.7/cffi
      copying cffi/error.py -> build/lib.mingw-2.7/cffi
      copying cffi/ffiplatform.py -> build/lib.mingw-2.7/cffi
      copying cffi/lock.py -> build/lib.mingw-2.7/cffi
      copying cffi/model.py -> build/lib.mingw-2.7/cffi
      copying cffi/pkgconfig.py -> build/lib.mingw-2.7/cffi
      copying cffi/recompiler.py -> build/lib.mingw-2.7/cffi
      copying cffi/setuptools_ext.py -> build/lib.mingw-2.7/cffi
      copying cffi/vengine_cpy.py -> build/lib.mingw-2.7/cffi
      copying cffi/vengine_gen.py -> build/lib.mingw-2.7/cffi
      copying cffi/verifier.py -> build/lib.mingw-2.7/cffi
      copying cffi/__init__.py -> build/lib.mingw-2.7/cffi
      copying cffi/_cffi_include.h -> build/lib.mingw-2.7/cffi
      copying cffi/parse_c_type.h -> build/lib.mingw-2.7/cffi
      copying cffi/_embedding.h -> build/lib.mingw-2.7/cffi
      copying cffi/_cffi_errors.h -> build/lib.mingw-2.7/cffi
      running build_ext
[... cut ...] de 1 in c:/users/shzheng/appdata/local/temp/pip-install-nzydnc/cffi/
  
ERROR: Command "C:/msys32/mingw32/bin/python.exe C:/msys32/mingw32/lib/python2.7/site-packages/pip install --ignore-installed --no-user --prefix c:/users/shzheng/appdata/local/temp/pip-build-env-cmbot3/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'setuptools>=40.6.0' wheel 'cffi>=1.8,!=1.11.3; python_implementation != '"'"'PyPy'"'"''" failed with error code 1 in None

dobairoland, 2019-07-09:
Hi @sheldon123z.

The `cryptography` package cannot  be installed by `pip` on MSYS2. You need to install it with `pacman`:
```
pacman -S mingw-w64-i686-python2-cryptography
```
If you have an old MSYS2 environment then you need to update the packages:
```
pacman -Syu
```
The easiest way of solving this issue is to download the newest version of ESP MSYS2 enviroment which contains everything ready-to-use:
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html#toolchain-setup

noyelseth, 2020-01-29:
Unable to find **mingw-w64-i686-python2-cryptography** target using pacman

[image]

projectgus, 2020-01-29:
@noyelseth It looks like MSYS2 has removed these Python 2 packages.

The temporary workaround for this is to download the latest ESP-IDF environment .zip file image from the IDF Getting Started guide for your ESP-IDF version. This zip file will already have this package installed.

As a long term fix, we'll be updating the instructions for current versions of ESP-IDF to use Python 3 instead, ASAP (this is supported now on ESP-IDF v3.3 and newer but all the instructions assume Python 2).

tuzhimin, 2020-05-08:
ERROR: Complete output from command D:/ESP32/msys32/mingw32/bin/python.exe D:/ESP32/msys32/mingw32/lib/python2.7/site-packages/pip install --ignore-installed --no-user --prefix 'c:/users/admini~1/appdata/local/temp/pip-build-env-yjinsr/overlay' --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'setuptools >= 40.8.0' wheel 'Cython >= 0.29.14' 'cffi >= 1.12.3 ; platform_python_implementation == '"'"'CPython'"'"'' 'greenlet>=0.4.14 ; platform_python_implementation == '"'"'CPython'"'"'':
  DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
  Collecting setuptools>=40.8.0
    Using cached https://files.pythonhosted.org/packages/ab/b5/3679d7c98be5b65fa5522671ef437b792d909cf3908ba54fe9eca5d2a766/setuptools-44.1.0-py2.py3-none-any.whl
  Collecting wheel
    Using cached https://files.pythonhosted.org/packages/8c/23/848298cccf8e40f5bbb59009b32848a4c38f4e7f3364297ab3c3e2e2cd14/wheel-0.34.2-py2.py3-none-any.whl
  Collecting Cython>=0.29.14
    Using cached https://files.pythonhosted.org/packages/9b/24/3416df8ae5962b09f6b522ad6d61e711dadb57816a6e4e76c3b99732f3c5/Cython-0.29.17-py2.py3-none-any.whl
  Collecting cffi>=1.12.3
    Using cached https://files.pythonhosted.org/packages/05/54/3324b0c46340c31b909fcec598696aaec7ddc8c18a63f2db352562d3354c/cffi-1.14.0.tar.gz
  Collecting greenlet>=0.4.14
    Using cached https://files.pythonhosted.org/packages/f8/e8/b30ae23b45f69aa3f024b46064c0ac8e5fcb4f22ace0dca8d6f9c8bbe5e7/greenlet-0.4.15.tar.gz
  Collecting pycparser (from cffi>=1.12.3)
    Using cached https://files.pythonhosted.org/packages/ae/e7/d9c3a176ca4b02024debf82342dab36efadfc5776f9c8db077e8f6e71821/pycparser-2.20-py2.py3-none-any.whl
  Installing collected packages: setuptools, wheel, Cython, pycparser, cffi, greenlet
    Running setup.py install for cffi: started
      Running setup.py install for cffi: finished with status 'error'
      ERROR: Complete output from command D:/ESP32/msys32/mingw32/bin/python.exe -u -c 'import setuptools, tokenize;__file__='"'"'c:/users/admini~1/appdata/local/temp/pip-install-_g2_h4/cffi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'c:/users/admini~1/appdata/local/temp/pip-record-fwdun9/install-record.txt' --single-version-externally-managed --prefix 'c:/users/admini~1/appdata/local/temp/pip-build-env-yjinsr/overlay' --compile:
      ERROR:
          No working compiler found, or bogus compiler options passed to
          the compiler from Python's standard "distutils" module.  See
          the error messages above.  Likely, the problem is not related
          to CFFI but generic to the setup.py of any Python package that
          tries to compile C code.  (Hints: on OS/X 10.8, for errors about
          -mno-fused-madd see http://stackoverflow.com/questions/22313407/
          Otherwise, see https://wiki.python.org/moin/CompLangPython or
          the IRC channel #python on irc.freenode.net.)

          Trying to continue anyway.  If you are trying to install CFFI from
          a build done in a different context, you can ignore this warning.

      running install
      running build
      running build_py
      creating build
      creating build/lib.mingw-2.7
      creating build/lib.mingw-2.7/cffi
      copying cffi/api.py -> build/lib.mingw-2.7/cffi
      copying cffi/backend_ctypes.py -> build/lib.mingw-2.7/cffi
      copying cffi/cffi_opcode.py -> build/lib.mingw-2.7/cffi
      copying cffi/commontypes.py -> build/lib.mingw-2.7/cffi
      copying cffi/cparser.py -> build/lib.mingw-2.7/cffi
      copying cffi/error.py -> build/lib.mingw-2.7/cffi
      copying cffi/ffiplatform.py -> build/lib.mingw-2.7/cffi
      copying cffi/lock.py -> build/lib.mingw-2.7/cffi
      copying cffi/model.py -> build/lib.mingw-2.7/cffi
      copying cffi/pkgconfig.py -> build/lib.mingw-2.7/cffi
      copying cffi/recompiler.py -> build/lib.mingw-2.7/cffi
      copying cffi/setuptools_ext.py -> build/lib.mingw-2.7/cffi
      copying cffi/vengine_cpy.py -> build/lib.mingw-2.7/cffi
      copying cffi/vengine_gen.py -> build/lib.mingw-2.7/cffi
      copying cffi/verifier.py -> build/lib.mingw-2.7/cffi
      copying cffi/__init__.py -> build/lib.mingw-2.7/cffi
      copying cffi/_cffi_include.h -> build/lib.mingw-2.7/cffi
      copying cffi/parse_c_type.h -> build/lib.mingw-2.7/cffi
      copying cffi/_embedding.h -> build/lib.mingw-2.7/cffi
      copying cffi/_cffi_errors.h -> build/lib.mingw-2.7/cffi
      running build_ext
      building '_cffi_backend' extension
      creating build/temp.mingw-2.7
      creating build/temp.mingw-2.7/c
      i686-w64-mingw32-gcc -fno-strict-aliasing -march=i686 -mtune=generic -O2 -pipe -fwrapv -D__USE_MINGW_ANSI_STDIO=1 -DNDEBUG -DNDEBUG -I/usr/include/ffi -I/usr/include/libffi -ID:/ESP32/msys32/mingw32/include/python2.7 -c c/_cffi_backend.c -o build/temp.mingw-2.7/c/_cffi_backend.o
      error: command 'i686-w64-mingw32-gcc' failed: No such file or directory
      ----------------------------------------
  ERROR: Command "D:/ESP32/msys32/mingw32/bin/python.exe -u -c 'import setuptools, tokenize;__file__='"'"'c:/users/admini~1/appdata/local/temp/pip-install-_g2_h4/cffi/setup.py'"'"';f=getattr(tokenize, '" [... cut ...]
ERROR: Command "D:/ESP32/msys32/mingw32/bin/python.exe D:/ESP32/msys32/mingw32/lib/python2.7/site-packages/pip install --ignore-installed --no-user --prefix 'c:/users/admini~1/appdata/local/temp/pip-build-env-yjinsr/overlay' --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'setuptools >= 40.8.0' wheel 'Cython >= 0.29.14' 'cffi >= 1.12.3 ; platform_python_implementation == '"'"'CPython'"'"'' 'greenlet>=0.4.14 ; platform_python_implementation == '"'"'CPython'"'"''" failed with error code 1 in None

help me，Jesus， i do‘t no do it，

amardhore, 2020-06-26:
I am trying to setup esp-idf with msys32. I did it before few times but this time I am having so much trouble. My setup is windows 10 fresh installation. Initially it failed on python -m pip install --user -r $IDF_PATH/requirements.txt.  I followed the instructions mentioned above but it fails while installing dependencies. Why this installing esp-idf process is not consistent and it does not always work. I am working on some research project and I do not know how to proceed from here. I might move away from esp32 if can not resolve it.  Any help would very much appreciated. This is my output in the attached filet:

[msys_log.txt](https://github.com/espressif/esp-idf/files/4838077/msys_log.txt)

projectgus, 2020-06-29:
Hi @amardhore,

What ESP-IDF version are you using?

Angus

amardhore, 2020-06-29:
@projectgus It's ESP-IDF 4.2.

projectgus, 2020-06-30:
Hi @amardhore ,

The MSYS2 legacy environment for v4.2 currently has some issues, as you've found. We were using Python 2 in these images but MSYS2 Python 2 packages are no longer available, so it's not possible to install any new python modules.

We plan to fix this before the v4.2 release, and will release a new MSYS2 precompiled image that uses Python 3.

There are some options to get started ASAP:

* [Switch to the current stable release v4.0.1](https://docs.espressif.com/projects/esp-idf/en/v4.0.1/get-started-legacy/windows-setup.html) and use the latest MSYS2 legacy environment from here.
* [Switch to the CMake-based build system which was made the default in ESP-IDF v4.0](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/)
* Wait for the new MSYS2 environments to be available, but this may take a week or two.

a-dhore, 2020-06-30:
@projectgus  I updated esp-idf to V4.0 and it worked. Appreciate your help.

amardhore, 2020-06-30:
@projectgus I updated esp-idf to V4.0 and it worked. Appreciate your help.