espressif/esp-idf #2481: MESH: When root loses connection with router it stays root...

edjeed, 2018-09-27:
ESP-MESH

When root loses connection with router it stays root. 
Is it possible for the root to give up the root status and start a new vote when Root loses connection with the router?

dongdong004, 2018-09-28:
Yes, it is possible.
* Use `esp_mesh_waive_root()` on root , all nodes will start vote.
* Use `esp_mesh_set_self_organized(true,true)` on root, root will give up the root status and start scan.

edjeed, 2018-09-29:
I was calling `esp_mesh_waive_root()`, vote started but nothing happened when vote finished. 
The docs states that `esp_mesh_set_self_organized` is enabled by default so I didn't call it.
I'll try this out, thanx again @dongdong004

Alvin1Zhang, 2018-10-09:
@edjeed Do you have any updates? Have your issue been resolved? If yes, would you please help close this issue? Thanks.

edjeed, 2018-10-09:
Issue is solved, but a small update to the documentation would make it more understandable.

"
To trigger a root node switch, the current root node must explicitly call esp_mesh_waive_root() to trigger a new election. The current root node will signal all nodes within the network to begin transmitting and scanning for beacon frames (see Automatic Root Node Selection) whilst remaining connected to the network (i.e. not idle). If another node receives more votes than the current root node, a root node switch will be initiated. The root node will remain unchanged otherwise.
"

I'm missing the esp_mesh_set_self_organized function in this context.

ESP-MESH-QL, 2018-10-13:
> Is it possible for the root to give up the root status and start a new vote when Root loses connection with the router?

When Root loses connections with the router, Root will always try to connect to the router until the connection is successful. But if you do really want to switch Root, esp_mesh_waive_root() can really help you achieve. If you found _vote started but nothing happened when vote finished_, that might be caused by the attempts you tried is not enough. The default attempts is 15. If the router can be scanned by at least one device and the attempts is enough, then Root switch will be realized.

ps: The number of attempts taken depends on whether there is a root node with a clearer advantage in the current network. 
> Issue is solved, but a small update to the documentation would make it more understandable.

The vote initiated by esp_mesh_waive_root()  is based on the whole network without being destroyed. While esp_mesh_set_self_organized(true, true) is equal to current Root breaks down, we donot recommend this approach, that's why we didn't mention this meshod in the docs.

Alvin1Zhang, 2018-10-15:
@edjeed Please help check our reply on the documentation you mentioned. Thanks.

edjeed, 2018-10-15:
If have tested the esp_mesh_waive_root(), by moving the current root-device far away from the router. Another mesh device is close to the router, after 20 vote attempts, root switch is not being started.

esp_mesh_set_self_organized(true, true), kills the whole current mesh so I agree it's not recommended, but should be documented,

closed 2019-02-14 by Alvin1Zhang

Robokishan, 2019-04-16:
how to use esp_mesh_waive_root it isn't working for me i just called 
esp_mesh_waive_root(NULL,MESH_VOTE_REASON_ROOT_INITIATED)
the root is selected but the previous root stays at it is . and the elected root does not convert into the node it just stays as the node. 
getting error :

E (592027) event: invalid static ip
W (592027) wifi: [beacon]interval:300ms
I (592029) mesh: write layer:1
I (592029) [mwifi, 66]: Parent is connected
I (592034) [mwifi_examples, 347]: event_loop_cb, event: 7
I (592039) [mwifi_examples, 355]: Parent is connected on station interface
I (592047) [mwifi_examples, 347]: event_loop_cb, event: 11
I (592053) [mwifi_examples, 347]: event_loop_cb, event: 14

and mesh doesn't change even the previous root is in next room.
and the elected root is beside the router.
thank you