espressif/esp-idf #2483: IO mux & GPIO mux config confusion

michprev, 2018-09-27:
## Environment
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 35785d08f

## Problem Description

I know that implementation of GPIO and IO mux on ESP32 is not simple but ESP-IDF makes even bigger mess in it.
From my understanding these are necessary steps to correctly setup ESP32 pin:

1. configure IO mux to assign one specific function to the pin (for example for pin 12: MTDI / HSPIQ / GPIO12)
2. in case that selected function is GPIO and you don't want to use the pin as GPIO you have to either assign one peripheral signal to the pin or assign the pin to one or more peripheral signals.

But looking at https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/gpio.html I am really confused.

`PIN_FUNC_SELECT` seems to be the option for step 1. But it is not even mentioned on the docs page.
`gpio_iomux_in` and `gpio_iomux_out` should be the option for step 2. But I am really confused by `gpio_iomux_out` function. It should take as an argument one of `*_OUT_IDX` signals from `soc/gpio_sig_map.h` file.

Also it should be clear at first sight what options you can set while using IO mux and what options to set while using GPIO mux.
Looking into TRM:
1. you can always set drive strength, input enable option, pullup/pulldown function (options in `IO_MUX_x_REG`)
2. if you set GPIO as the function of the pin you can additionaly set interrupt type, interrupt enable, wakeup enable and pad driver (options in `GPIO_PINn_REG`)

Looking at `gpio_config` function you are setting pullup/pulldown options and interrupt options in the same config struct. But interrupt configuration makes sense only while not using IO mux directly.

negativekelvin, 2018-09-27:
gpio_iomux_in and gpio_iomux_out are for step 1
gpio_matrix_in and gpio_matrix_out should are for step 2

gpio_iomux_in is confusing because one would expect PIN_FUNC_SELECT to configure both input and output and it is only ever used a few times in spi driver

Alvin1Zhang, 2019-03-21:
@michprev Hi, would you help share if any updates for the issue? Thanks.

Alvin1Zhang, 2019-03-27:
@michprev @negativekelvin Feel free to reopen the issue. Thanks.

EvgenyMuryshkin, 2025-07-09:
What is the resolution? gpio_matrix_in and gpio_matrix_out are no longer available in gpio.h (IDF 5.4.1).
What is the latest way of connecting things? 

I am currently trying to implement a direct connection between two pins via mux and no luck so far.

Thanks.

lnavlag1, 2025-07-17:
> What is the resolution? gpio_matrix_in and gpio_matrix_out are no longer available in gpio.h (IDF 5.4.1). What is the latest way of connecting things?
> 
> I am currently trying to implement a direct connection between two pins via mux and no luck so far.
> 
> Thanks.

Here!: https://esp32.com/viewtopic.php?t=19099