espressif/esp-idf #2484: How to use newlib's iconv on ESP32?

TridentTD, 2018-09-27:
I want to use newlib's libiconv for converting utf8 to iso-8859-11.
Have any example to using iconv on Arduino-ESP32 or ESP-IDF?

Thank you.

Alvin1Zhang, 2019-03-18:
@TridentTD Hi, sorry that get back to you so late. Does your issue still exist? Thanks.

TridentTD, 2019-03-18:
@Alvin1Zhang 
Yes. Is there any example how to use iconv on ESP32?

JaeHoLee1811, 2019-10-08:
this issue still problem?? cuz there is no solution this issue... help me :((
I used to four days to this problem...

allenm, 2019-12-16:
+1 ，need help

antmak, 2019-12-17:
Now the toolchain was built without the iconv support. No official support for iconv now.
 You can turn on that in https://github.com/espressif/crosstool-NG/blob/esp-2019r2/samples/xtensa-esp32-elf/crosstool.config and build it then as described https://docs.espressif.com/projects/esp-idf/en/latest/get-started/linux-setup-scratch.html

antmak, 2019-12-17:
Still no official iconv support in the toolchain and IDF's examples.

antmak, 2019-12-17:
Still no official iconv support in both the toolchain and IDF.

You can try build your code which uses iconv.h with https://dl.espressif.com/dl/toolchains/preview/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-41-g8ac0429-linux-amd64.tar.xz (iconv support enabled ) and give any feedback here.

Probably the support for iconv.h will be added to next release of the toolchain release (2020r1)

gw12266, 2024-05-14:
请问，解决了吗，如何在esp32上使用iconv