espressif/esp-idf #2485: Can Driver

snahmad, 2018-09-28:
Is CAN driver included in Version 3.2. I cannot find in 3.1 stable release.

negativekelvin, 2018-09-28:
The commit branch information shows it was only merged to master https://github.com/espressif/esp-idf/commit/1d2727f4c8f9a6ef5fffd8e7f2dd86500cc8284b#diff-78a55ecd614cdfc8d8f22e23faac4609

snahmad, 2018-10-05:
Can i use CAN driver only files with version 3.1 which is last stable release.

KrrishnaShah, 2018-12-03:
Is can driver removed from 3.1.1 ?

snahmad, 2018-12-03:
I think it is only available in version=3.2 Master branch.

giordano-cm, 2019-03-04:
https://github.com/espressif/esp-idf/issues/3124

I'm using PlatformIO and it seems that the v3.1.3 doesn't have the CAN driver. And now what?

Dazza0, 2019-03-05:
@snahmad If you need to use CAN driver in release 3.1, you'll need to branch off release 3.1 and add the CAN driver related changes yourself. The changes are as follows:

- Add `can.c` to the [components/driver directory](https://github.com/espressif/esp-idf/tree/v3.1.3/components/driver)
- Add `can.h` to the [components/driver/include/driver directory](https://github.com/espressif/esp-idf/tree/v3.1.3/components/driver/include/driver)
- Add `can_struct.h` to the [components/soc/esp32/include/soc directory](https://github.com/espressif/esp-idf/tree/v3.1.3/components/soc/esp32/include/soc)
- Add `#define DR_REG_CAN_BASE                         0x3ff6B000` to [`soc.h`](https://github.com/espressif/esp-idf/blob/cf5dbadf4f25b395887238a7d4d8251c279afa8c/components/soc/esp32/include/soc/soc.h#L122)
- Add `PROVIDE ( CAN = 0x3ff6B000 );` to [`esp32.peripherals.ld`](https://github.com/espressif/esp-idf/blob/cf5dbadf4f25b395887238a7d4d8251c279afa8c/components/esp32/ld/esp32.peripherals.ld#L29)

snahmad, 2019-03-05:
@Dazza0  Thanks for reply. late reply. I am happy with master branch release 3.2. working for me.

closed 2019-03-21 by Alvin1Zhang