espressif/esp-idf #2487: rmt_isr_deregister in rmt_driver_uninstall always returns ESP_ERR_INVALID_ARG

davctv, 2018-09-28:
## Environment

- Development Kit:      [none]

- Core (if using chip or module): [ESP-WROOM32|]
- IDF version (`85a01e9fc064ef1eb01070be11cfca4698bfdb0e) esp-v3.1

- Development Env:		[Eclipse]
- Operating System:		[Windows]
- Power Supply:         [external 3.3V]

rmt_driver_uninstall should return ESP_OK but always returns ESP_ERR_INVALID_ARG.

The error comes from rmt_isr_deregister function which calls esp_intr_free() function which return ESP_ERR_INVALID_ARG.

The following code of the rmt_driver_uninstall is not executed and this creates problem with resource release.

Thanks

xiongyumail, 2018-09-29:
Hi @davctv 
Returning ESP_ERR_INVALID_ARG is generally an input parameter error, you can look into the RMT code to see which parameter is wrong.

negativekelvin, 2018-09-29:
This may happen if rmt_driver_uninstall runs on a different core than rmt_driver_install if your task has no core affinity

davctv, 2018-09-29:
> Returning ESP_ERR_INVALID_ARG is generally an input parameter error, you can look into the RMT code to see which parameter is wrong.

@xiongyumail the RMT_CHECK macro at first line of rmt_driver_uninstall are correctly passed and I have checked the return value is from rmt_isr_deregister.

@negativekelvin I use rmt_driver_install in a specific "rmt" task which I create when program start-up.
I use rmt_diver_uninstall from main task. Should I call it from the "rmt"?
Or maybe I have to call rmt_driver_install from main task before create "rmt" task?
I will do some test and maybe I will report my test code.

Thanks.

stickbreaker, 2018-09-29:
@negativekelvin  Maybe `esp_intr_free()` should be changed to automatically use `esp_ipc_call_blocking()` to switch to the other core if the `free()` call fails because of processor affinity?

https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/ipc.html

Chuck.

davctv, 2018-10-01:
@negativekelvin @stickbreaker 

I have done some test and you can check here:

https://github.com/davctv/test_ir

If rmt_drive_install and rmt_driver_uninstall are called from MAIN task everything works.

If rmt_driver_install is called in another task and then rmt_driver_uninstall is called from MAIN task the error comes out. I will modify my code to do this, but it would be better if rmt driver can be uninstalled from different task (and probably different core).

Thanks.

koobest, 2018-10-15:
Hi, @davctv, Does the last master solve your problem?

davctv, 2018-10-15:
Hello @koobest,
No I didn't test on master. We are going to production and we are on v3.1. I have solved calling "rmt_drive_install" and "rmt_driver_uninstall" are called from MAIN task.

However I will do some test to eventually close the issue.

Thanks

Alvin1Zhang, 2018-10-24:
@davctv Could you help share if any updates for this issue? Thanks.

davctv, 2018-10-24:
Hello @Alvin1Zhang,

I confirm that with latest master (129d327) the issue is solved. RMT Driver can be installed in another  task and it can be uninstalled from another task.

On release v3.1 you have to install/uninstall driver from the same task.

I think the issue can be closed.

Regards

closed 2018-10-24 by davctv

Alvin1Zhang, 2018-10-24:
@davctv Thanks for the support.