espressif/esp-idf #2488: tcpip_adapter_init does not pass error codes to caller

TobleMiner, 2018-09-28:
The method [tcpip_adapter_init](https://github.com/espressif/esp-idf/blob/11b444b8f493165eb4d93f44111669ee46be0327/components/tcpip_adapter/tcpip_adapter_lwip.c#L98) does neither abort execution on error, nor pass error codes to the caller.

I'd suggest either changing the method to a `esp_err_t tcpip_adapter_init(void)` or if that is not possible for compatibility reasons adding a new method with above signature that returns errors that occured during execution to the caller.

At the time being it is impossible for the caller to determine if tcpip_adapter_init worked. IMHO that's pretty insane and makes witting reliable code for the ESP32 a lot harder than it should be.

liuzfesp, 2018-10-09:
Thanks @TobleMiner for pointing this, we will optimize it.

TobleMiner, 2018-10-09:
That's nice to hear. Thanks for taking care of it.

david-cermak, 2019-11-25:
Hi @TobleMiner 

in IDF v4.1 the tcpip_adapter is reworked into a new network interface library called [esp-netif](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/esp_netif.html), which returns an error code if the initialisation fails: https://github.com/espressif/esp-idf/blob/master/components/esp_netif/include/esp_netif.h#L54

Alvin1Zhang, 2019-11-29:
@TobleMiner Thanks for reporting the issue, the issue has been fixed in https://github.com/espressif/esp-idf/commit/ffe043b1a81a0f9e1cc2cfa8873e21318ec89143, feel free to reopen if needed. Thanks.

closed 2019-11-29 by Alvin1Zhang