espressif/esp-idf #2489: Passive background BLE scanning + WiFi connected => WD reset

jakkra, 2018-09-28:
## Environment

- Development Kit:      [none]
- Core (if using chip or module): [ESP32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    430eeef5a25209d9a01e75140bf82d79498ffd51 with this patch: https://github.com/espressif/esp-idf/issues/2337#issuecomment-418038595
- Development Env:		[Make]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description
After a few minutes if background BLE passive scanning while connected to a WiFi cause a watchdog reset with the following stacktrace.

### Expected Behavior
No Crash/WD reset
### Actual Behavior
WD reset and crash
### Steps to repropduce

1. Connect WiFi
2. Do BLE passive background scanning continiously

## Debug Logs
```
0x40094a88: invoke_abort at esp-idf/components/esp32/panic.c:653
0x40094c2b: abort at esp-idf/components/esp32/panic.c:653
0x400d24f3: task_wdt_isr at esp-idf/components/esp32/task_wdt.c:236
0x40082581: _xt_lowint1 at esp-idf/components/freertos/xtensa_vectors.S:1105
0x4008e02f: coex_force_wifi_mode at ??:?
```

## Other items if possible

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/2428531/sdkconfig.zip)

jkhax0r, 2018-10-01:
I'm having same issue.  @jakkra Is there a work-around?  You specifically mentioned passive scanning - does active scanning fix it?

jakkra, 2018-10-02:
@jkhax0r I have not tried to workaround it, maybe turning it on and off again once per minute or something. I would guess the problems is there for active scanning too.

lemonatorindustries, 2018-10-03:
I have the same issue. I start BLE scan for 60 sec, the stop BLE and start Wifi and do http POST, disconnect Wifi and reenable BLE to start BLE scan again. Crash always occurs within 24 hours. WiFi and BLE are never enabled simultaneously. Still crashing. Strange behavior. Anybody got a clue?

jakkra, 2018-10-08:
I did some more testing and issue still remains after disabling BLE passive scanning. 
Doing WiFi scan and afterwards classic Inquiry repeatedly also triggers the same problem.

TianHao-Espressif, 2018-10-09:
@jakkra , I fixed the bugs of watchdog timeout one week ago, but there's some other problems. So I posted another MR internal to fix the problem completely and it's in testing. It will be merged ASAP. Please wait for several days. Thanks.

TianHao-Espressif, 2018-10-09:
@jakkra ， if you need it urgent, I can provide you the patch here.
Please based on release/v3.1 branch (commit f11ac037c4958fc68e67a50134a0ce9a1634008c).
Put libbtdm_app.a into esp-idf/components/bt/lib and other libraries into esp-idf/component/esp32/lib.

You will see the bt library version is **7b0770a** and wifi version is **06a8362**. For further, the coexist version is **v1.1.0**, if you call esp_coex_version_get().

Any problem, please notice me, thanks.
[patch_for_coex_wdt_bug.zip](https://github.com/espressif/esp-idf/files/2458739/patch_for_coex_wdt_bug.zip)

jakkra, 2018-10-09:
Hi @TianHao-Espressif, Unfortunately that fix did not make a difference. 
We did try to switch core of BT Controller, and then we get this printout just before the crash:
`ASSERT_ERR((EM_BT_RXDESC_OFFSET + ld_env.curr_rxdesc_index * REG_EM_BT_RXDESC_SIZE) == bt_et_currentrxdescptr_currentrxdescptr_getf()), in ld_inq.c at line 278`

TianHao-Espressif, 2018-10-10:
@jakkra ， could you check the version information is correct?

It's not the same error as you mentioned firstly in this issue.

jakkra, 2018-10-10:
Hi @TianHao-Espressif It seems as the fix did fix the WD crash. However now instead we get this crash instead: 

```
0x40084510: r_assert_err at ??:?
0x4003b0ae: ?? ??:0
0x40085f0a: r_ld_fm_frame_isr at ??:?
0x400884be: r_rwbt_isr at ??:?
0x40088ffb: r_rwbtdm_isr_wrapper at intc.c:?
0x40082621: _xt_lowint1 at C:/Git/nina-esp/component/cb_rtsl_esp32/esp-idf/components/freertos/xtensa_vectors.S:1105
0x400891c5: btdm_bb_isr at intc.c:?
```

TianHao-Espressif, 2018-10-10:
@jakkra , Thanks for feedback.

 Please could you  tell me which BLE or classic bt  is running when crash. As this issues's title, you do BLE scanning + WIFI, but this crash indicatesa a classic bt problem. So your information is very important. If it's sure that it's doing BLE scanning, then crash happens, the debug direct need to be changed.

jakkra, 2018-10-10:
@TianHao-Espressif There where two scenarios where we got the original WD crash, one was when wifi connected and ble passive scan (Described above). The other scenario (not mentioned here) was doing inquiry, then discovery and then wifi scanning. I initialy tested your patch only with scenario two, which is why the confusion. 

So to conclude, wifi connected + BLE passive scanning works now. But in scenario two a new issue rised, after the WD crash was fixed. Do you want me to open a new issue for scenario two?

Scenario two is: 
1. WiFi scan for 5s
2. Classic inquiry 10s
3. LE Discovery 10s
4. Go to one.

TianHao-Espressif, 2018-10-15:
@jakkra , thanks. Could open another issues to take about this? So other people will know the Watchdog bug is fixed. Thanks.

Milan991, 2018-10-15:
@TianHao-Espressif When you say that Watchdog timer bug is fixed, are you saying about patch or some new commit that was pushed regarding this issue? 
Thank you in advance.

Milan991, 2018-10-19:
@TianHao-Espressif Great, thank you. Please let me know when that is done.

angryDuck2, 2018-10-27:
I am getting this error when I switch the files suggested
```
E (201) wifi: invalid magic number 3ffd6ad0, call WIFI_INIT_CONFIG_DEFAULT to init config
E (201) wifi: wifi_init_in_caller_task 707 ret=258
```

xielizyh, 2018-10-30:
@TianHao-Espressif , According your sugesstion, I used the the newest commit on Oct 26, 2018, but the WDT problem is reappear after a few hours. I started that ble is active scan continuously and wifi connected.

TianHao-Espressif, 2018-12-03:
H， this bug(Task Watch dog timerout in wifi task) is already fixed in ccddd3bc0198bf80809042d1c6cdd3838d9d107d. Please cose the issues. Any other problem, please create a new issues, thanks.

TianHao-Espressif, 2018-12-03:
@xielizyh ， please provide the crash log. I mean the problem which is firstly mentioned in this issues is fixed. I your problem is a new bug, please crate an new issue. Thanks.

closed 2018-12-03 by Alvin1Zhang