espressif/esp-idf #2491: Feature request: Improved/additional info in coredump

ammaree, 2018-09-28:
I strongly believe that the absolute majority of ESP32 based devices will be installed at remote locations without any local console attached.
I also believe that the long term success of the ESP32 as a platform will very much depend on the ability of  developers to create highly reliable and serviceable products and/or services built on the ESP32.
To achieve this required very high level of reliability will depend on whether developers are able to consistent capture at remote locations ALL the info required to effectively identify the cause of a crash.

The current mechanism of coredumps is already of great value but as we know certain critical info is missing from the coredump. What I suggest is that the functionality be extended to:
a) include the diagnostic info (reset cause, core/task info etc) that is currently dumped to the UART;
b) possibly add another segment attribute CRASH_ATTR to indicate portions of memory to be included in the coredump;
c) add the capability to the coredump decoder to identify threads, queues/semaphores and timers by their names in the text output.

I am sure there are many more and better suggestions that can/will come from more skilled developers, but further improving the ability to remotely identify the cause of a crash will significantly improve the adoption of the ESP32, an already superb product with a well supported SDK and an incredible team.

Making this enhanced crash capturing and analysis functionality a priority higher than adding additional features will, I believe, make it easier to support the additional new features and application in future.

igrr, 2018-09-29:
Hi @ammaree,

> a) include the diagnostic info (reset cause, core/task info etc) that is currently dumped to the UART;

Acknowledged, will look for a good way to do this.

b) possibly add another segment attribute CRASH_ATTR to indicate portions of memory to be included in the coredump;

Acknowledged, agree this would be useful

c) add the capability to the coredump decoder to identify threads, queues/semaphores and timers by their names in the text output.

Acknowledged about task names (they can be recovered from the TCB). About queues and timers, how do you propose to associate names with them?

ammaree, 2018-09-29:
Hi @igrr 

Have not looked at the FreeRTOS internals in detail but:
For queues I believe we can use the fields value added using vQueueAddToRegistry() 
For timers we could use xTimerCreate() fields "pcTimerName" and/or "pvTimerID"

It might require some indirect referencing into the timer and queue structures but should be possible?

ammaree, 2019-07-13:
Any chance of of an update on this request?

ammaree, 2019-10-28:
@gerekon Any chance of an update?

gerekon, 2019-10-29:
@ammaree There are a set of core dump improvements are about to be released. Almost all your items are in the list. (a) and threads identification mentioned in (c) should be included into the next release.

ammaree, 2019-10-30:
Any idea when we will see the initial items and also the thread/task identification released?

ammaree, 2024-02-14:
Have not checked in detail, very few crashes nowadays so coredump not playing a big role. Will reopen if it is still found to be an issue

closed 2024-02-14 by ammaree