espressif/esp-idf #2495: Softap + sta concurrent mode not work in the newest master and v3.1

outermanle, 2018-09-30:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): [v4]
- Core (if using chip or module): [ESP-WROOM-32D]
- IDF version (the master newest and v3.1):
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [USB]

## Problem Description
works in softap+station cocurrent mode.

1. two sta(phone) connect to softap; (just connected, no data transfer)
2. station connect to AP with internet access.(http post to server every 5s)

then about 1 minutes,  we get error log:
   W (743341) wifi: conn=0x3ffcf20c failed to alloc rx ba index
    W (743341) wifi: rap no hw index, tid=4

then http client get errors:
E (855185) HTTP_CLIENT: Connection failed, sock < 0
E (855185) http-client: HTTP POST request failed: [ESP_ERR_HTTP_CONNECT], err(38)

### Expected Behavior
works well in softap+sta mode.

### Actual Behavior
malfunction.

### Steps to repropduce
1. two sta(phone) connect to softap; (just connected, no data transfer)
2. station connect to AP with internet access.(http post to server every 5s)

then about 1 minutes,  we get error.

### Code to reproduce this issue

```cpp
    tcpip_adapter_init();
    ESP_ERROR_CHECK(tcpip_adapter_dhcps_stop(TCPIP_ADAPTER_IF_AP));

    tcpip_adapter_ip_info_t ip_info;
    memset(&ip_info, 0, sizeof(ip_info));
    IP4_ADDR(&ip_info.ip, 192, 168, 14, 1);
    IP4_ADDR(&ip_info.gw, 192, 168, 14, 1);
    IP4_ADDR(&ip_info.netmask, 255, 255, 255, 0);
    ESP_ERROR_CHECK(tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_AP, &ip_info));
    ESP_ERROR_CHECK(tcpip_adapter_dhcps_start(TCPIP_ADAPTER_IF_AP));

    ESP_ERROR_CHECK(esp_event_loop_init(wifi_event_handler, NULL));
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    wifi_config_t sta_config= {
        .sta = {
            .ssid = "zzzz",
            .password = "xxxxxxxx",
        }
    };
    wifi_config_t ap_config = {
        .ap = {
            .ssid = "xxxxx",
            .password = "asdfasdf",
            .ssid_len = 0,
            .max_connection = 4,
            .authmode = WIFI_AUTH_WPA_WPA2_PSK
        }
    };

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &ap_config));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &sta_config));

    ESP_ERROR_CHECK( esp_wifi_start() );

```

liuzfesp, 2018-10-09:
@outermanle could you help to check this commit: https://gitlab.espressif.cn:6688/idf/esp-idf/merge_requests/3374

liuzfesp, 2018-10-09:
@outermanle, this is just debug log, it should NOT impact the functionality of soft-ap + sta mode. Could you help to confirm it?

Moreover, we already have a fix for removing the debug log. It will be merged into the master branch this week.

outermanle, 2018-10-11:
ok, I will do a test. Thanks!

outermanle, 2018-10-13:
It works! Thanks!

liuzfesp, 2018-10-14:
It's fixed with commit: 299704cec3cb9e0d2208b08e01ccc89ebc777df8