espressif/esp-idf #2497: esp_wifi_scan_start failure after upgrade to 3.0.4

lukecyca, 2018-09-30:
## Environment

- Development Kit:      none
- Core (if using chip or module): ESP-WROOM32
- IDF version: **Problem exists in 3.0.4 and 3.1 (but not in 3.0.3)**
- Development Env:		Make
- Operating System:		MacOS
- Power Supply:         Tested on USB and external 5V (both with 3.3v voltage reg)

## Problem Description

I'm doing a "captive portal"-style setup procedure where it:
1. First attempts to connect to a previously-configured wifi network as a STA
2. If it fails, it starts an AP
3. A client (phone or computer) can connect to this AP and use a web interface to configure the device
4. The device performs a wifi scan to populate the web interface, so that a new wifi configuration can be set

### Expected Behavior

The `esp_wifi_scan_start()` should succeed and return a list of networks

### Actual Behavior

An error is produced:
```
wifi: Now is connecting, user scan invalid now!
ESP_ERROR_CHECK failed: esp_err_t 0x300c at 0x401337a1
0x401337a1: wifi_manager at /[...]/wifi_manager.c:522 (discriminator 1)
```

## Debug Logs

```
wifi: wifi firmware version: bd20748
I wifi: config NVS flash: enabled
I wifi: config nano formating: disabled
I system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I wifi: Init dynamic tx buffer num: 32
I wifi: Init data frame dynamic rx buffer num: 32
I wifi: Init management frame dynamic rx buffer num: 32
I wifi: wifi driver task: 3ffc3368, prio:23, stack:3584
I wifi: Init static rx buffer num: 10
I wifi: Init dynamic rx buffer num: 32
I wifi: wifi power manager task: 0x3ffc8078 prio: 21 stack: 2560
I phy: phy_version: 3663, 1d88daa, Aug  6 2018, 10:31:06, 0, 0
I wifi: mode : sta (24:0a:c4:9b:59:b8)
I CM03: Waiting for wifi
I BUTTON: 34 LONG
I CM03: Wifi disconnected
I CM03: Wifi disconnected
I CM03: EV: 36 1
I wifi: mode : sta (24:0a:c4:9b:59:b8) + softAP (24:0a:c4:9b:59:b9)
I wifi: sleep enable
I wifi: type: 1
HTTP Server listening...
I DNSSRV: DNS Server listening on 53/udp
I WIFIMGR: Reconnecting to VanHack.ca
I wifi: n:6 0, o:6 0, ap:6 0, sta:0 0, prof:6
I wifi: station: b4:8b:19:d5:3f:de join, AID=1, g, 20
I DNSSRV: Replying to DNS request (len=35) from 192.168.1.2
W wifi: Now is connecting, user scan invalid now!
ESP_ERROR_CHECK failed: esp_err_t 0x300c at 0x401337a1
0x401337a1: wifi_manager at /[...]/components/esp32-wifi-manager/./wifi_manager.c:522 (discriminator 1)

file: "/[...]/components/esp32-wifi-manager/./wifi_manager.c" line 522
func: wifi_manager
expression: esp_wifi_scan_start(&scan_config, true)

Backtrace: 0x4008a028:0x3ffca0b0 0x4008a395:0x3ffca0d0 0x401337a1:0x3ffca0f0
0x4008a028: invoke_abort at /[...]/esp-idf-3.0.4/components/esp32/./panic.c:572

0x4008a395: _esp_error_check_failed at /[...]/esp-idf-3.0.4/components/esp32/./panic.c:584

0x401337a1: wifi_manager at /[...]/components/esp32-wifi-manager/./wifi_manager.c:522 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57
```

XinDeng11, 2018-10-09:
hi@lukecyca, because connect need start a default scan itself, we call it connecting scan, if a scan command invoked when connecting scan is running, it will interrupt the connect scan and it can not send disconnect event instead of scan done event before 3.0.4, after our discussion, we deceide to change this behavior, user scan shouldn't interrput connect scan when connecting, so suggestion you invoke a disconnect before start scan so that it can scan

liuzfesp, 2019-03-18:
HI @snahmad, as you said, you need to call esp_wifi_disconnect() before calling esp_wifi_scan_start() if the ESP32 STA fails to connect the AP.  We will fix it in IDF master, but the change is big, we need time to do it.

Alvin1Zhang, 2019-06-30:
@lukecyca Thanks for reporting the issue, feel free to reopen the issue if it still exists. Thanks.

closed 2019-06-30 by Alvin1Zhang