espressif/esp-idf #2500: The following Python requirements are not satisfied

James-sjec, 2018-10-01:
[image]

James-sjec, 2018-10-01:
Hai all,
I was able to run esp-idf programs earlier , but after updating the esp-idf , I was getting the above error
I tried reinstalling from first , by installing the toolchains (https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20170918.zip) and esp-idf , but I am getting the same error.

chegewara, 2018-10-01:
https://github.com/espressif/esp-idf/issues/2480#issuecomment-424931850

projectgus, 2018-10-01:
In addition to comment linked above, you may also have to remove a pip-installed setuptools. Try these commands: 
https://github.com/espressif/esp-idf/issues/2480#issuecomment-425790054

Updated MSYS2 environment will be available ASAP.

James-sjec, 2018-10-02:
Thanks to all , for helping me to solve the issue

closed 2018-10-02 by James-sjec