espressif/esp-idf #2501: deep_sleep power consumption too high

mojo3344, 2018-10-01:
## Environment

- Development Kit:  NodeMCU and also custom pcb with WROOM32 module
- Core (if using chip or module): ESP-WROOM32
- IDF version: cded92bd (master on 10/1/18)
- Development Env: Make
- Operating System: Mac OS
- Power Supply:    External 3.3V

## Problem Description

/examples/system/deep_sleep does not product expected current consumption of ~5uA. I am measuring about ~100 uA using NodeMCU pcb with the usb-serial chip removed so that current consumption isn't biased by that. Same measurement with custom WROOM32 pcb as well.

### Expected Behavior

~5uA power consumption of WROOM32.

### Actual Behavior

~100uA power consumption.

### Steps to reproduce

To reproduce, clone, go into /examples/system/deep_sleep, make menuconfig and deselect example config for touch wakeup and ulp wakeup, then make and flash. Also, these boards don't have the button inputs, so comment out EXT1 wakeup sources:

````
    /*
    const int ext_wakeup_pin_1 = 25;
    const uint64_t ext_wakeup_pin_1_mask = 1ULL << ext_wakeup_pin_1;
    const int ext_wakeup_pin_2 = 26;
    const uint64_t ext_wakeup_pin_2_mask = 1ULL << ext_wakeup_pin_2;

    printf("Enabling EXT1 wakeup on pins GPIO%d, GPIO%d\n", ext_wakeup_pin_1, ext_wakeup_pin_2);
    esp_sleep_enable_ext1_wakeup(ext_wakeup_pin_1_mask | ext_wakeup_pin_2_mask, ESP_EXT1_WAKEUP_ANY_HIGH);
    */
````

Then run and measure current. I measure between 100uA-200uA here.

Then, checkout an old version of esp-idf:
    git checkout cded92bd3c39232295be6d901befe847db22bc74

I measure 10uA here.

It seems like something regressed between that old version and the latest. Any ideas?

cesarvizco, 2019-01-18:
Hi, any news about this issue?

Thanks

Cesar

liuzfesp, 2019-02-02:
@cesarvizco what's current if doesn't call esp_wifi_connect() after power on?

cesarvizco, 2019-02-04:
Hi, it's working now, I'm using this esp-idf version:

commit a62cbfec9ab681bddf8f8e45d9949b1b1f67a2ec (HEAD -> master, origin/master, origin/HEAD)
Merge: dbca85f96 bd8733f74
Author: Wang Jia Lin <wangjialin@espressif.com>
Date:   Tue Jan 15 17:44:51 2019 +0800

    Merge branch 'bugfix/fix_psram_eid' into 'master'
    
    bugfix(psram): fix psram size acquisition method, support esp32-pico chips
    
    See merge request idf/esp-idf!3750

Thanks

Cesar

projectgus, 2020-07-14:
Closing as the regression in power consumption was resolved.

closed 2020-07-14 by projectgus