espressif/esp-idf #2503: stack overflow in task btuT has been detected

chegewara, 2018-10-02:
@Weijian-Espressif here is my sdkconfig:
https://gist.github.com/chegewara/bb3ae58044ac8b1bff24740a35e90695

If i change ble logging level from DEBUG to default WARNING then i dont have this issue.
https://gist.github.com/chegewara/bb3ae58044ac8b1bff24740a35e90695#file-sdkconfig-L170-L345

Weijian-Espressif, 2018-10-08:
@chegewara please provide the log. is ESP32 scanning?

chegewara, 2018-10-10:
@Weijian-Espressif Yes, esp32 is scanning. This is not the same app, but the same issue with stack overflow when few bt log levels are higher than `WARNING`:
```
I (2824) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 30aea43f244e
E (2864) BLEDevice:  gattc_if: 3 == 3
E (2864) BLEDevice:  gattc_if: 3 == 3
E (2864) BLEDevice:  gattc_if: 3 == 3
E (2864) BLEDevice:  gattc_if: 3 == 3
I (2864) BLEClient: Get service information from remote device
E (2874) BLEDevice:  gattc_if: 3 == 3
I (2874) SampleClient: 7--> 5404
E (2884) BLERemoteCharacteristic:
E (2884) BLEDevice:  gattc_if: 3 == 3
I (2894) SampleClient: 8--> 5180
I (2894) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 30aea43f244e
I (2904) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 30aea43f244e
***ERROR*** A stack overflow in task btuT has been detected.
abort() was called at PC 0x4008fb08 on core 0
0x4008fb08: vApplicationStackOverflowHook at D:/msys32/home/imper/esp/esp-idf/components/esp32/panic.c:660

Backtrace: 0x4008f920:0x3ffd1d70 0x4008faf1:0x3ffd1d90 0x4008fb08:0x3ffd1db0 0x4008c49c:0x3ffd1dd0 0x4008e4c0:0x3ffd1df0 0x4008e476:0x3ffd2100 0x40082c9d:0x3ffd2120 0x400592fe:0x3ffd2140 0x400
5937a:0x3ffd2160 0x40058bbf:0x3ffd2180 0x400d9279:0x00000000
0x4008f920: invoke_abort at D:/msys32/home/imper/esp/esp-idf/components/esp32/panic.c:660

0x4008faf1: abort at D:/msys32/home/imper/esp/esp-idf/components/esp32/panic.c:660

0x4008fb08: vApplicationStackOverflowHook at D:/msys32/home/imper/esp/esp-idf/components/esp32/panic.c:660

0x4008c49c: vTaskSwitchContext at D:/msys32/home/imper/esp/esp-idf/components/freertos/tasks.c:3564

0x4008e4c0: _frxt_dispatch at D:/msys32/home/imper/esp/esp-idf/components/freertos/portasm.S:406

0x4008e476: _frxt_int_exit at D:/msys32/home/imper/esp/esp-idf/components/freertos/portasm.S:206

0x40082c9d: esp_log_write at D:/msys32/home/imper/esp/esp-idf/components/log/log.c:121

0x400d9279: __sprint_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:437
```

chegewara, 2018-10-11:
@Weijian-Espressif 
I have it enabled all the time and i dont print too much when i scan.

Weijian-Espressif, 2018-10-11:
@chegewara please Increase the BTU_TASK_STACK_SIZE  task size and try, thanks.

chegewara, 2018-10-11:
@Weijian-Espressif 
I did, i have 8 kB now and i think i did increase it up to 18-20kB. But sure i will do it now one more time to confirm. But in mean time i think this log can be useful too. When i reset one esp32 peripheral and central lost connection to it, then for some reason (timeout) all connected peripherals are disconnected. 

From now on all found peripherals are connected again if they are found:
```
I (24705) BT_GATT: GATT_GetConnIdIfConnected status=0

W (24705) BT_APPL: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=259 reason=0x0008
W (24705) BT_APPL: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=260 reason=0x0008
W (24715) BT_APPL: bta_gattc_conn_cback() - cif=5 connected=0 conn_id=261 reason=0x0008
I (24725) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
I (24775) BT_GATT: GATT_Disconnect conn_id=3
I (24775) BT_L2CAP: L2CA_RemoveFixedChnl()  CID: 0x0004  BDA: 30aea449a732
I (24785) BT_GATT: GATT_GetConnIdIfConnected status=0

W (24785) BT_APPL: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0016
W (24795) BT_APPL: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0016
W (24805) BT_APPL: bta_gattc_conn_cback() - cif=5 connected=0 conn_id=5 reason=0x0016
I (24815) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
I (24825) BT_GATT: GATT_Deregister gatt_if=3
I (24825) BT_GATT: GATT_Listen gatt_if=3
E (24835) BT_APPL: bta_gattc_deregister Deregister Failedm unknown client cif
I (24835) BT_GATT: GATT_Deregister gatt_if=4
I (24845) BT_GATT: GATT_Listen gatt_if=4
E (24845) BT_APPL: bta_gattc_deregister Deregister Failedm unknown client cif
I (24855) BT_GATT: GATT_Disconnect conn_id=517
I (24865) BT_L2CAP: L2CA_RemoveFixedChnl()  CID: 0x0004  BDA: 30aea443a172
I (24875) BT_GATT: GATT_GetConnIdIfConnected status=0
```

chegewara, 2018-10-11:
@Weijian-Espressif 
This is library with master and slave code i am using right now:
https://github.com/chegewara/esp32-snippets-enchancements-test/tree/multi_connect_test

There is few cpp files in main folder, but for tests i am using:
- central -> SampleClient.cpp
- peripheral -> SampleServer.cpp

Weijian-Espressif, 2018-10-11:
@chegewara Thanks for your help.  I will retry to reproduce the issue. By the way, when you register multiple gattc callbacks, one device disconnects Bluetooth and all callbacks trigger a disconnect event.

chegewara, 2018-10-11:
I have to check, but i think i have only one callback registered from which every client events are dispatcher according to gattc_if it is connected. But thanks for useful info.

chegewara, 2018-10-11:
Is there any particular reason why disconnect event is triggered on all callbacks when one peer device is disconnected?

chegewara, 2018-10-12:
@Weijian-Espressif btuT stack behaves very strange. Now i have test app as such:
- turn off debugging in menuconfig (option to decrease binary size), so there should not be anything that takes stack unnecesary,
- after bt initialization i am getting btuT handle and i am using it later to log stack size,
- when any device is connected to esp32 i am sending write to remote characteristic every 1000ms and i am printing to logs btuT minimal free stack size from watermark function,
Result:
- from time to time i am having very different values of btuT minimal free stack, from 322 to 1708 bytes, its just enough to do reset esp32 few times.

chegewara, 2018-10-13:
I am only guessing now, but maybe this is somehow related to this issue (btuT stack):
```
E (857) BT_OSI: osi_mem_dbg_clean invalid

W (30863) BT_L2CAP: L2CA_CancelBleConnectReq - no connection pending
E (30865) BT_OSI: osi_mem_dbg_clean invalid
```
After turning on `Debug bluetooth memory` in menuconfig im having BT_OSI error messages all the time.

Weijian-Espressif, 2018-10-31:
@chegewara Do you modify `BT DEBUG LOG LEVEL` in menuconfig ?

chegewara, 2018-10-31:
@Weijian-Espressif Yes, i have changed few BT DEBUG LOG LEVEL values to higher:
https://gist.github.com/chegewara/bb3ae58044ac8b1bff24740a35e90695#file-sdkconfig-L172-L338

Weijian-Espressif, 2018-10-31:
@chegewara  If you print too much debugging information, it will cause `btuT` exception. I have not found a good solution yet.

chegewara, 2018-10-31:
@Weijian-Espressif Dont worry about that, leave it for now. Im not complaining, just wanted to inform that some issue exists. Since i am only one, and it is only during extensive debugging it is not very important.
You could increase btuT stack size, but this would be waste of resources in this case. Thanks for support.

Weijian-Espressif, 2018-12-03:
@chegewara, For some customers who don't have enough memory, this modification reduce the free memory of ESP32 and will cause new problems, so I can't modify it.
If you need more btuT stack size, you can modify it.  .

chegewara, 2018-12-03:
I thout about 200-300 bytes, but thats fine. Thanks.

closed 2018-12-03 by chegewara