espressif/esp-idf #2507: mdns_query_a method is not working

AadiMehta, 2018-10-03:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v2]
- Core (if using chip or module): [ESP-WROOM32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 11b444b
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [USB]

## Problem Description
[mdns_query_a](https://docs.espressif.com/projects/esp-idf/en/stable/api-reference/protocols/mdns.html#_CPPv212mdns_query_aPKc8uint32_tP10ip4_addr_t) method returns ESP_ERR_NOT_FOUND when the host is present. ESP-IDF example for mdns also doesn't print the IP corresponding to its own service. All other methods e.g mdns_query_ptr, mdns_query_txt, etc are working as expected.
//Detailed problem description goes here.

### Expected Behavior
mdns_query_a should resolve the address correctly.

### Steps to repropduce
1. Run mdns esp-idf example. Make sure hostname specified in sdkconfig matches to that "mdns_query_host" function call in the example.
2. Run the example to see that it doesn't print the IP.

```W (350) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (520) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 2
I (540) wifi: mode : sta (24:0a:c4:11:b3:e8)
I (540) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1270) wifi: n:6 1, o:1 0, ap:255 255, sta:6 1, prof:1
I (2250) wifi: state: init -> auth (b0)
I (2260) wifi: state: auth -> assoc (0)
I (2260) wifi: state: assoc -> run (10)
I (2460) wifi: connected with Router, channel 6
I (2460) wifi: pm start, type: 1

I (5770) event: sta ip: 192.168.2.42, mask: 255.255.255.0, gw: 192.168.2.1
I (67320) mdns-test: Query A: esp32.local
W (69360) mdns-test: ESP_ERR_NOT_FOUND: Host was not found!
I (69360) mdns-test: Query PTR: _arduino._tcp.local
W (72460) mdns-test: No results found!
I (72460) mdns-test: Query PTR: _http._tcp.local
W (75560) mdns-test: No results found!
I (75560) mdns-test: Query PTR: _printer._tcp.local
W (78660) mdns-test: No results found!
I (78660) mdns-test: Query PTR: _ipp._tcp.local
W (81760) mdns-test: No results found!
I (81760) mdns-test: Query PTR: _afpovertcp._tcp.local
W (84860) mdns-test: No results found!
I (84860) mdns-test: Query PTR: _smb._tcp.local
W (87960) mdns-test: No results found!
I (87960) mdns-test: Query PTR: _ftp._tcp.local
W (91060) mdns-test: No results found!
I (91060) mdns-test: Query PTR: _nfs._tcp.local
```

**Zeroconf browser on different machine shows that an mdns service with esp.local hostname is already running in the network.**
[image]

me-no-dev, 2018-10-03:
mDNS will not resolve it's own IP :) it will never receive that query. You can query the IP of any other host though.

AadiMehta, 2018-10-03:
@me-no-dev Hi, I have tried querying IP of other host as well, and it still doesn't print the IP.

```
I (542) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1262) wifi: n:6 1, o:1 0, ap:255 255, sta:6 1, prof:1
I (2252) wifi: state: init -> auth (b0)
I (2252) wifi: state: auth -> assoc (0)
I (2262) wifi: state: assoc -> run (10)
I (2472) wifi: connected with Router, channel 6
I (2472) wifi: pm start, type: 1

I (5272) event: sta ip: 192.168.2.42, mask: 255.255.255.0, gw: 192.168.2.1
I (17022) mdns-test: Query A: esp32.local
W (19062) mdns-test: ESP_ERR_NOT_FOUND: Host was not found!
I (19062) mdns-test: Query PTR: _foresttech._tcp.local
1: Interface: STA, Type: V4
  PTR : Gateway-b827eb3931af
1: Interface: STA, Type: V4
  SRV : Gateway-b827eb3931af.local:80
1: Interface: STA, Type: V4
  TXT : [3] is_configured=false; gateway_id=false; common_name=Gateway-b827eb3931af; 
I (22452) mdns-test: Query A: Gateway-b827eb3931af.local
W (24562) mdns-test: ESP_ERR_NOT_FOUND: Host was not found!
```

Here is the screenshot of the zeroconf browser
[image]

david-cermak, 2018-10-04:
Hi @AadiMehta, 

There might be an issue with querying hostnames just after `mdns_hostname_set` (initial probing state may interact with sending queries). At this moment mdns query works but not immediately after init. You may be able to workaround with:

* not calling host_name_set at all (if you don't need to advertise your own mdns hostname)
* wait 2-3 seconds before querying, or query repeatedly
* increase the timeout for querying

Could you please check whether what you are seeing is the same issue? i.e. ask repeatedly for a known host with timeout of 5 seconds if it resolves the corresponding IP?

Thank you,
David

AadiMehta, 2018-10-09:
Hi @david-cermak 

I have your above suggestions, and it did work for one instance and didn't for other. So I have 2 instances of same service running inside my network(one on windows 10 and other on ubuntu). Here are the screenshots of the zeroconf browser:

[image]

[image]

I have tried all the points mentioned in your comment:
**Not calling host_name_set**
> not calling host_name_set at all (if you don't need to advertise your own mdns hostname)

**Queried repeatedly after waiting 2-3 seconds**
> wait 2-3 seconds before querying, or query repeatedly

**Increased timeout to 5 seconds**
> increase the timeout for querying 

IP is retrieved for windows service instance and not for linux based mdns service instance:
```
I (1121) wifi: n:6 2, o:1 0, ap:255 255, sta:6 2, prof:1
I (2101) wifi: state: init -> auth (b0)
I (2111) wifi: state: auth -> assoc (0)
I (2111) wifi: state: assoc -> run (10)
I (2261) wifi: connected with Router, channel 6
I (2271) wifi: pm start, type: 1

I (4251) event: sta ip: 192.168.2.54, mask: 255.255.255.0, gw: 192.168.2.1
I (9151) mdns-test: Query PTR: _foresttech._tcp.local
1: Interface: STA, Type: V4
  PTR : Awesome1
2: Interface: STA, Type: V4
  PTR : Gateway-b827eb3931af
1: Interface: STA, Type: V4
  SRV : 5d39d3ae97cf48fdb4edc4c9f071d121.local:80
1: Interface: STA, Type: V4
  TXT : [4] is_configured=false; common_name=Awesome1; gateway_id=5d39d3ae97cf48fdb4edc4c9f071d121; ip=192.168.2.45; 
I (12861) mdns-test: Query A: Gateway-b827eb3931af.local
W (17951) mdns-test: ESP_ERR_NOT_FOUND: Host was not found!
I (17951) mdns-test: Query A: 5d39d3ae97cf48fdb4edc4c9f071d121.local
I (18181) mdns-test: 192.168.2.45
```

david-cermak, 2018-10-10:
OK, thanks for checking and sharing the details. I can see that Query protocol gets the Gateway-b827eb3931af, but Query A does not. There is certainly something wrong, will check.
Yes, resolution should be case-insensitive (and indeed it is on my machine:-)

david-cermak, 2018-10-10:
Whilst the first symptom (not resolving after init) I am able to replicate (+ fix), the other two (case sensitive resolution, not resolving other instances) not. 

Would you mind trying this patch in your setup? if so please modify the file `components/mdns/mdns.c` line 3468 (function `_mdns_search_send_pcb`) from:
```
    if (_mdns_server->interfaces[tcpip_if].pcbs[ip_protocol].pcb && _mdns_server->interfaces[tcpip_if].pcbs[ip_protocol].state == PCB_RUNNING) {
```
to:
```
    if (_mdns_server->interfaces[tcpip_if].pcbs[ip_protocol].pcb && _mdns_server->interfaces[tcpip_if].pcbs[ip_protocol].state > PCB_INIT) {
```
Real fix gonna be a bit more complex, but that's just a gist of it. This should fix what I had in mind and maybe we are just lucky and fixes something else as well :smile: 

Thanks

AadiMehta, 2018-10-11:
@david-cermak 

Your patch works well, mdns resolution now starts working immediately after it get IP. 

However there can be one other issue as well(or the issue can be on python zeroconf package). So I have in my home setup, my own mdns service(using python zeroconf package) running on 2 instances(one windows and one linux) and a google home on which its own mdns service is running. 

On querying both services, following is the response snippet:
```
I (359) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (479) wifi: n:6 2, o:1 0, ap:255 255, sta:6 2, prof:1
I (1459) wifi: state: init -> auth (b0)
I (1469) wifi: state: auth -> assoc (0)
I (1529) wifi: state: assoc -> run (10)
I (1889) wifi: connected with Router, channel 6
I (1889) wifi: pm start, type: 1

I (4209) event: sta ip: 192.168.2.54, mask: 255.255.255.0, gw: 192.168.2.1
I (4809) mdns-test: Query PTR: _foresttech._tcp.local
1: Interface: STA, Type: V4
  PTR : Gateway-b827eb3931af
1: Interface: STA, Type: V4
  SRV : Gateway-b827eb3931af.local:80
1: Interface: STA, Type: V4
  TXT : [3] is_configured=false; common_name=Gateway-b827eb3931af; gateway_id=false; 
I (8339) mdns-test: Query PTR: _googlezone._tcp.local
1: Interface: STA, Type: V4
  PTR : f0a85d61-0297-cb4f-0ec0-b8260b9da65f
  SRV : f0a85d61-0297-cb4f-0ec0-b8260b9da65f.local:10001
  TXT : [2] id=C133E28350E7CF0C005AE509760400B9; __common_time__=1|0; 
  A   : 192.168.2.49
1: Interface: STA, Type: V4
  SRV : f0a85d61-0297-cb4f-0ec0-b8260b9da65f.local:10001
I (16229) mdns-test: Query A: gateway-b827eb3931af.local
W (21299) mdns-test: ESP_ERR_NOT_FOUND: Host was not found!
I (21309) mdns-test: Query A: gateway-123123123.local
W (26399) mdns-test: ESP_ERR_NOT_FOUND: Host was not found!
I (26409) mdns-test: Query A: f0a85d61-0297-cb4f-0ec0-b8260b9da65f.local
I (26679) mdns-test: 192.168.2.49
I (26679) mdns-test: Query A: 5d39d3ae97cf48fdb4edc4c9f071d121.local
W (31699) mdns-test: ESP_ERR_NOT_FOUND: Host was not found!
```

I need to query python zeroconf mdns service using all methods "mdns_query_ptr", "mdns_query_srv", "mdns_query_txt", "mdns_query_a" to retrieve the complete info, whereas for google service all info is printed all at once by using just "mdns_query_ptr" query.

david-cermak, 2018-10-11:
Thanks @AadiMehta for testing it! 

I will the fix to IDF and test mdns thoroughly (plus  with python-zeroconf to see if I can get to the other issue)

AadiMehta, 2018-10-11:
@david-cermak 

If you require, here is a sample python-zeroconf mdns server.

[sample_mdns_server.zip](https://github.com/espressif/esp-idf/files/2467792/sample_mdns_server.zip)

For this zeroconf python library will be required.

david-cermak, 2018-10-15:
Hi @AadiMehta,

The first issue (slow resolving) is being fixed, plus preparing automated tests to avoid such bugs in future.

Still not able to recreate the other issue, running  your zroconf server (Thanks for sharing the code!) on two platforms (Mac, linux), but QueryA works as expected. Could you please share the exact python code for your two instances (one resolving, the other not)? ...assuming there're just different names, etc as in the sample code you posted, but what if...

AadiMehta, 2018-10-15:
No resolution when server name starts with capital letters. Here is the code for that.
[Not_resolving_sample_mdns_server.zip](https://github.com/espressif/esp-idf/files/2479334/Not_resolving_sample_mdns_server.zip)

Second issue was that for python zeroconf server I have to use all APIs for resolution of PTR, SRV, TXT and A fields WHEREAS for google mdns service all fields are resolved directly with using only **mdns_query_ptr** API

AadiMehta, 2018-10-16:
@david-cermak Thank you david for the response. I will try it out once in my environment.

Alvin1Zhang, 2018-10-24:
@AadiMehta Could you help share if any updates for this issue? Thanks.

AadiMehta, 2018-10-25:
@Alvin1Zhang I have tried @david-cermak example in my environment and I confirm it works.

AadiMehta, 2018-10-28:
Hi @david-cermak , there might be an another issue or it might be a feature request.

So esp32 has successfully resolved a .local hostname, but when calling a REST API using this hostname(after this resolution), esp32 fails to send it to the hostname. This work completely correct when used from windows machine or any other linux machine.

I understand that windows/linux machines use combination of mdns and dns to resolve the hostnames but .local hostnames are only resolved using mdns and are then sent to appropriate IP addr. In my project I cannot directly use this resolved IP to call my API since my tls-hostname verification fails in that case.

negativekelvin, 2018-10-28:
You could try LWIP_DNS_SUPPORT_MDNS_QUERIES

AadiMehta, 2018-10-28:
@negativekelvin Thank you so much! It works for me!

closed 2018-11-18 by igrr

markg85, 2021-01-17:
Hi,

I'm having this very same issue too.
I'm in a linux setup where Avahi is handling mDNS for my pc's.
The pc's themselves can pin each other just fine using `<their name>.local`.
The ESP8266 could reach those same machines just fine too with mDNS.

I took a little excursion into the code along with wireshark and found one interesting clue that confuses everyone.
When i, in code, do a mdns lookup for `foo.local` (so the .local included!) then wireshark tells me that the lookup was actually for `foo.local.local`! Note the double ".local"!

So the "fix" is to not include the ".local" in my host.
If i then do a mdns lookup using some espressif sample code for a dns lookup, it magically works.

**However**, if i now pass in these hostnames without ".local", other parts of the code apparently think i want to do a **DNS** lookup as opposed to an **mDNS** lookup. And as i don't have those hostnames in my network without the .local suffix those lookups fail in WiFiGeneric.cpp line 654:
`[E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for foo`

That ".local" is added internally is confirmed in ESPmDNS.h itself in it's comment block:
```
Usage:
- Include the ESP32 Multicast DNS library in the sketch.
- Call the begin method in the sketch's setup and provide a domain name (without
  the '.local' suffix, i.e. just provide 'foo' to resolve 'foo.local'), and the
  Adafruit CC3000 class instance.  Optionally provide a time to live (in seconds)
  for the DNS record--the default is 1 hour.
```

Other libraries (i'm in particular having troubles with the PubSubClient library) connect to the hostname via a passed in Client object. That is often a WiFiClient where eventually this code is called:
```
int WiFiClient::connect(const char *host, uint16_t port, int32_t timeout)
{
    IPAddress srv((uint32_t)0);
    if(!WiFiGenericClass::hostByName(host, srv)){
        return 0;
    }
    return connect(srv, port, timeout);
}
```

As you can see in the snippet, the name is resolved by `WiFiGenericClass::hostByName` which looks like this:
```
int WiFiGenericClass::hostByName(const char* aHostname, IPAddress& aResult)
{
    ip_addr_t addr;
    aResult = static_cast<uint32_t>(0);
    waitStatusBits(WIFI_DNS_IDLE_BIT, 5000);
    clearStatusBits(WIFI_DNS_IDLE_BIT);
    err_t err = dns_gethostbyname(aHostname, &addr, &wifi_dns_found_callback, &aResult);
    if(err == ERR_OK && addr.u_addr.ip4.addr) {
        aResult = addr.u_addr.ip4.addr;
    } else if(err == ERR_INPROGRESS) {
        waitStatusBits(WIFI_DNS_DONE_BIT, 4000);
        clearStatusBits(WIFI_DNS_DONE_BIT);
    }
    setStatusBits(WIFI_DNS_IDLE_BIT);
    if((uint32_t)aResult == 0){
        log_e("DNS Failed for %s", aHostname);
    }
    return (uint32_t)aResult != 0;
}
```

At this point mDNS is nowhere to be found. The Lookup is done in regular DNS and therefore fails in my case. Remember, if i omit the `.local` i get a name that isn't known in my network. It **must** be ending with .local here.

I think there's at least 1 bug here, possible 2.
1. that '.local' should not be added behind the scenes. It should be visible. Thus you should just provide "foo.local" instead of "foo".
2. Perhaps just having this already fixes everything. If not, then somewhere in the host name handling mDNS should be checked for. I don't see it happening in the ESP32 not the ESP8266 code yet it's working in the ESP8266 and not in ESP32.

markg85, 2021-02-21:
@igrr can this issue be re-opened please?
It's still downright wrong.

See my comment from above that explains it in detail what is wrong and how to work around it. Moral of the story, ESP32 should **remove** it's default addition of ".local".

negativekelvin, 2021-02-21:
@markg85  if you want general dns lookup to support mdns then lwip has to be compiled with LWIP_DNS_SUPPORT_MDNS_QUERIES

As for the esp-idf mdns component limitations you could open a new feature request to allow domain name to be specified at runtime as well as multiple instances of mdns

igrr, 2021-02-21:
@markg85 Could you please open a new issue with your findings? You can reuse your comment above (click `...`, then choose "Reference in new issue").

markg85, 2021-02-21:
@negativekelvin I don't think i want that.
I'm reporting a bug, not requesting a feature. In my opinion the current behavior is bugged.

@iggr Yup, will do! Thank you for that tip, i didn't know that!

commit 174b3b89 (2018-11-21): mdns: sending search packets also in probing and announcing state

mdns queries did not work properly when send imeadiately after set_hostname, which cuased reinit of pcb and therefore restarted probing, so search packets were blocked until probing finished
closes #2507, closes #2593
commit e885ff23 (2019-01-11): mdns: sending search packets also in probing and announcing state

mdns queries did not work properly when send imeadiately after set_hostname, which cuased reinit of pcb and therefore restarted probing, so search packets were blocked until probing finished
closes #2507, closes #2593