espressif/esp-idf #2512: ASSERT_PARAM(64,0), in rwble.c at line 222

jkhax0r, 2018-10-03:
## Environment

- Development Kit:      [ESP-WROOM-32 custom PCB]
- Core (if using chip or module): [ESP-WROOM32]
- IDF version 
Latest release/3.0 as of yesterday (5b061a0530e208bdcc08969a10518d2cf1cd43c8).  Also tried on release/3.1 with same ASSERT_PARAM(64,0) message

- Development Env:		[Make|Eclipse]
- Operating System:		[Windows]
- Power Supply:         [external 5V]

## Problem Description

I have an application that is doing BLE scanning, BLE advertising, and WIFI and it crashes periodically with an ASSERT_PARAM(64,0), in rwble.c at line 222

### Actual Behavior
Crashes.  Can be up to hours until it happens

### Steps to repropduce

1. Enable ble scanning, ble advertising (gatts enabled but no connections), and wifi station mode
2. Connect to wifi access point automatically on boot
3. Run for a while listening for BLE messages.  No connection over BLE needs to happen and not much else happening in the application.  

I believe this is related to scanning because we have been running on our ESP "platform" for various other products for a long time without this issue.  The only thing different about this is that I have added scanning into the mix.  It is likely the same thing from long ago that was described here: https://github.com/espressif/esp-idf/issues/880#issuecomment-320475254.  This was the last time we used scanning in one of our projects.  

### Code to reproduce this issue

Here's my custom config which should mostly explain the various 
```cpp

static const ble_controller_config_t ble_controller_config = {
	.gatts_default_device_name = ble_name,
	.app_adv_params = {
			.adv_int_min        = 0x0200,
			.adv_int_max        = 0x0400,
			.adv_type           = ADV_TYPE_IND,
			.own_addr_type      = BLE_ADDR_TYPE_PUBLIC,
			.channel_map        = ADV_CHNL_ALL,
			.adv_filter_policy 	= ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY, //< To be changed after 2 minute timer.
		},
	.app_adv_data = {
				{
					.flag = (ESP_BLE_ADV_FLAG_GEN_DISC | ESP_BLE_ADV_FLAG_BREDR_NOT_SPT),
					.set_scan_rsp = false,
					.include_name = false,
					//.include_txpower = true,
					//.min_interval = 0x20,
					//.max_interval = 0x40,
					//.appearance = 0x00,
					.p_service_uuid = PUBSUB_SERVICE_UUID,
					.service_uuid_len = sizeof(PUBSUB_SERVICE_UUID),
				},
				{
					.set_scan_rsp = true,
					.include_name = true,
					.manufacturer_len = sizeof(mfg_data_t),
					.p_manufacturer_data = (uint8_t *)&mfg_data,   //NOTE: This must be RAM pointer
				}
			},
	.app_scan_params = {
	    .scan_type              = BLE_SCAN_TYPE_PASSIVE,    // Change to _ACTIVE to request scan responses
	    .own_addr_type          = BLE_ADDR_TYPE_PUBLIC,
	    .scan_filter_policy     = BLE_SCAN_FILTER_ALLOW_ALL,
	    /// IMPORTANT: WIFI disconnects and has errors if the scan interval is set to the same as the scan window (100% scan duty)
	    .scan_interval          = 0x58, // Set scan interval (0.625msec per tick)
	    .scan_window            = 0x20, // Set scan window (0.625msec per tick)
		.scan_duplicate         = BLE_SCAN_DUPLICATE_DISABLE,
	},
	.ble_power_level = ESP_PWR_LVL_P7, ///< Default is 1dbm, using 7dbm should double tx range
	.services_flags = {
		.enable_gattc = 0,
		.enable_gatts = 1,
		.enable_bonded = 1,
		.enable_user_set_device_name = 1,
		.enable_gap_scanning = 1,
		.enable_multiple_gatts_connections = 1
	},
};
```

## Debug Logs

```
ASSERT_PARAM(64 0), in rwble.c at line 222
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Register dump:
PC      : 0x400853a0  PS      : 0x00060034  A0      : 0x80087f6a  A1      : 0x3ffc05b0
0x400853a0: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054
A6      : 0x3ffc1028  A7      : 0x60008050  A8      : 0x8008539d  A9      : 0x3ffc0590
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x400852d5  LEND    : 0x400852dc  LCOUNT  : 0x00000000
0x400852d5: r_assert_param at ??:?

0x400852dc: r_assert_param at ??:?

Backtrace: 0x400853a0:0x3ffc05b0 0x40087f67:0x3ffc05d0 0x4008886f:0x3ffc05f0 0x40083801:0x3ffc0610 0x400d19ed:0x00000000
0x400853a0: r_assert_param at ??:?

0x40087f67: r_rwble_isr at ??:?

0x4008886f: r_rwbtdm_isr_wrapper at intc.c:?

0x40083801: _xt_lowint1 at C:/WORKING/bjn/TankCheckGateway/firmware/esp-idf/components/freertos/xtensa_vectors.S:1105

0x400d19ed: esp_vApplicationIdleHook at C:/WORKING/bjn/TankCheckGateway/firmware/esp-idf/components/esp32/freertos_hooks.c:85

CPU halted.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

jkhax0r, 2018-10-05:
I'm pretty sure I've tried with both sleep enabled and not, and had the same results.  Also I had the same error when I had a product using BLE scanning from over a year ago (https://github.com/espressif/esp-idf/issues/880#issuecomment-320475254, long before sleep).

I think there's a lot of confusion in these forums b/c that `ASSERT_PARAM(64 0), in rwble.c at line 222` looks very similar in a lot of posts, but I suspect the **64** assert code is the key here and that every failure with a different number may be largely unrelated.  If you look at my stack trace its quite different from the thread at #2070 

I'm sure the espressif guys could clarify but thats my gut.  And given all these ASSERT_PARAM results in the forums, maybe it would be nice if Espressif could add some friendly text descriptions to help guide a bit better.

Weijian-Espressif, 2018-10-30:
@jkhax0r We are doing fix "ASSERT_PARAM(64 0), in rwble.c at line 222", and other error may related to this error.

chegewara, 2018-11-18:
This log looks different, but i thought i will ask if this is related before i open new issue:
```
W (260712)ASSERT_ERR((((uint16_t)nextptr << 0) & ~((uint16_t)0x00007FFF)) == 0), in lld_pdu.c
at line 85
 SampleServer: 15 ms after sending notify
W (260724) SampleServer: 15 ms after sending notify
W (260736) SampleServer: 15 ms after sending notify
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40086f09  PS      : 0x00060234  A0      : 0x8004afde  A1      : 0x3ffbab60
0x40086f09: r_assert_err at ??:?

A2      : 0x00000001  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffbdcd0
A6      : 0x00060a20  A7      : 0x00000000  A8      : 0x80086f04  A9      : 0x3ffbab40
A10     : 0x0000005f  A11     : 0x0000005f  A12     : 0x00000010  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffd  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40086ead  LEND    : 0x40086eb4  LCOUNT  : 0x00000000
0x40086ead: r_assert_err at ??:?

0x40086eb4: r_assert_err at ??:?

Backtrace: 0x40086f09:0x3ffbab60 0x4004afdb:0x3ffbab80 0x4004b884:0x3ffbabb0 0x40089c63:0x3ffb
abe0 0x4015df51:0x3ffbac20 0x4004240b:0x3ffbac40 0x4001a637:0x3ffbac60 0x40019d11:0x3ffbac90 0
x40055b4d:0x3ffbacb0 0x40157c3b:0x3ffbacd0 0x40158045:0x3ffbacf0 0x4008fe29:0x3ffbad20
0x40086f09: r_assert_err at ??:?

0x40089c63: r_lld_pdu_tx_prog at ??:?

0x4015df51: r_lld_pdu_data_send at ??:?

0x40157c3b: r_rw_schedule at ??:?

0x40158045: btdm_controller_task at ??:?

0x4008fe29: vPortTaskWrapper at D:/msys32/home/imper/esp/esp-idf/components/freertos/port.c:43
5
```

diegojpalaquibay, 2019-01-04:
Hello
I just update my SDK 3.1.1 to 3.1.2 
My application has the ble sleep mode enabled and the wifi/bt coexistence enabled
and I get this error when i try to start ble adv after stop the ble scan:
Error:
ASSERT_PARAM(64 0), in rwble.c at line 222
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
 
I didnt have this error in the sdk 3.1.1
I tested and I got that this error appears since the commit 
________________________________________________________________________________________________________
Commit: 195f7274c454bd08e9d5cc4011bc72cada1dd381 [195f727]
Merge branch 'bugfix/btdm_controller_disable_v3.1' into 'release/v3.1'
Fix two bugs when disable bluetooth controller(backport v3.1)
_________________________________________________________________________________________________________
@Weijian-Espressif  There is some configuration that i need to change when I start to use the ble scan and advertise in the sdk 3.1.2, because I dont have any problem in the sdk 3.1.1

Weijian-Espressif, 2019-01-14:
@diegojpalaquibay, The branch  'bugfix/btdm_controller_disable_v3.1' is to fix crash when controller disable.It seems that it has nothing to do with your issue. Can you provide your test code? This will help me to quickly reproduce the issue.

lucascoxBAF, 2019-02-01:
@Weijian-Espressif, I'm also periodically getting "ASSERT_PARAM(64 0), in rwble.c at line 230".

esp-idf version: v3.3-beta1-326-gd7a7a6835

lucascoxBAF, 2019-02-11:
@Weijian-Espressif , any luck on this?

Alvin1Zhang, 2020-07-30:
Thanks for reporting and sorry for slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/2f6f842d0f2a0e57541ab8a71a1ad5ba382a956c. Feel free to reopen if the issue still happens. Thanks.

closed 2020-07-30 by Alvin1Zhang