espressif/esp-idf #2515: ble SM issue

chegewara, 2018-10-04:
I dont know exactly where the issue is, but it is easy to reproduce with small change in gatt_security_server example. Its enough to change this part of code:
```
        case ESP_GATTS_DISCONNECT_EVT:
            ESP_LOGI(GATTS_TABLE_TAG, "ESP_GATTS_DISCONNECT_EVT");
            /* start advertising again when missing the connect */
            remove_all_bonded_devices();   <--- add this line to clear whitelisted devices on disconnect
            esp_ble_gap_start_advertising(&heart_rate_adv_params);
            break;
```
Behavior:
- expected: after disconnecting from server peer device is deleted from whitelist and next connection is performing with authentication request because peer device is no longer whitelisted,
- actual: after disconnecting from server peer device first time is connecting without authentication request as is still whitelisted, only next time is connecting with authentication request.

```
I (355763) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT
D (355764) BT_SMP: SMDBG l2c smp_connect_callback

I (355765) BT_APPL: BTA_DmSetBleAdvParamsAll: 256, 256

D (355769) BT_BTM: btm_sec_disconnected before update sec_flags=0x1280

I (355775) BT_APPL: adv_type = 0, addr_type_own = 3, chnl_map = 7, adv_fil_pol = 0

D (355782) BT_BTM: btm_sec_disconnected after update sec_flags=0x1080

D (355796) BT_APPL: BTA got event 0x13d

D (355800) BT_APPL: bta_dm_sm_execute event:0x3d
I (355805) BT_BTM: BTM_IsAclConnectionUp: RemBdAddr: 5a5374723e93

I (355811) BT_BTM: BTM_IsAclConnectionUp: RemBdAddr: 5a5374723e93

D (355818) BT_BTM: btm_ble_resolving_list_remove_dev

D (355824) BT_BTM: btm_ble_suspend_bg_conn

D (355829) BT_BTM: btm_ble_suspend_bg_conn

D (355833) BT_BTM: BTM: BTM_DeleteStoredLinkKey: delete_all_flag: FALSE
D (355840) nvs: nvs_open_from_partition bt_config.conf 1
D (355841) BT_APPL: BTA got event 0x104

D (355845) nvs: nvs_set_blob bt_cfg_key0 426
D (355849) BT_APPL: bta_dm_sm_execute event:0x4
D (355859) nvs: nvs_close 43
I (355860) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
D (355861) SEC_GATTS_DEMO: ESP_GAP_BLE_REMOVE_BOND_DEV_COMPLETE_EVT status = 0
D (355868) BT_APPL: BTA got event 0x104

I (355876) SEC_GATTS_DEMO: ESP_GAP_BLE_REMOVE_BOND_DEV
D (355880) BT_APPL: bta_dm_sm_execute event:0x4
I (355885) SEC_GATTS_DEMO: -----ESP_GAP_BLE_REMOVE_BOND_DEV----
D (355890) BT_APPL: BTA got event 0x124

I (355897) SEC_GATTS_DEMO: 5a 53 74 72 3e 93
D (355901) BT_APPL: bta_dm_sm_execute event:0x24
I (355906) SEC_GATTS_DEMO: ------------------------------------
D (355910) BT_BTM: BTM_BleSetAdvParamsStartAdv

D (355922) BT_BTM: update params for an active adv

D (355926) BT_BTM: btm_ble_start_adv

D (355930) BT_BTM: BTM_SUCCESS

D (355933) BT_APPL: BTA got event 0x1f02

W (355937) BT_APPL: BTA got unregistered event id 31

D (355943) BT_APPL: BTA got event 0x1f0c

W (355947) BT_APPL: BTA got unregistered event id 31

I (355953) SEC_GATTS_DEMO: advertising start success
D (372378) BT_BTM: btm_identity_addr_to_random_pseudo
D (372378) BT_BTM: btm_ble_resolve_random_addr
D (372378) BT_BTM: btm_ble_match_random_bda rec_index = 0
D (372382) BT_BTM: btm_ble_match_random_bda rec_index = 1
D (372388) BT_BTM: btm_ble_match_random_bda rec_index = 2
D (372393) BT_BTM: btm_ble_match_random_bda rec_index = 3
D (372399) BT_BTM: btm_ble_match_random_bda rec_index = 4
D (372404) BT_BTM: btm_ble_match_random_bda rec_index = 5
D (372410) BT_BTM: btm_ble_match_random_bda rec_index = 6
D (372415) BT_BTM: btm_ble_match_random_bda rec_index = 7
D (372420) BT_BTM: btm_ble_match_random_bda rec_index = 8
D (372426) BT_BTM: btm_ble_match_random_bda rec_index = 9
D (372431) BT_BTM: btm_ble_match_random_bda rec_index = 10
D (372437) BT_BTM: btm_ble_match_random_bda rec_index = 11
D (372442) BT_BTM: btm_ble_match_random_bda rec_index = 12
D (372448) BT_BTM: btm_ble_match_random_bda rec_index = 13
D (372454) BT_BTM: btm_ble_match_random_bda rec_index = 14
D (372459) BT_BTM: btm_ble_match_random_bda rec_index = 15
D (372465) BT_BTM: btm_ble_resolve_address_cmpl p_mgnt_cb->index = 15
D (372471) BT_BTM: btm_ble_resolve_random_addr_on_conn_cmpl

D (372477) BT_BTM: btm_ble_connected
D (372481) BT_BTM: btm_sec_alloc_dev

I (372485) BT_BTM: BTM_InqDbRead: bd addr [574280c74a61]

D (372491) BT_L2CAP: l2c_ble_link_adjust_allocation  num_hipri: 0  num_lowpri: 1  low_quota: 10  r
ound_robin_quota: 0  qq: 10
D (372502) BT_L2CAP: l2c_ble_link_adjust_allocation LCB 0   Priority: 0  XmitQuota: 10
D (372510) BT_L2CAP:         SentNotAcked: 0  RRUnacked: 0
D (372516) BT_L2CAP: CID:0x0044 FCR Mode:0 Priority:2 TxDataRate:1 RxDataRate:1 Quota:200
D (372524) BT_BTM: btm_find_or_alloc_dev

D (372530) BT_BTM: btm_ble_read_remote_features_complete
I (372535) BT_GATT: GATT_GetConnIdIfConnected status=1

I (372540) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
D (372541) BT_SMP: SMDBG l2c smp_connect_callback

I (372546) BT_APPL: BTA_DmSetEncryption
D (372551) BT_APPL: BTA got event 0x104

D (372558) BT_APPL: bta_dm_sm_execute event:0x4
I (372563) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
D (372570) BT_APPL: BTA got event 0x104

D (372574) BT_APPL: bta_dm_sm_execute event:0x4
D (372579) BT_APPL: BTA got event 0x10c

D (372583) BT_APPL: bta_dm_sm_execute event:0xc
I (372587) BT_BTM: Security Manager: BTM_SetEncryption Handle:65535 State:0 Flags:0x80 Required:0x
6

D (372597) BT_SMP: SMP_Pair state=0 br_state=0 flag=0x0

I (372603) BT_L2CAP: L2CA_ConnectFixedChnl() CID: 0x0006  BDA: 574280c74a61
D (372610) BT_L2CAP: CID:0x0004 FCR Mode:0 Priority:2 TxDataRate:1 RxDataRate:1 Quota:200
D (372619) BT_L2CAP: CID:0x0045 FCR Mode:0 Priority:2 TxDataRate:1 RxDataRate:1 Quota:200
D (372627) BT_SMP: SMDBG l2c smp_connect_callback

D (372632) BT_SMP: smp_connect_callback()  for pairing BDA: 574280c74a61  Event: connected

D (372640) BT_SMP: main smp_sm_event

D (372644) BT_APPL: io mitm: 1 oob_data:0

D (372648) BT_SMP: main smp_sm_event

D (372652) BT_SMP: smp_send_cmd on l2cap cmd_code=0xb

D (372657) BT_SMP: smp_build_security_request

D (372662) BT_SMP: opcode=11 auth_req=0x1
D (372666) BT_SMP: smp_send_msg_to_L2CAP
```

chegewara, 2018-10-04:
This is logging from nRF connect (second connection) when esp32 is connected without auth request:
```
nRF Connect, 2018-10-04
ESP_BLE_SECURITY (76:0C:FB:6A:77:5B)
D	04:09:52.347	gatt.close()
D	04:09:52.351	wait(200)
V	04:09:52.552	Connecting to 76:0C:FB:6A:77:5B...
D	04:09:52.552	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	04:09:53.254	[Server callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	04:09:53.254	[Server] Device with address 76:0C:FB:6A:77:5B connected
D	04:09:53.301	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D	04:09:53.321	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	04:09:53.321	Connected to 76:0C:FB:6A:77:5B
D	04:09:53.325	wait(1600ms)
W	04:09:53.545	Connection parameters update failed with status: UNACCEPT CONN INTERVAL (0x3b) (interval: 48.75ms, latency: 0, timeout: 5000ms)
V	04:09:54.926	Discovering services...
D	04:09:54.926	gatt.discoverServices()
D	04:09:54.961	[Callback] Services discovered with status: 0
I	04:09:54.961	Services discovered
V	04:09:54.998	Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
- Central Address Resolution [R] (0x2AA6)
Heart Rate (0x180D)
- Heart Rate Measurement [N] (0x2A37)
   Client Characteristic Configuration (0x2902)
- Body Sensor Location [R] (0x2A38)
- Heart Rate Control Point [R W] (0x2A39)
D	04:09:54.999	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	04:09:55.004	gatt.setCharacteristicNotification(00002a37-0000-1000-8000-00805f9b34fb, true)
I	04:10:23.383	Connection parameters updated (interval: 48.75ms, latency: 0, timeout: 5000ms)
D	04:10:26.453	[Server callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I	04:10:26.453	[Server] Device disconnected
D	04:10:26.487	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
D	04:10:26.547	[Callback] Connection state changed with status: 19 and new state: DISCONNECTED (0)
I	04:10:26.547	Disconnected
W	04:10:26.571	Connection terminated by peer (status 19)
```

Weijian-Espressif, 2018-10-15:
@chegewara I found that remove bond list is not complete, I will fix it as soon as possible.

chegewara, 2018-10-15:
@Weijian-Espressif No problem, thanks for investigating and no rush.

Weijian-Espressif, 2018-10-30:
@chegewara please update to the latest v3.1 and try, thanks.

chegewara, 2018-11-05:
@Weijian-Espressif What is event 32 please:
`BTA got unregistered event id 31`

Weijian-Espressif, 2018-11-14:
@chegewara ,  please refer to esp-idf/components/bt/bluedroid/bta/gatt/include/bta_gattc_int.h. 
enum {
    BTA_GATTC_API_OPEN_EVT   = BTA_SYS_EVT_START(BTA_ID_GATTC),
    BTA_GATTC_INT_OPEN_FAIL_EVT,
    BTA_GATTC_API_CANCEL_OPEN_EVT,
    BTA_GATTC_INT_CANCEL_OPEN_OK_EVT,

    BTA_GATTC_API_READ_EVT,
    BTA_GATTC_API_WRITE_EVT,
    BTA_GATTC_API_EXEC_EVT,
    BTA_GATTC_API_CFG_MTU_EVT,

    BTA_GATTC_API_CLOSE_EVT,

    BTA_GATTC_API_SEARCH_EVT,
    BTA_GATTC_API_CONFIRM_EVT,
    BTA_GATTC_API_READ_MULTI_EVT,
    BTA_GATTC_API_REFRESH_EVT,

    BTA_GATTC_INT_CONN_EVT,
    BTA_GATTC_INT_DISCOVER_EVT,
    BTA_GATTC_DISCOVER_CMPL_EVT,
    BTA_GATTC_OP_CMPL_EVT,
    BTA_GATTC_INT_DISCONN_EVT,

    BTA_GATTC_INT_START_IF_EVT,
    BTA_GATTC_API_REG_EVT,
    BTA_GATTC_API_DEREG_EVT,
    BTA_GATTC_API_LISTEN_EVT,
    BTA_GATTC_API_BROADCAST_EVT,
    BTA_GATTC_API_DISABLE_EVT,
    BTA_GATTC_ENC_CMPL_EVT,  
    BTA_GATTC_API_CACHE_ASSOC_EVT,
    BTA_GATTC_API_CACHE_GET_ADDR_LIST_EVT,
};

chegewara, 2018-11-14:
@Weijian-Espressif I did, but there is less than 31 events in this enum, thats why i am asking here.

chegewara, 2018-11-14:
@Weijian-Espressif 
`BTA_GATTC_API_OPEN_EVT = BTA_SYS_EVT_START(BTA_ID_GATTC)` =>
`#define BTA_SYS_EVT_START(id)       ((id) << 8)` where id --> `#define BTA_ID_GATTC        31`
i have really bad day today, but this is 0x1F00, no way to have event id 31

dhrishi, 2018-11-14:
@chegewara The id that is printed in the statement has been created by right shifting the system event by 8 bits . You can check this here: https://github.com/espressif/esp-idf/blob/fb7ba1baa068a2267ef54b7105cf3c8a004fe4eb/components/bt/bluedroid/bta/sys/bta_sys_main.c#L492

chegewara, 2018-11-14:
Great, now is another question. If the event is knows as `BTA_GATTC_API_OPEN_EVT `, why it is log printed as unregistered? It should be registered in bt stack or am i wrong?

chegewara, 2018-11-14:
Sorry for last post, my brain is very strange today. 
After investigating code, and this message `BT_APPL: BTA got unregistered event id 31` i have last one question. Why gatt client callback `BTA_ID_GATTC <-- event 31` is triggered in gatt_secure_server example?

Weijian-Espressif, 2018-11-15:
@chegewara  When you remove bonded device,  ESP32 needs to remove all pending background connection before unpair and remove all cached GATT information. you can refer to `bta_dm_process_remove_device()` and `BTA_GATTC_Refresh()` in `bta_dm_process_remove_device()`.

closed 2018-11-15 by chegewara