espressif/esp-idf #2516: OTA certificate generation issue

kmakhil90, 2018-10-04:
I am not able togenerate the proper certificate files for my connection as detailed in readme file for OTA sample code.
Please help to probe this problem.

Thanks,

Alvin1Zhang, 2018-10-08:
Hi, @kmakhil90 Thanks for taking time to report this issue, would you please help provide more details about your issues, based on the template https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md ? Thanks.

Alvin1Zhang, 2018-10-24:
@kmakhil90 Could you help share if any updates for this issue? Thanks.

kmakhil90, 2018-10-24:
I could solve this problem by using a different openssl command to generate certificate. The command mentioned in readme not worked.

jitin17, 2018-10-27:
@kmakhil90 Were you trying to generate certificates using openssl on Windows?
Anyways, it would be nice if can specify the command that worked for you and the host machine OS, so that we can modify the README accordingly. Thanks.

kmakhil90, 2018-10-29:
Hi,
I tried the command: 
openssl req -newkey rsa:2048 -nodes -keyout key.pem -x509 -days 365 -out certificate.pem
from following link: https://www.ibm.com/support/knowledgecenter/en/SSWHYP_4.0.0/com.ibm.apimgmt.cmc.doc/task_apionprem_gernerate_self_signed_openSSL.html

jitin17, 2018-11-06:
@kmakhil90 `openssl req -x509 -newkey rsa:2048 -keyout ca_key.pem -out ca_cert.pem -days 365`
This is the command found in README.md. The difference between this command and the one that works for you is that your command has a `--nodes` flag. So, I think if your command works then the one provided in the README should also work. 

So, can you post the error that you get on running the command specified in the README.

On a side note, `--nodes` is not recommended as it does not encrypt the private key.