espressif/esp-idf #2519: Can Bus off

MrAlexandrKolesnikov, 2018-10-04:
Hi! Is there any way to restart the kana module when bus off without waiting for 128 bus free signal?
Reset Mode or full reinit has no effect.

MrAlexandrKolesnikov, 2018-10-09:
I use only can_types.h on idf version 3.0. Any way restart can use cleaning and setting can module bits?

Dazza0, 2018-10-09:
I'm not sure where you got `can_types.h`. However can try the following
```
#include "driver/periph_ctrl.h"
...

periph_module_disable(PERIPH_CAN_MODULE);
periph_module_enable(PERIPH_CAN_MODULE);
```
This should have the effect of resetting the CAN module on the hardware level (gates the clock signal and sets the reset line) after which the registers in the CAN module should have the same default values as a power-on reset.

MrAlexandrKolesnikov, 2018-10-09:
I use only direct access to can registers like CAN object on your code. And can work fine. But sometimes i get bus-off interrupt from it. I try clear register and set it as you do in  `can_driver_uninstall();
can_driver_install(&g_config, &t_config, & f_config);
can_start();` 
but it has no effect. I also try
 "#include "driver/periph_ctrl.h"
periph_module_disable(PERIPH_CAN_MODULE);
periph_module_enable(PERIPH_CAN_MODULE);"
the call my init function 
but it has no effect

Dazza0, 2018-10-09:
Do you have a code snippet you could show of your which registers you set and clear after a bus off?

MrAlexandrKolesnikov, 2018-10-09:
Of course!

When I get bus-off interrupt i do:
`CAN.mode_reg.reset = 1;`
`(void)CAN.interrupt_reg.val`
`(void)CAN.arbitration_lost_captue_reg.val`
`(void)CAN.error_code_capture_reg.val`
`periph_module_disable(PERIPH_CAN_MODULE);`
`periph_module_enable(PERIPH_CAN_MODULE);`
`init();`

Dazza0, 2018-10-10:
I'm still unable to recreate the issue. Could you send your `init()` function as well? I suspect the error may be due to a fault re-initialization.

MrAlexandrKolesnikov, 2018-10-10:
Sory, i can't show you everything, but this is the whole part responsible for initializing the driver and our register description
[can_driver.txt](https://github.com/espressif/esp-idf/files/2463433/can_driver.txt)
[can_driver.txt](https://github.com/espressif/esp-idf/files/2463434/can_driver.txt)

Dazza0, 2018-10-10:
I don't see anything wrong with your init. Try running the following code but check the following points listed below.
```
periph_module_disable(PERIPH_CAN_MODULE);
periph_module_enable(PERIPH_CAN_MODULE);
init();
```
- Check that the CAN module is in the correct state after re-enabling and re-initializing by reading the values of the mode register, status register,  and both error counter registers. Confirm that reset mode has been exited, the bus-status bit is 0, and that both error counters have been reset to 0.

How do you determine that the re-init was unsuccessful? Are you attempting to transmit/receive a frame after re-initialization with no results? If so, check the following 
- Is the condition that caused the bus-off still present on the bus (i.e. is another node on the bus pulling the bus into a dominant state)? What is the state of the CAN bus after re-initialization? Please verify with a logic analyzer.
- Is there any activity on the CAN bus when attempting to transmit or receive after re-initialization. Please verify with a logic analyzer.
- Try re-initializing into self-test mode and transmit a frame to verify that the connection between the CAN module and the external transceiver is still operating correctly (i.e. the GPIOs have been configured correctly). Failure to do so can cause a CAN module to constantly interpret an attempted transmission as a loss in arbitration resulting in no activity on the CAN bus.

MrAlexandrKolesnikov, 2018-10-10:
Ok, i see that RXERROR counter not clear after enable/disable. I try clear it but it steel equal 128. I check the line and see than CAN H and CAN L have the same level.  Do you have any example how init into self-test mode?

Dazza0, 2018-10-10:
[CAN Self Test Example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/can/can_self_test)

Here's what should ideally occur
1. When bus off occurs, the CAN module should automatically be put into reset mode. The RX error counter should be set to 0 and the TX error counter is set to 128.
2. After a hardware reset (module disable then enable), the CAN module should be automatically be put into reset mode. Both error counters, ALC and ECC should be cleared to zero. TX and RX status should both be set. The rest of the configuration registers should remain unchanged.
3. After reset mode exits (i.e. after calling `init()`), TX and RX status will be unset after observing 11 consecutive recessive bits on the bus.

Could you verify at which stage the RX error counter becomes 128 and ALC becomes 7. If the RX counter is 0 after a hardware reset and becomes 128 after reset mode exits, it suggests that something on the bus or the RX pin is causing the CAN module to interpret RX errors after re-initialization.

If the RX error counter becomes 128 after calling `init()` (i.e. reset mode has exit) WITHOUT transmitting anything. This suggests something external to the CAN module causing this increase. Double check that your selected GPIOs are not being shared with any other functionality (e.g. UART, SPI) and that your GPIO reconfiguration is correct. Anything that pulls the RX pin low can cause RX error counter to increase (e.g. GPIO pad being set to a pull down).

If the RX error counter becomes  128 after calling `init()` whilst something is being transmitted, the attempted transmission could be what is causing the RX error counter to increase.

MrAlexandrKolesnikov, 2018-10-10:
After a hardware reset ECC clear to zero, ALC not clear. My bad - RX error counter after init clear to zero but Receive Status register equal 0x1.

Dazza0, 2018-10-10:
Is the status register = 0x1 or just the receive status bit in the status register?

MrAlexandrKolesnikov, 2018-10-10:
receive status bit steel equal 0x1

Dazza0, 2018-10-10:
As mentioned above, once `init()` is called, both the receive status bit and transmit status bit should both be set to 0x1 until 11 recessive bits are observed on the bus. Try putting a short delay (e.g. 500ms) after `init()` and check the status register again. If receive status bit is still 0x1 after the delay, this indicates that the CAN module is receiving a message. If that is the case, you should check the state of your RX pin.

Alvin1Zhang, 2018-10-24:
@MrAlexandrKolesnikov Could you help share if any updates for this issue? Thanks.

Dazza0, 2019-08-19:
@MrAlexandrKolesnikov closing this issue due to lack of updates. Please feel free to open a new issue if you are still experiencing problems and suspect it is a bug in ESP-IDF. Please also take a look at the  ESP32 ECO document to see if any of the CAN Errata are relevant to the issues you are facing.

Dazza0, 2020-05-21:
Reopening as further testing has shown a possible edge case that can cause Bus Off Recovery to get stuck. When the Bus Off interrupt occur, the REC can still increase after Bus Off. The ISR handles this by freezing both counters. However, if the ISR is delayed in running, the REC will be non zero when Bus Off recovery begins. This in turn will result in interrupt that indicates Bus Off recovery completion to fail to trigger.

Workaround: In the ISR that handles Bus Off condition. Set the TEC to 0 and the back to 255 immediately. I'll push a commit to fix this shortly.

closed 2021-04-01 by espressif-bot