espressif/esp-idf #2522: ESP32 Wrover Flash/Monitor/Boot issue

ctag-fh-kiel, 2018-10-05:
## Environment

- Core (if using chip or module): [ESP32-Wrover]
- IDF version
- Development Env:		[Make]
- Operating System:		[Ubuntu VirtualBox under Win10]
- Power Supply:         [USB|external 5V]

## Problem Description

Hi, I have designed a PCB with a WROVER module and the SiliconLabs CP2102N with the option to use external power or USB power.
I also use the SDCARD interface, where IO2 is in use. I adapted the USB circuitry from the WROVER dev board.

I have frequently issues with flashing / booting and / or monitoring.
Flashing: 
Sometime the module seems dead, i.e. flash process doesn't start, can't connect to device.

Booting (Occasional stuck in boot process):
rst:0x10 (RTCWDT_RTC_RESET),boot:0x2b (HSPI_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 

Monitoring (get into monitoring mode but then nothing is being received from the module): 
Leaving...
Hard resetting via RTS pin...
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

Schematics are here: 
https://drive.google.com/file/d/1Npo_qVMd848Ku3gqBL1uUXvQw_8Wpzlh/view

Thanks a lot for any hints...

igrr, 2018-10-12:
boot:0x2b means that at reset, GPIO0 was low but GPIO2 was high (see https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#boot-mode-message). This indicates a problem with your boot mode selection circuit. On WROVER-KIT boards starting from version 3, there is a third transistor which forces GPIO2 low together with GPIO0, when entering download mode. If you already have this transistor in your design, then my suggestion would be to connect a logic analyzer and observe the sequence in which bootstrapping signals change when esptool is trying to reset the module.

Alvin1Zhang, 2018-10-24:
@ctag-fh-kiel Could you help share if any updates for this issue? Thanks.

ctag-fh-kiel, 2018-10-24:
I am currently lacking the hardware, but will provide more details on this
issue asap. It also happens when using a native Mac environment... Thanks
for keeping the issue open for the moment.

ctag-fh-kiel, 2018-12-01:
Ok, I have been further investigating the issue. It seems that it was a bad wrover module. I have purchased 4 Wrovers from Aliexpress (2 without IPEX and 2 with). All 4 of them have issues it seems. 
Now I am using the 16MB Flash version available at Mouser, no issues so far --> closing this issue.

closed 2018-12-01 by ctag-fh-kiel