espressif/esp-idf #2527: Got application/json as default response content type

trinhvc, 2018-10-06:
Hi,
I'm having a problem with http server in the master branch. That is:
As stated in following function's description: httpd_resp_send, httpd_resp_send_chunk, httpd_resp_set_type; If no status code and content-type were set, the default content type is 'text/html'.

But when I send response without setting the content type as below:
httpd_resp_send(req, buf, buf_len);
The content type I get is application/json

I found that there may be a mistake here in file httpd_parse.c, line 570:
ra->content_type = (char *)HTTPD_TYPE_JSON;

I don't know if anyone having the same problem like me.
Or could any one confirm about the issue.
Thank you,

chegewara, 2018-10-06:
According to docs it should be `text/html`
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/http_server.html#_CPPv219httpd_resp_set_typeP11httpd_req_tPKc

anurag-kar, 2018-10-08:
Yes , that seems like a mistake. Line 570 should read 
```
ra->content_type = (char *)HTTPD_TYPE_TEXT;
```
Thanks for letting us know. Will fix this issue ASAP

closed 2018-10-18 by igrr