espressif/esp-idf #2531: Can't connect to access point

almini, 2018-10-08:
## Environment

- Development Kit: ESP32-bit
- IDF version: 7313e39fd
- Development Env:	Make
- Operating System: Arch Linux
- Power Supply: USB

## Problem Description

Fails to connect to access point (iPhone hotspot).
esp-idf v3.1 works without issues, but current master does not.

### Expected Behavior

Connect successfully.

### Actual Behavior

Fails to connect.

### Steps to repropduce

1. clone https://github.com/espressif/esp-idf-template
2. adjust ssid and password
3. setup esp-idf
4. make erase_flash
5. make flash

### Code to reproduce this issue

```cpp
#include "freertos/FreeRTOS.h"
#include "esp_wifi.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"
#include "driver/gpio.h"
#include "esp_log.h"

const char* TAG = "Wifi";

esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch (event->event_id) {
        case SYSTEM_EVENT_STA_START:
            ESP_LOGI(TAG, "Wifi driver started.");
            break;
        case SYSTEM_EVENT_STA_CONNECTED: {
            ESP_LOGI(TAG, "Wifi connected.");
            break;
        } case SYSTEM_EVENT_STA_GOT_IP: {
            ESP_LOGI(TAG, "IP Address: %s", ip4addr_ntoa(&event->event_info.got_ip.ip_info.ip));
            break;
        } case SYSTEM_EVENT_STA_DISCONNECTED: {
            ESP_LOGI(TAG, "Disconnected (reason=%d).", event->event_info.disconnected.reason);
            // TODO: Cleanup
            ESP_ERROR_CHECK(esp_wifi_stop());
            break;
        } case SYSTEM_EVENT_STA_STOP:
            ESP_LOGI(TAG, "Wifi driver stopped.");
            break;
        default:
            break;
    }
    return ESP_OK;
}

void app_main(void)
[... 672 lines cut ...]
CONFIG_SPIFFS_API_DBG=
CONFIG_SPIFFS_GC_DBG=
CONFIG_SPIFFS_CACHE_DBG=
CONFIG_SPIFFS_CHECK_DBG=
CONFIG_SPIFFS_TEST_VISUALISATION=

#
# TCP/IP Adapter
#
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_TCPIP_LWIP=y

#
# Virtual file system
#
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096
```

markfink, 2018-10-21:
+1
I have a similar issue. And I know that my hardware is fine since connecting to the AP using AmazonFreeRTOS works fine. Anyway I want to use ESP-IDF but I can not figure out how to fix the wifi connection so far. I tried porting aws_wifi with little success yet.
I tried the following samples (none works for me):
* esp-idf/examples/wifi/getting_started/
* /home/mark/esp/esp-idf/examples/protocols/aws_iot/

what would be a good / the best approach to get proper wifi connectivity with WPA2 and DHCP on ESP-32?

liuzfesp, 2018-10-23:
Hi @almini  @markfink,  could you help to try IDF commit d57132b2129838a0d7193fbebc363391ba3e1b6d? 

In your demo code, **the WiFi only have one chance to connects the AP**, it's not robust because the first time connecting may fail.  The most common way is to call `esp_wifi_connect()` instead of `esp_wifi_stop()` when getting event SYSTEM_EVENT_STA_DISCONNECTED.

markfink, 2018-10-23:
Hi @liuzfesp... I realized that my wifi problem was in fact caused by a silly mistake. I spelled the AP name in lowercase letters instead of uppercase - my bad. I still need to investigate what aws-wifi does differently and what they do to make it more resilient... once I find some time to do so I will open a separate ticket for that.

almini, 2018-10-23:
Hello  @liuzfesp,
thanks for the tip. After a few tries it successfully connects to my iPhone AP. 
v3.1 is able to always connect on the first try and it therefore needs much less time to connect.

closed 2018-10-24 by Alvin1Zhang