espressif/esp-idf #2532: COMPONENTS should ignore test directories

nicola-lunghi, 2018-10-08:
feature request: i want to add a single component dir without adding the full component folder

I have an external project with a components subfolder

like this

```
my_awesome_component
├── component.mk
├── include
│   └── my_awesome.h
├── awesome.c
├── awesome_obj.cpp
├── README.md
└── test
    ├── component.mk
    ├── awesome_obj_test.cpp
    └── awesome_test.c
```

The problem is if I want to add **only the component dir** to EXTRA_COMPONENTS_DIR but also the test component folder is picked up

I solved this with

```
EXTRA_COMPONENT_DIRS += my_awesome_component_path
EXCLUDE_COMPONENTS += test
```

Could be nicer to add in 
https://github.com/espressif/esp-idf/blob/221eced06daff783afde6e378f6f84a82867f758/make/project.mk#L148-L155

a line like
```
COMPONENTS := $(filter-out $(subst ",,test, $(COMPONENTS)) 
```

Also I don't know how this can affect cmake

projectgus, 2019-03-01:
Hi @nicola-lunghi ,

Sorry noone got back to you about this.

I think the root cause should be fixed in 7df598a0626456afa71c629bba434ef8892c2728 now. Can you please try with current master and see if this is still an issue?

Thanks and sorry again,

Angus

Alvin1Zhang, 2019-05-17:
@nicola-lunghi Hi, would you help share if any updates for this issue? Thanks.

nicola-lunghi, 2019-05-23:
Yes thanks

closed 2019-05-23 by nicola-lunghi