espressif/esp-idf #2536: TCP connections closed after a few hours

aksdfauytv, 2018-10-09:
## Environment

- Core: ESP-WROOM32
- IDF version: 3276a13
- Development Env:		Make
- Operating System:		Ubuntu
- Power Supply:         external 3.3V

## Problem Description

The bug was introduced in LWIP commit c600d1eb00dd654f5782eafc8bce2d2e98d78570.
When you look at line 1110, 1122 and 1133 in dhcp.c, you will see

    dhcp->t0_timeout = timeout;
    ...
    dhcp->t1_timeout = timeout;
    ...
    dhcp->t2_timeout = timeout;

The assignment looses upper 16 bits of timeout, since t0_timeout, t1_timeout and t2_timeout are unsigned shorts.
"Timeout" is in seconds, so right now maximum lease time supported is ~18 hours, which in and of itself isn't world ending.
The bigger problem manifests only for certain lease times, where after truncation timers end up in wrong order - t0_timeout is shorter than t1_timeout. This causes DHCP client to never renew lease and instead restarts dhcp process from scratch. This in turn causes LWIP to close all TCP connections.

### Expected Behavior

Open connections should not be reset every few hours.

### Actual Behavior

Open connections are reset every few hours.

### Steps to repropduce

1. Configure your DHCP server to give lease time of 10 days.
2. Start ESP32
3. Notice the timeouts are truncated, which changes their order. t0 is 12032 while t1 is 38784.
4. If you are patient, wait until first timer elapses. You will see it was t0, which causes DHCP to restart.

stuarthatchbaby, 2019-10-08:
We're getting hit with this. Comcast in the US set a 7-day t0 timer in some of their routers. When truncated to 16bits this ends up being just over 4h which is even shorter than t2 in that case. We too see persistent connections which are disconnected.

What was the rationale for changing DHCP_COARSE_TIMER_SECS in the IDF?

esp-HarveyRong, 2019-11-07:
@aksdfauytv The problem has been fixed and will be synced to Github over time

Alvin1Zhang, 2020-06-08:
@aksdfauytv Thanks for reporting and sorry for slow reply, the fix has been available at https://github.com/espressif/esp-lwip/commit/9b9d126b. Feel free to reopen if the issue still happens. Thanks.

closed 2020-06-08 by Alvin1Zhang